nkolban/esp32-snippets #813: The future of this repo ...

nkolban, 2019-02-16:
A couple of years ago I fell in love with ESP32.  Loved it, loved it, loved it (still do).  I tinkered, I created a repo for snippets for samples ... primarily for my own benefit.  When I would post to the forum, I could then post a link to a snippet and say "Does this help?".   I then started studying BLE and, knowing C++, started a class library for BLE.  It got some traction and good folks started playing with it.   Mr @chegewara started tinkering with it and we became Internet buddies.

In the summer of 2018, I got a new job in an area 100% unrelated to IoT and in an area where I had low skills ... it became untenable for me to devote time to my hobby of ESP32 as well as study for my new job and have a life (Battlefield V and fishing) outside of all this.  As such, I backed off ESP32 time.  I thought maybe I could continue to tinker occasionally ... but that hasn't happened.

Now let us talk about the future of this repo.

First, Mr @chegewara has first refusal at complete, 100% ownership.  If he wants this repo and wishes to manage it as he sees fit, then it is his for the asking.  If we he wants it, he can do anything with it ... ANYTHING.

Second, if Mr @chegewara doesn't want it, then should Espressif want its content ... it can become theirs.  I think that extends to Mr @chegewara as well.

Third, if neither of the above want it, then it falls to anyone else who may want it to have it.  If in the unlikely event that there are multiple folks wanting it, then a combination of @chegewara, @me-no-dev , Espressif and, if asked ... myself ... will choose.

If none of the above ... then the repo will NOT be deleted ... but it will not be tended to (by my) either.  I have switched off notifications on it.

As for the location of the repo ... since I no longer have skin in the game, I don't care if it remains in the nkolban group or elsewhere.  I do require that it continue to be Apache Licensed and free of charge for any and all uses (private or commercial).

And ... that's it ...

mitchjs, 2019-02-17:
@nkolban  I want to thank you for your work on the ESP32 BLE c++ classes
without them I would maybe never gotten through the native ESP-IDF BLE code
and got a working project!

chegewara, 2019-02-17:
Hi @nkolban 
when i started working with esp32 i knew nothing, literally nothing about embedded devices. I didnt know what is button debouncing. Then i found this awesome library. About BLE i knew only it is some sort of wireless communication. From your library and with your help i learned a lot, almost everything i know now.
I spent literally thousands of hours working with esp32 and learning how it works, studying ble, usb hid, esp-idf etc in past 18 months or so. With all respect to you and your great work sorry i have to say that, but i dont want this repository, its yours. I dont want to be insulted because repositories (this and arduino ble) are managed this way, only because people dont remember how hard it was to get c++ support in arduino-esp32 and how it slow down development.

@mitchjs You mentioned in other post i started working with this library because i needed it to my project, thats not true. I started to work with this library because i thought its awesome, i still think like that. I will fork it and develop it more and more without complaining that someone is doing it good or bad, because grown people should do it like that.

So, thanks You for all Neil. Without You i still would write simple code to blink LEDs, but now i know i can do projects like this one:
https://esp32.com/viewtopic.php?f=17&t=9292

BobaTrek, 2019-02-17:
Hi @chegewara 

I and 99.9% of the users of this code are thrilled with the work and effort that you have put in to this repository.  Please don't abandon the support of this repository under duress because a single person has insulted you.  If that single person does not like the way this repository is being managed, then let him create his own fork that he can manage any way he likes.  Not you!

mitchjs, 2019-02-17:
@chegewara more now that my guess was wrong that you NEEDED the BLE and worked on it cause it was awesome... I love your attitude… thank you again for all the work.... if you fork it... I will follow your fork for sure

mpbejo, 2019-02-20:
I am very sorry for what happened in these days. It may be true that the repository is not managed at best ( I am not a github expert). But the most important thing is that in these months I have found a special person (@chegewara) who has always helped me every time I needed it. Many people here could not have done anything if there had not been @chegewara responding to everyone. I hope he continues his work serenely.
Best Regards
Marco

1technophile, 2019-06-03:
I've just read this discussion and want to thank you @nkolban and @chegewara for your huge contributions. Good luck Niel to your new job and wish you the best @chegewara with your interesting paraglider assistant project !

mishafarms, 2020-01-01:
Thank you all for your work on this repository.

I have found it useful.

I am not offering any support or anything, But I did get the cpp_utils and bootwifi compiling with v4.0 of the IDF. so if anyone wants it https://github.com/mishafarms/esp32-snippets branch v4.0. compiles using CMakeLists.txt.  You should add the 2 directories into the top level CMakeLists.txt like this
`set(EXTRA_COMPONENT_DIRS "components/esp32-snippets/cpp_utils components/esp32-snippets/networking/bootwifi")`

Hope this is useful to someone.
Michael

gb53smith, 2020-01-29:
I am finding source code in many places.

1. X:\Arduino\libraries
2.  .platformio/packages/framework-arduinoespressif32/libraries/
3. C:\Users\myname\.platformio\lib\ESP32 BLE Arduino_ID1841
4. https://github.com/espressif/arduino-esp32
5. https://github.com/nkolban/esp32-snippets
6. https://github.com/nkolban/ESP32_BLE_Arduino

How are these related?  
What is the best stable version to use? 
How are updates propagated?
I am using #2 -  .platformio/packages/framework-arduinoespressif32/libraries/  with some success implementing a UART server.

chegewara, 2020-01-31:
I think 4. is the best at the moment. At least i am using it with few minor changes.

me-no-dev, 2020-01-31:
2 should be the same as 4. both come from the same source.
6 used to be the source for 2 and 4, but after 5 went defunct, 4 had to split (therefore 2 as well)

gb53smith, 2020-01-31:
2 is older than 4. see changes to setPower in BLEDevice.cpp.  (one of the bugs hat affected me)
I agree that 4 is the maintained source for updates although these changes are slow or don't to propagate to Arduino or Platformio libraries.  Unfortunately, the [readme ](https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/README.md)file in 4 indicates that the source code is maintained in 5 which is not true.  Also, the library.properties file is misleading as these are not the sources files for version 1.0.1 in #1. Maybe add a note or remove this file.

me-no-dev, 2020-01-31:
> although these changes are slow or don't to propagate to Arduino or Platformio libraries

4 is the github version that is released to "Arduino" and PIO. It's normal to have some changes from the releases... it's what we work on.

> Unfortunately, the readme file in 4 indicates that the source code is maintained in 5 which is not true. Also, the library.properties file is misleading as these are not the sources files for version 1.0.1 in #1. Maybe add a note or remove this file.

Yup maybe something should get done.

F99, 2020-10-01:
Hi!
On behalf of 100.000 people that owe NK and never paid, 
I'd like to express my honest gratitude and best wishes for the future.
The ESP world would be much less inviting without your shared work.
regards

konacurrents, 2021-07-11:
Using an ESP32, and the Arduino  environment (not Espressif), I have good success with bridging MQTT and Bluetooth using the #6 version above (the ESP32_BLE_Arduino library). I can connect to another ESP32 running as a Bluetooth server.

Visiting the page says the code is now part of ArduinoBLE. But going there is states that ESP32 is not supported.

> Enables BLE connectivity on the Arduino MKR WiFi 1010, Arduino UNO WiFi Rev.2, Arduino Nano 33 IoT, and Arduino Nano 33 BLE.

So what is the future of BLE libraries for the ESP32? 
thanks.

gb53smith, 2021-07-11:
No future here.
Take a look at [ESPHome support of BLE](https://esphome.io/components/esp32_ble_server.html?highlight=ble) on ESP32.
They have copied a lot of this code and made some useful BLE components.  There will be some community support at ESPHome.  
Compared to Arduino or PlatformIO there is hardly any programming involved.  If you are using Home Assistant you don't even have to setup MQTT as they have an API that is well integrated with Home Assistant.  You can setup a MQTT component with ESPHome if you need to for some other controller/front end.  
I tried examples from this repository but went with a [Motenio M0 board](https://lowpowerlab.com/shop/product/184) with a RFM69 radio  and built a RFM69 to WiFi gateway using [MySensors](url). 
Much better range than BLE.

chegewara, 2021-07-11:
@gb53smith for you maybe there is no future here, for me its a lot future. If you dont believe you can check my upwork profile.
With this library, either with arduino or esp-idf, i am earning $$$.

ESPHome is maybe good, but not for everything, just like AWSfreeRTOS, it all depends what you need
A lot people is trying to move and to use nimble library by @h2zero.
RFM69 maybe have better range, but what standard it is? Can you use it with iBeacons, with sensors which are on market or even with smartphones? Good for you that you have found something which is giving you fun, but please dont speak about things you dont know

1technophile, 2021-07-11:
> So what is the future of BLE libraries for the ESP32?

h2zero library is a good way to go:
https://github.com/h2zero/NimBLE-Arduino

It leverages NimBLE instead of Bluedroid, less memory, and less program size, among other advantages.

gb53smith, 2021-07-12:
I built both a BLE client and a BLE to WiFi/MQTT gateway using code from this repository. 
Both the BLE 4 client and gateway had to be in the same room for reliable communication.  The ESP32 BLE client power consumption was too high to run off a battery.  
Sure, if you have to communicate with commercial BLE devices a ESP32 gateway is a good solution. 
I found that the ESP32 was not suitable for low power, long distance communication applications.
The ESP32 C3 with BLE 5 may be worth a second look but I doubt power used processing the BLE protocol would improve much.

h2zero, 2021-07-12:
There is nothing inherently wrong with this repo and no reason to consider it without a future. Sure there are some things that could be better and some inefficiencies, but it does the job well enough. I've used this for years and it's great, that's why my NimBLE repo was based on it.

>Both the BLE 4 client and gateway had to be in the same room for reliable communication.

That's not a typical issue for most people, I would guess you had a module/config problem if that is the case.

>I found that the ESP32 was not suitable for low power, long distance communication applications.

This is sadly true no matter what you do, if you need a BLE device to run on a CR2032, esp32 is not the optimal choice.