nkolban/esp32-snippets #819: When does "onRead" callback call?

MikhailNatalenko, 2019-02-22:
I use `BLECharacteristicCallbacks` class for writing and reading characteristics. 

I thought that `onRead`callback calls **before** the actual reading moment, so you can set BLE value in this step, but it seems that onRead calls **after** actual reading. That thing prevents me to bind my model and characteristics. 

Am i understand this thing well, and can you suggest us some of best practice for binding model and ble?

mpbejo, 2019-02-22:
Hi I am new on BLE, but in my little experience, the onRead and the onWrite are events fired when You try to read and write the characteristics.
I do this in my project:
1 the client write the characteristic on the server "Hey, somenthing new?"
2 the onWrite in the server is fired, if  there is something new the server write the new value in the characterisic 
3 after some milliseconds the client read the characteristics and have the new value.

MikhailNatalenko, 2019-02-23:
I see. I forgot to explain what i want to achieve. 

I have a model, that contains some value (just a class with data). And also there is a BLE value, that actually should present the model data. I can copy my model's data to BLE stack in loop for example, but i think it is not a food way. It would be more convenient to bind it somehow. And `onWrite` `onRead` can make this binding. 

Advantages: 
Async binding, BLE just calls 'onWrite' 'onRead' and our model become updated "automatically"

Disadvantages:
??? 
Maybe it's convinent only for my purposes? I would like to know your opinion.

chegewara, 2019-02-23:
@MikhailNatalenko You are correct, it was my mistake and onRead callback should be called before sending response. Here is code that needs to be changed to fix it in arduino and which seems to be fixed in this repo:
https://github.com/nkolban/ESP32_BLE_Arduino/blob/master/src/BLECharacteristic.cpp#L385-L407
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLECharacteristic.cpp#L389-L414

MikhailNatalenko, 2019-02-26:
@chegewara  Oh. nice to hear it. 
Should i make a pull request? I'm working on a project with remote CI, it compiles firmware for me with standart libraries (platformio), so i can't solve this problem with haks. 

Maybe i can help you to fix this bug ASAP

MikhailNatalenko, 2019-03-01:
@chegewara I'm sorry, when this bugfix will appear in arduino repo? I really need that fix

chegewara, 2019-03-02:
Hi @MikhailNatalenko 
sorry, but i dont maintain any longer BLE library.

vicatcu, 2019-12-19:
fwiw, as of today, on 1.0.4 of the arduino-esp32 BSP, this seems to behave as one would expect. Before returning data to the client (i.e. phone) the onRead callback provides an opportunity for the firmware to modify the value of the characteristic (i.e. call pCharacteristc->setValue...) and that is what the client will receive.