nkolban/esp32-snippets #820: Adding WiFi to a BLE sketch => immediate crash of the app

cristian-spiescu, 2019-02-22:
Hello,

I am aware that combining BLE + WiFi may not be trivial. I.e. to make the 2 services cohabit in a production quality app. I'm not there yet. I have a more trivial issue; I didn't spot posts about issues that manifest as "wild" as mine. **Explanation**:

When I modify a working BLE sample (such as `BLE_scan`) to make it work with wifi, and when I add e.g. the instruction `WiFi.mode(WIFI_STA);` the application resets reporting the following:

```
ets Jun  8 2016 00:22:57

rst:0x3 (SW_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1100
load:0x40078000,len:10088
load:0x40080400,len:6380
entry 0x400806a4
```

I also tried [this sample](https://github.com/nkolban/esp32-snippets/issues/685#issuecomment-454118436) from @chegewara. Same result.

Something curious that's maybe worth mentioning. When I compile the app including the BLE lib, even if I don't call any of its functions/objects, calling e.g. `WiFi.mode()` still results in a crash.

I'm using ESP32 board v1.0.1. The BLE lib v1.0.1. I'm using Sloeber (eclipse) to build. I have currently only one ESP32 board: [M5Stack Core](https://docs.m5stack.com/#/en/core/gray). May this be related to the peripherals installed on this board?

Thanks in advance for any hints!

Regards,
Cristian.

cristian-spiescu, 2019-02-22:
I think I found the issue. It seems that Sloeber doesn't verify if the total size of the app exceeds the capacity of the MCU. I discovered this while trying to run the sketch w/ Arduino IDE:

```
Sketch uses 1468614 bytes (112%) of program storage space. Maximum is 1310720 bytes.
```

After browsing a bit the code, I find that this BLE lib is extraordinary. Probably the nice OOP architecture takes a bit its toll on the flash size.

Tarem, 2019-02-23:
Stuck at this point as well a view days ago.
Actually there is a pretty simple solution to this. 

Watch this video:
https://www.youtube.com/watch?v=Qu-1RK4Fk7g

In short:
your ESP32 contains 4Mb storage. Without any adaptions to the Arudino IDE, you can only use about 1MB as default. If you are not interested in some "services" provided by the esp (OAT & Spiff) you can reduce their available memory. You will find these options in your Arduino IDE (Tools, Partition Scheme)
following the video, you can increase your program memory to 3145728 bytes with the Partition Scheme: "Huge APP (3MB no OAT)

`Sketch uses 1172673 bytes (37%) of program storage space. Maximum is 3145728 bytes.
Global variables use 42760 bytes (13%) of dynamic memory, leaving 284920 bytes for local variables. Maximum is 327680 bytes.`