nkolban/esp32-snippets #821: nRF24L01 using SPI in int, missing intr_flags initialization

DrClaudio73, 2019-02-27:
I had to add this line for initializing SPI bus:

bus_config.intr_flags      = ESP_INTR_FLAG_LEVEL1; //added to avoid error "intr flag not allowed". see esp_intr_alloc.h

after this addition everything worked perfectly.
I don't know if someone else encountered this problem.

SharpQuenty, 2019-05-09:
Hey, same here !
Not sure witch of these is allowed 
I tried ESP_INTR_FLAG_INTRDISABLED and ESP_INTR_FLAG_LEVEL6 didn't work.

#define ESP_INTR_FLAG_LEVEL1		(1<<1)	///< Accept a Level 1 interrupt vector (lowest priority)
#define ESP_INTR_FLAG_LEVEL2		(1<<2)	///< Accept a Level 2 interrupt vector
#define ESP_INTR_FLAG_LEVEL3		(1<<3)	///< Accept a Level 3 interrupt vector
#define ESP_INTR_FLAG_LEVEL4		(1<<4)	///< Accept a Level 4 interrupt vector
#define ESP_INTR_FLAG_LEVEL5		(1<<5)	///< Accept a Level 5 interrupt vector
#define ESP_INTR_FLAG_LEVEL6		(1<<6)	///< Accept a Level 6 interrupt vector
#define ESP_INTR_FLAG_NMI			(1<<7)	///< Accept a Level 7 interrupt vector (highest priority)
#define ESP_INTR_FLAG_SHARED		(1<<8)	///< Interrupt can be shared between ISRs
#define ESP_INTR_FLAG_EDGE			(1<<9)	///< Edge-triggered interrupt
#define ESP_INTR_FLAG_IRAM			(1<<10)	///< ISR can be called if cache is disabled
#define ESP_INTR_FLAG_INTRDISABLED

nurikk, 2019-06-19:
try to set 
`bus_config.intr_flags = 0;`

HVHung149, 2020-09-30:
Did you solve this problem yet? Please help me, I have the same problem, when I move all code in the spi_task() to app_main() .

> I had to add this line for initializing SPI bus in SPI.cpp in cpp_utils:
> 
> bus_config.intr_flags = ESP_INTR_FLAG_LEVEL1; //added to avoid error "intr flag not allowed". see esp_intr_alloc.h
> 
> after this addition everything worked perfectly.
> I don't know if someone else encountered this problem.
> 
> Thanks again to Mr. Kolban for precious work and engagement.

DrClaudio73, 2020-10-01:
Yes after adding this one line of code , it worked. You might refer to my repo nRF24L01-helloworld