nkolban/esp32-snippets #823: BLE for Wemos D1 Mini

rlbruderick, 2019-03-02:
I've had good success with your BLE libraries on these ESP32 boards: WEMOS LOLIN32 and MH-ET LIVE ESP32DevKit.  But I'd like to use the WeMos D1 Mini board for some projects.  These BLE libraries won't compile for it.  Are there BLE libraries that will work?  Are there any workarounds?  Everyone sells this WeMos D1 Mini board as a Bluetooth capable board, but I can't find any example of it being used with Bluetooth.  What am I missing?

2010019970909, 2019-03-25:
Hi, the [WEMOS D1 mini](https://wiki.wemos.cc/products:d1:d1_mini) does not embed an ESP32 processor but an ESP8266, which doesn't have integrated support for Bluetooth.

rlbruderick, 2019-03-31:
Good answer, though bad news. Thank you.