nkolban/esp32-snippets #825: ESP32 BT-HID Mouse arduino

senthilkumarh, 2019-03-03:
HI, I used the sketch you attached BLE_Mouse.ino and it is working fine with android phone but its not working when connected to windows10. Please can you help on this
[BLE_Mouse.ino.txt](https://github.com/nkolban/esp32-snippets/files/2923380/BLE_Mouse.ino.txt)

13080433, 2019-03-15:
@senthilkumarh 
Could you tell me what are the hardware connection for this HID mouse?
I just uploaded this code to my ESP32 device. But I don't know how to what are the pin connection for ESP32, arduino and othre requie hardware.
Could you tell me soe steps for this which can be appreciated.

looking forward to your response.

chegewara, 2019-03-15:
Just read code:
```
const int button1Pin = 39;     // the number of the pushbutton pin
const int button2Pin = 38;     // the number of the pushbutton pin
const int ledPin =     5;      // the number of the LED pin
```
I have m5stack with 3 buttons connected to pins 37, 38, 39. I am using button 1 (pin 39) and button 2 (pin 38) to test and move mouse cursor.