nkolban/esp32-snippets #829: max # of characteristics supported?

hellowtisch, 2019-03-08:
I have set up a BLE service and added 6 characteristics.
Any BLE client I used only finds 5 of the characteristics. The last one is always not found.
I changed the sequence of characteristics added and still, the last one added is not found.

Is there a limit on max # of characteristics supported per service on a esp32?

hellowtisch, 2019-03-08:
in addition, I have another strange observation:
for characteristics 0-3 I do get notifications, with characteristic 4 however, notifications do not work. It is possible to read the values, but changes do not get pushed. So this seems a strange behavior.
Could this be due to performance constraints on the ESP32?

chegewara, 2019-03-08:
numHandles:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEServer.h#L67

hellowtisch, 2019-03-08:
that works! 
@chegewara thank you very much!

closed 2019-03-08 by hellowtisch

oliwilson, 2019-06-12:
> that works!
> @chegewara thank you very much!

Could you possibly share the line of code you implemented to fix this issue? I have the exact same issue and am struggling to implement numHandles to resolve it without syntax errors.

hellowtisch, 2019-06-13:
here's the line of code that worked for me:
BLEService *pService = pServer->createService(BLEUUID(SERVICE_UUID), 30, 0);

Cheers

MohammedYusufMitha, 2021-01-26:
> here's the line of code that worked for me:
> BLEService *pService = pServer->createService(BLEUUID(SERVICE_UUID), 30, 0);
> 
> Cheers

I was facing the same issue, this resolved it, thanks a lot!!!

Sladerix, 2021-11-08:
I found a way that helps me to find the correct numHandles value. Is this making any sense?

`numHandles = (# of Characteristics)*2  +  (# of Services) + (# of Characteristics with BLE2902)`

the # of Characteristics is multiplied by 2 because of Reading Channel and Writing Channel (I supposed)

chegewara, 2021-11-09:
Yes, that seems to be correct, more or less.

Axliyor, 2023-08-31:
@chegewara thanks!!!

WelliSolutions, 2024-10-22:
The method is called `createService()`. Why should I add up the number of services then? What do the other services have to do with the service I am just creating?

chegewara, 2024-10-22:
Its not number of services. It is a number of handles reserved in service.
Each characteristic and descriptor requires some handles and also is occupying some memory, thus arbitrary default value.
If your setup requires more handles, because you are using a lot of characteristics and/or descriptors then you can use this param.

WelliSolutions, 2024-10-22:
So the [formula](https://github.com/nkolban/esp32-snippets/issues/829#issuecomment-963393585)
```cpp
numHandles = (# of Characteristics)*2  +  (# of Services) + (# of Characteristics with BLE2902)
```
should not include `(# of Services)`.

chegewara, 2024-10-22:
Probably yes.
Like i said, it is more or less correct formula. As you can see this formula does not include descriptors other than 2902.

You have 2 options:
- experiment with code and see how many handles you really need
- make some assumption and set higher value you need, it wont hurt too much in meaning of memory usage

PhLuReh, 2025-02-27:
Nice, that I finally found this solution here. But I don't know, why the API is not raising runtime errors (return values) or at least log outputs, which would make the behavior more predictable. I stumbled upon the issue while adding more characteristics and noticing that previously working functionality got lost again - or even LoadProhibited exceptions.

I would consider reopening this issue.