nkolban/esp32-snippets #833: ESP32 as BLE Server & Client, The connected server are also connected to my BLE server.

zz-zsys, 2019-03-11:
Hi everyone,

at the moment I'm trying to implement ESP32 as BLE Server and Client. The ESP32 as BLE Client are connected to 3-4 BLE Servers, collect the data from them and advertise it as BLE Server for the user. 
I'm using the latest BLE library file from cpp-utils. On the basis of the BLE Client example, I've tested every BLE Server individually and there is no problem on that. By implementing the ESP32 as BLE Server then however, I've noticed something but I'm not really sure if it is should be happened like that or not.
Before I told what I've noticed, let me described the flow of my BLE setup.

1. BLEDevice init
2. Setup the BLE scannner
3. Setup the BLE client for every BLE Server, later when connect to the device, notify callback will be set.
4. Setup the BLE server with Server Callback

So what I noticed are:
1.  When the BLE server is found, ESP32 will try to connect to it. After the connection is formed, onConnect BLEClientCallbacks is called, meaning the ESP32 is successfully connected to the server.
BUT then my BLE server callback is also called (as if they try to connect to my server, it shouldn't be because I'm the client and they are the server). By using the below function I could detected which mac address is connected to my server. And to my surprise, the server that I try to connect to are connected to my server. Perhaps anyone has any idea why it is happened? The onDisconnect server callback is not in cpp-utils, that's the one that I've added the same like the onConnect server callback.

```
class myServerCallback : public BLEServerCallbacks {

	void onConnect(BLEServer* pServer, esp_ble_gatts_cb_param_t *param) {

		char remoteAddress[18];

		sprintf(
			remoteAddress,
			"%.2X:%.2X:%.2X:%.2X:%.2X:%.2X",
			param->connect.remote_bda[0],
			param->connect.remote_bda[1],
			param->connect.remote_bda[2],
			param->connect.remote_bda[3],
			param->connect.remote_bda[4],
			param->connect.remote_bda[5]
		);

		ESP_LOGI(LOG_TAG, "myServerCallback onConnect, MAC: %s", remoteAddress);

		deviceConnected = true;
	}

	void onDisconnect(BLEServer* pServer, esp_ble_gatts_cb_param_t *param) {

		char remoteAddress[20];

		sprintf(
			remoteAddress,
			"%.2X:%.2X:%.2X:%.2X:%.2X:%.2X",
			param->disconnect.remote_bda[0],
			param->disconnect.remote_bda[1],
			param->disconnect.remote_bda[2],
			param->disconnect.remote_bda[3],
			param->disconnect.remote_bda[4],
			param->disconnect.remote_bda[5]
		);

		ESP_LOGI(LOG_TAG, "myServerCallback onDisconnect, MAC: %s", remoteAddress);

		deviceConnected = false;
	}
};
```

2. Furthermore, I try to connect to my server using NRF connect apps in Android. As the other BLE server also connected to my server, I noticed that if only two of them connected to my ESP32 server, I can connect to the server with my handy, but if three or more connected to my server, I couldn't connect to the server with my handy anymore. Is there any limitation on connected device besides the maximum connection of 7? I've set the CONFIG_BT_ACL_CONNECTIONS in sdkconfig.h to 7. Is there anymore definition that I should changed? Has anyone any solution on that?

Best regards / Viele Grüße,
ZZ

chegewara, 2019-03-11:
Hi
2. If you are using esp-idf then you have to change ble settings connection counts in menuconfig, but remember there is 2 places you have to change this value (for HCI driver and BLE)

1. looks like there is small bug, or rather missing small piece of code in BLEServer.cpp:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEClient.cpp#L218-L219
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEServer.cpp#L178

zz-zsys, 2019-03-12:
Hi @chegewara,

Thank you for your reply.

2. I saw your question on esp-idf github [https://github.com/espressif/esp-idf/issues/2524](url). I'm using Arduino IDE instead of esp-idf. Through esp-idf, if i open menuconfig, I could see the  Bluetooth ->Bluedroid enable -> BT/BLE MAX ACL CONNECTIONS option but not Bluetooth ->Bluetooth controller -> BLE Max Connections option. Is there any possibility to do that in sdkconfig.h or somewhere else in the files? Did you manage to solve it through esp-idf?

chegewara, 2019-03-12:
Yes, with esp-idf there is no problem anymore. With arduino you have to ask @me-no-dev to change it in settings for next compilation.

zz-zsys, 2019-03-12:
@chegewara Thanks again. I'll ask @me-no-dev  then or create an issue under arduino-esp32 repository then. Are you using the latest release from esp-idf?

Regarding problem 1, is that the place where the bugs may be happened? Thanks again!

zz-zsys, 2019-03-12:
I just update the latest release from esp-idf, and find Bluetooth ->Bluetooth controller -> BLE Max Connections option.

qlamerand, 2019-09-02:
@chegewara Hi, I have the same issue (1.) and tried to fix it by adding the tests you suggested in BLEServer::handleGATTServerEvent but it's not sufficient. My server callback is still called when my client connects to another server...
Any help would be welcome :)

chegewara, 2019-09-03:
@qlamerand I can try to help with this, but i need to see logs. What exactly i need is some logs from custom callbacks, nothing special just info about event type called:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEDevice.h#L60-L62

qlamerand, 2019-09-09:
Sorry, in the end we went back using ESP-IDF in pure C. Thanks anyway :) !

chegewara, 2019-09-09:
Its not a problem, sorry i couldnt help you more.

Sakib1263, 2019-10-13:
I wanted to connect 2 servers to one client (BLE-esp32). So far, I can do it 1 to 1. What can I change in the client code so that it can receive data simultaneously from both servers and display it in the serial monitor? I am providing both codes for reference. 

**Client Code:**
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEScan.h>
#include <BLEAdvertisedDevice.h>

static BLEUUID serviceUUID("7cddf5af-453f-40fa-808e-37ae6ad8facd"); //Interested Device or Service
static BLEUUID    charUUID("3383a686-a53c-42e4-a88c-e0e1de4c4bda"); //Interested Characterstics

static BLEAddress *pServerAddress;
static boolean doConnect = false;
static boolean doScan = false;
static boolean connected = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice *myDevice;
BLEScan *pBLEScan; //Name the scanning device as pBLEScan

static void notifyCallback(BLERemoteCharacteristic *pBLERemoteCharacteristic, uint8_t* pData, size_t length, bool isNotify) {
      /*Serial.print("Notify callback for characteristic ");
      Serial.println(pData[0]);
      for (int i = 0; i < length; i++) {
          Serial.print(pData[i]);
          Serial.print(" ");
      }
      Serial.println();*/
}

class MyClientCallback : public BLEClientCallbacks {
  void onConnect(BLEClient* pclient) {
  }
  void onDisconnect(BLEClient* pclient) {
    connected = false;
    Serial.println("onDisconnect");
  }
};

bool connectToServer() {
     Serial.print("Forming a connection to ");
     Serial.println(myDevice->getAddress().toString().c_str());

     BLEClient *pClient = BLEDevice::createClient();
     Serial.println("Client created");

     // Connect to the remove BLE Server.
     pClient->connect(myDevice);  // if you pass BLEAdvertisedDevice instead of address, it will be recognized type of peer device address (public or private)
     Serial.println("Connected to server");
     
     // Obtain a reference to the service we are after in the remote BLE server.
     BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
     if (pRemoteService == nullptr) {
        Serial.print("Failed to find our service UUID: ");
        Serial.println(serviceUUID.toString().c_str());
        return false;
       }
     Serial.println("Found our service");

     // Obtain a reference to the characteristic in the service of the remote BLE server.
     pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
     if (pRemoteCharacteristic == nullptr) {
        Serial.print("Failed to find our characteristic UUID: ");
        Serial.println(charUUID.toString().c_str());
        return false;
       }
     Serial.println("Found our characteristic");
  
     pRemoteCharacteristic->registerForNotify(notifyCallback);
      
     connected = true;
[... 110 lines cut ...]
 void loop () {
   if (clientConnected) {

     temperature = bme.readTemperature();
     humidity    = bme.readHumidity();
    
     char humid  [8];
     char temper [8];
     
     std::string temperatureString = dtostrf (temperature, 3, 2, temper);
     std::string humidityString = dtostrf (humidity, 3, 2, humid);
     std::string space = " ";
     //std::string space2 = "*C, ";
     //std::string space3 = "Humidity: ";
     //std::string space4 = "%Rh";
     std::string sensorDataString = temperatureString + space + humidityString + space;
     
     temphumidSense.setValue (sensorDataString);
     temphumidSense.notify();  // Send the Temperature reading to the application!

     Serial.print("Sent Temperature: " + String(temper) + "*C | ");
     Serial.println("Sent Humidity: " + String(humid) + "%Rh");
   }
     delay(500);
}