nkolban/esp32-snippets #836: BLESecurity how to.handle auth not complete

wegunterjr, 2019-03-14:
onAuthenticationComplete(esp_ble_auth_cmpl_t)

ESP_GAP_BLE_AUTH_CMPL_EVT

I am using this event but having issues if pin fails.  Is there an auth fail event?

chegewara, 2019-03-15:
You have to use the same callback. Here is how parameter structure looks like:
```
typedef struct
{
    esp_bd_addr_t         bd_addr;               /*!< BD address peer device. */
    bool                  key_present;           /*!< Valid link key value in key element */
    esp_link_key          key;                   /*!< Link key associated with peer device. */
    uint8_t               key_type;              /*!< The type of Link Key */
    bool                  success;               /*!< TRUE of authentication succeeded, FALSE if failed. */
    uint8_t               fail_reason;           /*!< The HCI reason/error code for when success=FALSE */
    esp_ble_addr_type_t   addr_type;             /*!< Peer device address type */
    esp_bt_dev_type_t     dev_type;              /*!< Device type */
    esp_ble_auth_req_t    auth_mode;             /*!< authentication mode */
} esp_ble_auth_cmpl_t;                           /*!< The ble authentication complete cb type */
```
Look at `success` value.

axa88, 2019-08-21:
Using the BLESecurity callback as mentioned here, where could i decode meaning of [esp_ble_auth_cmpl_t](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/esp_gap_ble.html#_CPPv419esp_ble_auth_cmpl_t) ?
It seems fail_reason in the GAP API [documentation ](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/bluetooth/esp_gap_ble.html#_CPPv4N19esp_ble_auth_cmpl_t11fail_reasonE) has no explanation.

Currently I always need to pair twice as the first pair results with a value of 81 (i think 0x51) .  Is this a bit mask for something?

trodrigu, 2020-08-10:
@axa88 I’ve ran into this as well.  Can you please show me what you mean when pairing twice?  I’ve been having trouble just getting them to pair.

axa88, 2020-08-10:
> @axa88 I’ve ran into this as well. Can you please show me what you mean when pairing twice? I’ve been having trouble just getting them to pair.

literally i would need to pair my esp32 device to my host, an Android phone, once to which it would fail not pairing, and then the second time to which it would pair.

here I was hoping to get a better description of the actual error.

chegewara, 2020-08-10:
I dont know that error description, but you could open issue on esp-idf or even try to find old issue.

jakubdybczak, 2020-08-23:
> literally i would need to pair my esp32 device to my host, an Android phone, once to which it would fail not pairing, and then the second time to which it would pair.
>
> here I was hoping to get a [...]

I have same issue - ESP32 only pairs device for second time. And I also have fail_reason = 81 (ACI_STATUS_ERROR_INTERNAL). Did you find any solution to this?

chegewara, 2020-08-24:
It is good practice to erase flash every time when you work with ble security, also you should clear pairing on smartphone each time. If you wont do that it may lead to many problems with testing/debugging.

mehmethekim, 2022-09-06:
I had the same issue; ESP_GAP_BLE_AUTH_CMPL_EVT event comes without success. I came across with esp's examples and they were using "esp_restart()" function to solve this issue. But it was inefficient. What you can do is, you can use `esp_ble_gap_disconnect(param->ble_security.auth_cmpl.bd_addr)` function to first disconnect from the device, then restart scanning process using `esp_ble_gap_stop_scanning()`and `esp_ble_gap_start_scanning(seconds)` functions. disconnecting solved my problem. The device automatically connects again and ESP_GAP_BLE_AUTH_CMPL_EVT gives success.