nkolban/esp32-snippets #839: Delete BLEDevice and then init new one in Arduino IDE

akotowski, 2019-03-20:
I am trying to "Delete BLEDevice and then init new one" as in Issue  #630 but using the Arduino IDE.
It works as far as BLEDevice::deinit(false).
When I try to setup the bluetooth device again BLEDevice::createServer hangs up waiting for a Semaphore.
Attached is the monitor  log.
[image]
Here is the code as a .TXT file.
[ESP32_BLE_Test.TXT](https://github.com/nkolban/esp32-snippets/files/2989241/ESP32_BLE_Test.TXT)
Here are the sections at issue:
`  Serial.println("deinit");
  BLEDevice::deinit(false);
  Serial.println("deinit complete");
  delay(5000);

  Serial.println("setupBluetooth");
  setupBluetooth ();
  Serial.println("setupBluetooth complete");`

`int setupBluetooth (){
  
   Serial.println("BLEDevice::init");
  // Create the BLE Device
  BLEDevice::init("WM4A");

   Serial.println("BLEDevice::createServer");
  // Create the BLE Server
  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

   Serial.println("pServer->createService");
  // Create the BLE Service
  BLEService *pService = pServer->createService(BLEUUID(SERVICE_UUID), uint16_t(30));              //Custom device service, call for 25 handles

  // Create the BLE Battery Service
  BLEService *pBatService = pServer->createService(BATTERY_SERVICE_UUID);   //Standard Batttery Service

  // Create BLE Characteristics
  
  pBatRCharacteristic = pBatService->createCharacteristic(                                //Battery Service Battery Level Characteristic (pBatService)
                      CHARACTERISTIC_UUID_BAT_STATUS,
                      BLECharacteristic::PROPERTY_NOTIFY | BLECharacteristic::PROPERTY_READ);
  pBatRCharacteristic->addDescriptor(new BLE2902());
  pBatRCharacteristic->setCallbacks(new MyBatCallbacks());                                //Callbacks (for Bat)

  pTempRCharacteristic = pService->createCharacteristic(                                  //Temperature Characteristic (pService)
                      CHARACTERISTIC_UUID_TEMP,
                      BLECharacteristic::PROPERTY_NOTIFY | BLECharacteristic::PROPERTY_READ);
  pTempRCharacteristic->addDescriptor(new BLE2902());
  pTempRCharacteristic->setValue("101.1");

  //Set Advertising INterval
  pServer->getAdvertising()->setMinInterval(64);         //no effect
  pServer->getAdvertising()->setMaxInterval(128);

  // Start the service
  pBatService->start();
  pService->start();
  pServer->getAdvertising()->start();
}
`
Any help or suggestions welcome.

bribur, 2019-05-16:
Did you get this to work?  I am having the same issue, locks trying to create server after deinit.

akotowski, 2019-05-16:
No, never found a solution.

arthurkafer, 2019-10-08:
`BLEDevice::init` allows only one initialization, and its commented on the BLEDevice.cpp https://github.com/nkolban/ESP32_BLE_Arduino/blob/adc2aee2f0d01eb6b30dd5ad3589f2cc89934beb/src/BLEDevice.cpp#L341

If you use `BLEDevice::deinit`, it sets `initialized = false;`, so you should restart your ESP if you need to reuse BLE in your code.

chegewara, 2019-10-10:
@arthurkafer 
First of all, link from your comment points to old library, it is no longer maintained. Now library is part of arduino-esp32 repository.
Obviously, when you deinit() BLE you have to init() it again to use it.

arthurkafer, 2019-10-11:
@chegewara 
My bad, didn't noticed about the repository. 
Anyway, some weeks ago I had the problem of deiniting BLE and initing it again, because it didn't work for me. In my case, after I call deinit(), my routine resets my ESP, so no problems

chegewara, 2019-10-12:
This shouldnt happen. If you have some logs i could look at, please paste it here.

chegewara, 2019-10-16:
Maybe this has something to do with it?
```
[E][BLEScan.cpp:221] start(): esp_ble_gap_set_scan_params: err: 259, text: Unknown ESP_ERR error
```

chegewara, 2019-10-16:
You cant deinit(true). This will release memory and you wont be able to use BLE anymore:
```
void loop() {
  delay(1000);
  // put your main code here, to run repeatedly:
  BLEScanResults foundDevices = pBLEScan->start(scanTime, false);
  Serial.print("Devices found: ");
  Serial.println(foundDevices.getCount());
  Serial.println("Scan done!");
  pBLEScan->clearResults();   // delete results fromBLEScan buffer to release memory

  BLEDevice::deinit(true); <--- HERE
  delay(2000);
  newSetup();
}
```

arthurkafer, 2019-10-16:
I've tried using `BLEDevice::deinit()` and `BLEDevice::deinit(false)`, it gets the same results and logs as releasing memory.

I'll get the logs tomorrow, I'm not at my work PC right now.
Thanks for the quick reply

arthurkafer, 2019-10-17:
I'm testing it now with the updated library, and it works fine with deinit(false). 

The problem happening with @akotowski might be about old libraries, trying his code here, it seems to be working fine too, just with a loop.

Akshay2804, 2020-04-19:
Hello @arthurkafer @akotowski @chegewara  !
I am working on similar application with ESP32 but I am unable to kill or de-initialize the BLE server and then create a new BLE or Bluetooth server.
Any help on the same will be really appreciated.

Ben79543, 2020-06-15:
> I'm testing it now with the updated library, and it works fine with deinit(false)

@arthurkafer Can you detail which is the updated library you used?
I am getting the same issue : "Scan works fine in the first try, but after calling deinit, it won't work until you reset ESP." and same "esp_ble_gap_set_scan_params: err: 259" error.

arthurkafer, 2020-06-18:
@Ben79543 [this](https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE) is the library you should use. There has been some recent updates almost every month, so if some new problem occours, try downloading an new update

arthurkafer, 2020-06-18:
> Hello @arthurkafer @akotowski @chegewara !
> I am working on similar application with ESP32 but I am unable to kill or de-initialize the BLE server and then create a new BLE or Bluetooth server.
> Any help on the same will be really appreciated.
@Akshay2804 
like chegewara said, if you want to call deinit BLE and still work on it on the code, you should call `BLEDevice::deinit(false);`

mattgaidica, 2024-12-12:
This still seems to be an issue 4 years later...

12:24:24.678 -> After sleep  - Free heap: 286520 bytes, Min free heap: 275912 bytes
12:24:24.743 -> Started advertising... 
12:24:30.294 -> Done advertising.
12:24:32.070 -> Before sleep - Free heap: 281992 bytes, Min free heap: 211132 bytes

This is trying to init/deInit BLE through many advertisement cycles with lightsleep in between. If deInit(false), the leak still occurs but the program runs OK. If deInit(true), the leak still occurs but the program crashes after a few bouts of advertising.

mattgaidica, 2024-12-14:
I had to patch multiple core files to get this working. You would think other people need to toggle BLE, or maybe they are and just not looking for the 1-5kB leaks. I'm going to contribute those changes, hopefully they get picked up in v>3.7.0.

chegewara, 2024-12-18:
Hi,
if you want to contribute you should do it in arduino-esp32, because this repository is abandoned and included in espressif arduino-esp32.