nkolban/esp32-snippets #842: ESP32 crashes (when used along with MQTT over ssl)

m-menon, 2019-03-25:
I encountered an (strange?)error when running my Arduino Sketch on a ESP32 dev board. I have two different sketches for testing purposes. 

1. Uses [MQTT](https://github.com/knolleary/pubsubclient) (over SSL) library. 
2. Uses this [ESP32 Bluetooth LE library](https://github.com/nkolban/ESP32_BLE_Arduino).

Both these works perfectly well when used independently. There's a sketch that uses both these libraries and then the ESP32 crashes repeatedly! I have tried all possibilities and poked around with the code since couple of days but I can't seem to get a fix. 

I am still pretty new to ESP32/Arduino etc so any pointers would greatly help. Here are some logs. I even used the [ESP Exception library](https://github.com/me-no-dev/EspExceptionDecoder) to copy paste the trace and it shows me additional information regarding the crash.

```
Connecting to Home-WiFi
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:345] _eventCallback(): Event: 0 - WIFI_READY
. [D][WiFiGeneric.cpp:345] _eventCallback(): Event: 4 - STA_CONNECTED
. [D][WiFiGeneric.cpp:345] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:389] _eventCallback(): STA IP: 192.168.1.201, MASK: 255.255.255.0, GW: 192.168.1.188
.  ...connected!
mqttServerIP = xxx.xxx.xxx.xxx (IP masked)
MQTT connecting ...[I][ssl_client.cpp:45] start_ssl_client(): Free heap before TLS 116464
[I][ssl_client.cpp:47] start_ssl_client(): Starting socket
[I][ssl_client.cpp:75] start_ssl_client(): Seeding the random number generator
[I][ssl_client.cpp:84] start_ssl_client(): Setting up the SSL/TLS structure...
[I][ssl_client.cpp:97] start_ssl_client(): Loading CA cert
[I][ssl_client.cpp:153] start_ssl_client(): Performing the SSL/TLS handshake...
[I][ssl_client.cpp:173] start_ssl_client(): Verifying peer X.509 certificate...
[I][ssl_client.cpp:183] start_ssl_client(): Certificate verified.
[I][ssl_client.cpp:186] start_ssl_client(): Free heap after TLS 72128
connected
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c46c  PS      : 0x00060730  A0      : 0x8012b7d6  A1      : 0x3fff6950  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x0000001c  A5      : 0x00000000  
A6      : 0x40204dbc  A7      : 0x00000001  A8      : 0x800eeb4c  A9      : 0x3fff6910  
A10     : 0x3fffffc0  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3fffff60  
A14     : 0x3fffc918  A15     : 0x00000084  SAR     : 0x00000008  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4000c46c:0x3fff6950 0x4012b7d3:0x3fff6960 0x4012c0b5:0x3fff6980 0x40107eb5:0x3fff69a0 0x40107eef:0x3fff69c0

Rebooting...
ets Jun  8 2016 00:22:57
```
The result from the ESP Exception Decoder

```
PC: 0x4000c46c
EXCVADDR: 0x00000000

Decoding stack results
0x4012911f: bta_ar_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/ar/bta_ar.c line 72
0x40129a01: bta_sys_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/sys/bta_sys_main.c line 186
0x401057c5: btu_task_start_up at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btu/btu_task.c line 301
0x401057ff: btu_task_thread_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btu/btu_task.c line 226
```

m-menon, 2019-03-25:
I moved the BT initialization to bottom ( that is after MQTT connection is successful). So now the MQTT connects properly every time but the moment I try to initialize the Bluetooth Library the ESP32 crashes. 
Greatly appreciate any pointers. Spent a great many hours poking at everything I possibly could.

The following lines are the ones used for initializing

```
      Serial.println(F("Initialize Bluetooth ..."));

      BLEDevice::init("");
      pBLEScan = BLEDevice::getScan(); //create new scan
      pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
      pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
      pBLEScan->setInterval(100);
      pBLEScan->setWindow(99);  // less or equal setInterval value
```

The crash log
```
Initialize Bluetooth ...
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c46c  PS      : 0x00060f30  A0      : 0x80129386  A1      : 0x3fff6d30  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x0000001c  A5      : 0x00000000  
A6      : 0x40202730  A7      : 0x00000001  A8      : 0x800ec6dd  A9      : 0x3fff6cf0  
A10     : 0x3ffffe9c  A11     : 0x00000000  A12     : 0x00000001  A13     : 0x3ffffe0c  
A14     : 0x3fffccfc  A15     : 0x00000084  SAR     : 0x00000008  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4000c46c:0x3fff6d30 0x40129383:0x3fff6d40 0x40129c65:0x3fff6d60 0x40105a2d:0x3fff6d80 0x40105a67:0x3fff6da0

Rebooting...
ets Jun  8 2016 00:22:57
```

The result from [ESP exception decoder](https://github.com/me-no-dev/EspExceptionDecoder)
```
PC: 0x4000c46c
EXCVADDR: 0x00000000

Decoding stack results
0x40129383: bta_ar_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/ar/bta_ar.c line 72
0x40129c65: bta_sys_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/bta/sys/bta_sys_main.c line 186
0x40105a2d: btu_task_start_up at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btu/btu_task.c line 301
0x40105a67: btu_task_thread_handler at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/bt/bluedroid/stack/btu/btu_task.c line 226
```

[image]

m-menon, 2019-03-28:
Ok this is related to the memory issues when using WiFi + BLE. I'll work my way around this. Thanks anyways.

closed 2019-03-28 by m-menon

tommy-mor, 2019-05-06:
What do you mean related to memory issues? Having same problem. Did you work your way around this?

mmrvelj, 2020-05-22:
I am experiencing the same issue. 

Using MQTT without SSL works, but with SSL either fails SSL connection or crashes system (depends on the order of invocation).