nkolban/esp32-snippets #843: Connection to all gatt server break down when disconnecting one of gatt server

zz-zsys, 2019-03-25:
For my test function, I'm connecting one client to multiple server (the plan was 5 servers in total but I could only connected to 3 servers in total). The test platform is the latest BLE library from cpp-utils under Arduino IDE.

Using the esp-idf with the same release as arduino-esp32, I managed to compiled the new BLE stack static library, so that maximum 7 devices could be connected at the same time, but it was not successful, where the maximum connected device are still up to 3 devices. So I'm just stick to 3 devices for further testing.

However, what I realise is, when disconnecting the pClient using disconnect function, all the connected servers are also disconnected. Is this library related? or esp-idf stack related?

In disconnect function, esp_ble_gattc_close function is used. Is there any reason why this function is used instead of esp_ble_gap_disconnect? I try to test both functions but it seems both give same result. Any advice?

chegewara, 2019-03-25:
Hi,
if you want to compile esp-idf and use in arduino like you did, you have to change 2 settings in menuconfig to connect more than 3 devices. Even if you can setup more than 6 max allowed connections is 6 https://github.com/espressif/esp-idf/issues/2914

> However, what I realise is, when disconnecting the pClient using disconnect function, all the connected servers are also disconnected. Is this library related? or esp-idf stack related?

Its hard to say, i think its still issue in this library, but some time ago when i asked the same question on esp-idf issue tracker ive been told its how esp-idf works.

zz-zsys, 2019-03-25:
> if you want to compile esp-idf and use in arduino like you did, you have to change 2 settings in menuconfig to connect more than 3 devices. Even if you can setup more than 6 max allowed connections is 6 [espressif/esp-idf#2914](https://github.com/espressif/esp-idf/issues/2914)

i do read your question on esp-idf https://github.com/espressif/esp-idf/issues/2524#issuecomment-427734622 and based on that I have set both of them as 7 but still gattc_if still hanged when gattc_if = 6 arrived. The improvement that I see between the available bt-stack on arduino-esp32 and the new bt-stack is when connecting gattc_if begin from 3 on the new one, where gattc_if on default bt-stack begin from 4.
 
> Its hard to say, i think its still issue in this library, but some time ago when i asked the same question on esp-idf issue tracker ive been told its how esp-idf works.
I think I do read about your question on it also somewhere either in the forum or in github.

gmartau, 2019-05-15:
I did not open another topic because appears to be related.

I have three sensors and initially I connected to each with its own client object using the known address.
If there are clients already connected and working (the notifications are received) and the client try to connect to a remaining sensor that is turned off, the client receive the disconnect event for the failed connection and also there are disconnect notifications for all already connected clients (the clients do not share the notification classes).
To avoid this problem, I scan for client and try to connect only if it is advertised. The problem is that if a sensor become unavailable, there is a disconect event for it and for all connected the clients. However, the connected clients continue to receive the notifications.
I enabled the debug log and I don't find a problem in the library. The unexpected disconnect event is raised with specific Id for the connected clients.