nkolban/esp32-snippets #847: Multiple Beacon broadcasting at same time

rohitjust24idpl, 2019-04-02:
@chegewara @nkolban 
This looks like a superb library to quickly implement the beacons ( iBeacon and Eddystone ) with just few lines of code.

**Current Scenario:** Right now on a button press, I am broadcasting Eddystone data for first 2 seconds and then iBeacon data for later 2 seconds. When I init for iBeacon, eddystone is automatically de-init. Which means i cannot create multiple BLEDevice objects at same time ?
Is that it ?

**Problem Statement:** Is there a way I can broadcast them at the same time, simultaneously? Using threads or blocks ?

chegewara, 2019-04-02:
Hi,
i asked espressif programmers similar question and ive been told that esp32 can advertise only 1 advertisement at the same time. Only option is to send 1 beacon, stop advertising and send other beacon.

In theory you could send 1 beacon as normal advertising and 1 beacon as scan response, but im not sure its good idea.

rohitjust24idpl, 2019-04-02:
@chegewara 
Thanks for your quick response.  

1 beacon advertising and another 1 beacon scan response does not suits my scenario.
Anyways this is not a show stopper for me. We are good to close this issue.

closed 2019-04-02 by chegewara