nkolban/esp32-snippets #851: Parsing BLEClient.getServices reply

LordJakson, 2019-04-10:
Hi I am new to advanced c++ programming and struggling with pointer conversions.

As described in the `BLEClient.h` file the getServices() function returns a pointer to a std::map of the lib.

`std::map<std::string, BLERemoteService*>* getServices();`

I then assign it to my local variable:

`std::map<std::string, BLERemoteService*> *foundServices = newClient.getServices();`

With `foundServices->size();` I get the amount of the found services but how can I get the UIDs or the service itself?
The standard iterators don't work because its a pointer and I found no example to cast or type it propertly.

`for (auto &myPair : foundServices)`

Thanks in advance for answering my noob question.

chegewara, 2019-04-10:
https://github.com/nkolban/ESP32_Explorer/blob/master/main/BLEExplorer.cpp#L153

LordJakson, 2019-04-10:
Ah you have to define the begin and end manually over pointer mebers.
I always thought I have to cast the result.
So this is it:

`for (auto myPair = foundServices->begin() ; myPair != foundServices->end() ; myPair++){}`

closed 2019-04-10 by LordJakson