nkolban/esp32-snippets #852: Can the server get the rssi of the connected client?

Wyctus, 2019-04-10:
I would like to know how far is the client which tries to connect to the server. 
That's why I planning to approximate it based on the rssi. 
I spent hours looking for such function in the esp-idf, but can't really find it.

In the ```onConnect(BLEServer* pServer, esp_ble_gatts_cb_param_t* param)``` function I found the client address in the ```param``` and seems like the ```esp_ble_gap_read_rssi(esp_bd_addr_t remote_addr)``` can help me, but I'm stucked since it doesn't return the rssi, only a status message.
How can I get the rssi value?

Thanks in advance!

chegewara, 2019-04-10:
Hi,
yes, there is no implemented in BLEServer function to read RSSI, but you can use method you described. What you will need to add is custom gap handler and in that handler you will need to catch rssi read event.

Wyctus, 2019-04-11:
Thank you, works fine!

closed 2019-04-11 by Wyctus

Consistometru, 2020-06-03:
Hi,
Is there an example of this function for Arduino IDE? I'm having a hard time translating from ESP-IDF

chegewara, 2020-06-03:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEClient.cpp#L366