nkolban/esp32-snippets #853: Notifications for several bluetooth clients.

ArTyGill, 2019-04-11:
Esp32 acts as a server, several devices are connected to it. And at the moment, I could send a response notification only to the last connected client. I propose to add the ability to send a message to all clients simultaneously, and to each individually, in the " void BLECharacteristic :: notify (bool is_notification)" function.

ArTyGill, 2019-04-11:
You can use conn_id as an identifier for sending data.

chegewara, 2019-04-11:
Hi @ArTyGill 
as far as i remember notifications should be send to all connected devices, but its possible your test are more thorough and i missed something.

ArTyGill, 2019-04-11:
@chegewara yes that's right, I use the old library, and I fixed it myself, but sometimes the notification can only be addressed to one of the clients, if the client writes a characteristic to ble, I use its conn_id to send a response message, and then the notification comes only to a specific client , I also realized for myself the possibility of sending messages to all clients if I pass to the function -> notify (-1) instead of conn_id.