nkolban/esp32-snippets #854: Unable to reconnect properly

bilogic, 2019-04-14:
Hi,

I'm trying to turn the ESP32 into a BLE to wifi proxy for my MiScale.
Briefly, 

1. My sketch calls ```createClient()```, connects to the MiScale and enables notifications.
2. However, this connection disconnects after about 30 seconds, ```onDisconnect``` gets printed on the serial, not sure if disconnection was by the ESP32 or MiScale.
3. After disconnection, the MiScale no longer notifies the ESP32 when there are new measurements.
4. Thus, I made the sketch reconnect upon disconnection.
5. However, after about 2-3 cycles of this re-connection, the ESP32 hangs inside ```pClient->connect(pAddress);```.
6. I inserted a flag to call ```pClient = BLEDevice::createClient();``` only once per reboot, but it made things worse, now the ESP32 cannot reconnect even 1 time

I see #209 and #397 as being related issues, but both are already closed.
Also my ESP32 BLE Arduino Library is from https://github.com/wakwak-koba/ESP32_BLE_Arduino as it fixes a stack smashing issue.

My questions would be:

1. Is disconnection supposed to take place?
2. I had the impression that it is possible to achieve push notifications from the MiScale, but if the ESP32 has to keep reconnecting, I assume this is going to deplete the MiScale's battery. Is this correct?
3. Any ideas how push notifications can be achieved?
4. If re-connection is the only option, any idea why the ESP32 is hanging?

Thank you.

```
/*
 *
 * This sketch emulates a Heart rate watch and is able to receive BLE signals of a Polar H7 Heart Rate Sensor.
 * It shows the received values in Serial and is also able to switch notificaton on the sensor on and off (using BLE2902)

   Copyright <2017> <Andreas Spiess>

  Permission is hereby granted, free of charge, to any person obtaining a copy of this software and associated documentation files (the "Software"),
  to deal in the Software without restriction, including without limitation the rights to use, copy, modify, merge, publish, distribute, sublicense,
  and/or sell copies of the Software, and to permit persons to whom the Software is furnished to do so, subject to the following conditions:

  The above copyright notice and this permission notice shall be included in all copies or substantial portions of the Software.

  THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
  FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
  LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER
  DEALINGS IN THE SOFTWARE.

   Based on Neil Kolban's example file: https://github.com/nkolban/ESP32_BLE_Arduino
 */

#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID(BLEUUID((uint16_t)0x181D));
// The characteristic of the remote service we are interested in.
static BLEUUID charUUID(BLEUUID((uint16_t)0x2A9D));

static BLEAddress *pServerAddress;
static BLEClient *pClient = NULL;
static boolean doConnect = false;
static boolean connected = false;
static BLERemoteCharacteristic *pRemoteCharacteristic;

const uint8_t notificationOff[] = {0x0, 0x0};
const uint8_t notificationOn[] = {0x1, 0x0};
bool onoff = true;

static void notifyCallback(
    BLERemoteCharacteristic *pBLERemoteCharacteristic,
    uint8_t *pData,
    size_t length,
    bool isNotify)
{

    Serial.print("Notify callback for characteristic ");
    for (int i = 0; i < length; i++)
    {
        Serial.print(pData[i]);
        Serial.print(" ");
    }
    Serial.println();
}

class MyClientCallback : public BLEClientCallbacks
{
    void onConnect(BLEClient *pclient)
    {
        connected = true;
        Serial.println("onConnect");
    }

    void onDisconnect(BLEClient *pclient)
    {
        connected = false;
        Serial.println("onDisconnect");
    }
};

bool createClient()
{
    pClient = BLEDevice::createClient();
    Serial.println(" - Created client");

    pClient->setClientCallbacks(new MyClientCallback());
}

bool connectToServer(BLEAddress pAddress)
{
    Serial.print("Forming a connection to ");
    Serial.println(pAddress.toString().c_str());

    // create a client, do this once per reboot OR once per disconnect?
    // if (pClient == NULL)
    createClient();

    // Connect to the remote BLE Server.
    pClient->connect(pAddress);
    Serial.println(" - Connected to server");

    // Obtain a reference to the service we are after in the remote BLE server.
    BLERemoteService *pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr)
    {
        Serial.print("Failed to find our service UUID: ");
        Serial.println(serviceUUID.toString().c_str());
        return false;
    }
    Serial.println(" - Found our service");

    // Obtain a reference to the characteristic in the service of the remote BLE server.
    pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
    if (pRemoteCharacteristic == nullptr)
    {
        Serial.print("Failed to find our characteristic UUID: ");
        Serial.println(charUUID.toString().c_str());
        return false;
    }
    Serial.println(" - Found our characteristic");

    // Read the value of the characteristic.
    std::string value = pRemoteCharacteristic->readValue();
    Serial.print("The characteristic value was: ");
    Serial.println(value.c_str());

    pRemoteCharacteristic->registerForNotify(notifyCallback);
}

/**
   Scan for BLE servers and find the first one that advertises the service we are looking for.
 */
class MyAdvertisedDeviceCallbacks : public BLEAdvertisedDeviceCallbacks
{
    /**
        Called for each advertising BLE server.
    */
    void onResult(BLEAdvertisedDevice advertisedDevice)
    {
        Serial.print("BLE Advertised Device found: ");
        Serial.println(advertisedDevice.toString().c_str());

        // We have found a device, let us now see if it contains the service we are looking for.
        if (advertisedDevice.haveServiceUUID() && advertisedDevice.getServiceUUID().equals(serviceUUID))
        {

            Serial.print("Found our device!");
            advertisedDevice.getScan()->stop();

            pServerAddress = new BLEAddress(advertisedDevice.getAddress());

            doConnect = true;
        }
    }
};

void setup()
{
    Serial.begin(115200);
    Serial.println("Starting Arduino BLE Client application...");
    BLEDevice::init("");

    BLEScan *pBLEScan = BLEDevice::getScan();
    pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
    pBLEScan->setActiveScan(true);
    pBLEScan->start(30);
}

void loop()
{
    if (!connected)
    {
        if (connectToServer(*pServerAddress))
        {
            Serial.println("We are now connected to the BLE Server.");
        }
        else
        {
            Serial.println("We have failed to connect to the server; there is nothin more we will do.");
        }
    }
    else
    {
        if (onoff)
        {
            Serial.println("Notifications turned on");
            pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t *)notificationOn, 2, true);
            onoff = false;
        }
    }

    delay(1000);
}
```

bilogic, 2019-04-19:
Fixed by overhauling parts of the BLE library. If anyone is interested, I will fork https://github.com/wakwak-koba/ESP32_BLE_Arduino to share the code. Thank you.

h2zero, 2019-04-19:
You shouldn't create a new client every time you reconnect, you should make pClient a global and call create client once in setup(). Also you may want to test if pClient->connect() returns true and set the connected flag after than call because occasionally the onConnect call back is called even when the connection fails as disconnect event gets sent before the open event that triggers the callback.

bilogic, 2019-04-20:
Yes, that was how I did it eventually. Thank you.

closed 2019-04-20 by bilogic

mblaise0, 2021-06-02:
@bilogic can  you please show your code work on how you handled this situation. I am having similar issues. I am running mqtt client (pubsubclient) and BLE library on an esp32 via wifi connection which is working effectively but once onDisconnect(BLEClient* pclient) routine is called I stop getting notification from the BLE server until i reset the board. your reply on how you solved this will be much appreciated. Thanks in anticipation.

mblaise0, 2021-06-02:
> You shouldn't create a new client every time you reconnect, you should make pClient a global and call createcClient() once in setup(). Also you may want to test if pClient->connect() returns true and set the connected flag after that call because occasionally the onConnect call back is called even when the connection fails as disconnect event gets sent before the open event that triggers the callback.

Can you please illustrate this with a sample code, I tried what you said here but I believe am doing it wrongly because my esp32 resets over and over which makes me believe am doing it wrongly. @bilogic

bilogic, 2021-06-02:
@mblaise0 I understand how it feels to be stuck with no solution. But it has really been a while, i'm not even sure what my current BLE uses. Let me get back.

mblaise0, 2021-06-02:
@bilogic You really get me,am right here waiting.  thank you so much.

h2zero, 2021-06-02:
@mblaise0 I forgot about a lot of the subtleties of this repo as I use NimBLE now. 

My comment here was simply to make the client pointer a global variable and call createClient once during initialization with the result stored in the global pointer. Then you could use the disconnect callback to set a flag or whatever and test that in a loop to reconnect when needed. I'm not sure if this still applies to this repo now as the code has since changed.

mblaise0, 2021-06-03:
@h2zero thanks for your reply, I actually did exactly what you said but I didn't get results so I imagined i did it wrongly. You mentioned using NimBLE, is that more efficient than the ESP32 Arduino BLE Library? I am curious to know if the issues am having with this can be solved using the NimBLE.

bilogic, 2021-06-03:
@mblaise0 eventually I discarded this and used https://github.com/shmuelzon/esp32-ble2mqtt

mblaise0, 2021-06-03:
@bilogic oh wow thanks alot but is there a away to compile this using the Arduino platform?

bilogic, 2021-06-03:
@mblaise0 unfortunately not, however I work on it inside VSCode, so, it makes little difference to me.

h2zero, 2021-06-03:
> @h2zero thanks for your reply, I actually did exactly what you said but I didn't get results so I imagined i did it wrongly. You mentioned using NimBLE, is that more efficient than the ESP32 Arduino BLE Library? I am curious to know if the issues am having with this can be solved using the NimBLE.

@mblaise0 yes to both questions. Check it out here
https://github.com/h2zero/NimBLE-Arduino

It's also very compatible with the repo, not much code change would be needed, probably just the header.

mblaise0, 2021-06-03:
@h2zero  oh wow! this is really brilliant let me jump to trying it out, i will give you my feedback soon enough. Its much understandable using the samples from the Refactored Original examples. Thank you so much.