nkolban/esp32-snippets #855: A client can't connect to 4 or more servers

happynet95, 2019-04-15:
Hi. I'm trying to connect four servers to a single client.
When the number of servers were less than four, this arduino code worked really well.
However, when the number of the server increases, the client still gets connected with server_A, server_B, and server_C while it fails to connect server_D.
When I turn off the server_A, I can connect server_B, server_C and server_D. So I dont think that I miscoded the server or client, and I think this library itself has an issue on connecting 4 or more servers at a time. This is based on "BLE_notify.ino" and "BLE_client.ino". Is there anyone else struggling with this problem?

here is my code : BLE_client.ino

/**
  A BLE client example that is rich in capabilities.
  There is a lot new capabilities implemented.
  author unknown
  updated by chegewara
*/

#include "BLEDevice.h"
// The remote service we wish to connect to.
static BLEUUID serviceUUID_A("4fafc201-1fb5-459e-8fcc-c5c9c331914a");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID_A("beb5483e-36e1-4688-b7f5-ea07361b26a8");

static boolean doConnect_A = false;
static boolean connected_A = false;
static boolean doScan_A = false;
static BLERemoteCharacteristic* pRemoteCharacteristic_A;
static BLEAdvertisedDevice* myDevice_A;

// The remote service we wish to connect to.
static BLEUUID serviceUUID_B("4fafc201-1fb5-459e-8fcc-c5c9c331914b");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID_B("beb5483e-36e1-4688-b7f5-ea07361b26b8");

static boolean doConnect_B = false;
static boolean connected_B = false;
static boolean doScan_B = false;
static BLERemoteCharacteristic* pRemoteCharacteristic_B;
static BLEAdvertisedDevice* myDevice_B;

// The remote service we wish to connect to.
static BLEUUID serviceUUID_C("4fafc201-1fb5-459e-8fcc-c5c9c331914c");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID_C("beb5483e-36e1-4688-b7f5-ea07361b26c8");

static boolean doConnect_C = false;
static boolean connected_C = false;
static boolean doScan_C = false;
static BLERemoteCharacteristic* pRemoteCharacteristic_C;
static BLEAdvertisedDevice* myDevice_C;

// The remote service we wish to connect to.
static BLEUUID serviceUUID_D("4fafc201-1fb5-459e-8fcc-c5c9c331914d");
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID_D("beb5483e-36e1-4688-b7f5-ea07361b26d8");

static boolean doConnect_D = false;
static boolean connected_D = false;
static boolean doScan_D = false;
static BLERemoteCharacteristic* pRemoteCharacteristic_D;
static BLEAdvertisedDevice* myDevice_D;

boolean A_connect = false;
boolean B_connect = false;
boolean C_connect = false;
boolean D_connect = false;

uint32_t synchro_count[4] = {0,0,0,0};

uint32_t PACKET_A[13];
uint32_t PACKET_B[13];
uint32_t PACKET_C[13];
uint32_t PACKET_D[13];
[... 569 lines cut ...]
  }

  
  
  if ((isfull_A == 1 && isfull_B >= 1 && isfull_C >= 1 && isfull_D >= 1)||(isfull_A >= 1 && isfull_B == 1 && isfull_C >= 1 && isfull_D >= 1)||(isfull_A >= 1 && isfull_B >= 1 && isfull_C == 1 && isfull_D >= 1)||(isfull_A >= 1 && isfull_B >= 1 && isfull_C >= 1 && isfull_D == 1)) {
    synchronize(synchro_count);
    Serial.println("packet counts are successfully synchronized!");
    isfull_A++;
    isfull_B++;
    isfull_C++;
    isfull_D++;
    Serial.print(" A : ");
    Serial.print(isfull_A);
    Serial.print(" B : ");
    Serial.print(isfull_B);
    Serial.print(" C : ");
    Serial.print(isfull_C);
    Serial.print(" D : ");
    Serial.print(isfull_D);

    }

  value++;
  // Delay a second between loops.
} // End of loop

h2zero, 2019-04-19:
If I'm not mistaken I think the arduino libraries were configured with a max of 3 connections. If you really need to connect to more than that simultaneously you may have to recompile the IDF with the menuconfig settings changed and copy the bt lib and sdkconfig file to your arduino folder.

djohns27, 2019-09-24:
I am having the same problem as @happynet95 .

@h2zero would you happen to know what files would need to be altered in the arduino libraries in order for it to work?

chegewara, 2019-09-27:
@djohns27 You have to use arduino as component in esp-idf environment.

GianCann, 2020-02-26:
@chegewara I'm sorry... can you explain better "You have to use arduino as component in esp-idf environment"?
I need to use an ESP32 as client for many server device (same type) using che [legoino library](https://github.com/corneliusmunz/legoino) (a wrapper for BLE connection/communication with these device).
I don't able to connect more than 3 device.
I read how to rebuild the ESP32-arduino libray, and how change the values in the sdkconfig file (I followed the instruction explained here: https://github.com/espressif/esp32-arduino-lib-builder ).
Now, I've a new set of file/library: how should I use them correctly?
I try to replace the file in the folder C:\Users\Gianluca\AppData\Local\Arduino15\packages\esp32 but without success.

Can you explain the correct way to enable and ESP32 to manage more of 3 BLE device from Arduino?
Thank you very much.

chegewara, 2020-02-26:
Asking about rebuilding arduino better place is arduino-esp32 repository, i never been doing this.
Instead i am using platformio with framework set to `arduino,espidf`. This allows to build app like in arduino ide, with all arduino libraries etc and to have sdkconfig with options you can setup as you need. Only issue is that you need to edit sdkconfig manually, make menuconfig is not available or i didnt find it.

In both cases you have to change 2 options in menuconfig that are responsible for limiting number of simultaneous connections, both in Bluetooth submenu (dont remember which exactly).

GianCann, 2020-02-26:
Ok, thank you @chegewara .
Can you give me some starting point (information/documentation) to configure platformio?
It's no problem to manuel edit the the sdkconfig file (it's only two parameters) but this is only the final step, of all.

Thank you for any help.

chegewara, 2020-02-26:
You can start with sdkconfig from arduino-esp32 repository.
If you encounter any problems i suggest to ask on gitter, there is few very wise and helpful people.