nkolban/esp32-snippets #856: Manually restarting server with deinit() and init()

MutatedGamer, 2019-04-21:
I am writing an app that needs multiple ESP32s to be both a BT server and client at the same time. After researching this and doing several experiments, I concluded the best way to do this would be to reinitialize my server after every each (very quick) BT exchance as a client. I have a function to start the server:

```void startServer() {
  // Make Bluetooth server
  BLEDevice::init(SERVER_NAME);
  // Server
  pServer = BLEDevice::createServer();
  pService = pServer->createService(SERVICE_UUID);
   //  Create characteristiccs for reading server kerb and uploading client kerb
  BLECharacteristic *pCharacteristicServer = pService->createCharacteristic(
                      SERVER_KERB_CHAR_UUID,
                      BLECharacteristic::PROPERTY_READ
                    );
  pCharacteristicServer -> setValue(MY_KERB);
  BLECharacteristic *pCharacteristicClient = pService->createCharacteristic(
                    CLIENT_KERB_CHAR_UUID,
                    BLECharacteristic::PROPERTY_READ |
                    BLECharacteristic::PROPERTY_WRITE
                  );
  pCharacteristicClient->setCallbacks(new ClientKerbWriteCallback());

  // Start the service
  pService->start();
  // Start advertising
  pAdvertising = pServer-> getAdvertising();
  pAdvertising->addServiceUUID(pService->getUUID());
  pAdvertising->start();
}```

I call this function in my setup() method, since the devices default to being a server.

When the user pushes a button, I call a method to scan for all servers advertising the same service it is providing

```void getDevices() {
  devices.clear();
  // Scan for servers to connect to
  BLEScan* pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  BLEScanResults foundDevices = pBLEScan->start(scanTime);
  delay(500);
}```

where `MyAdvertisedDeviceCallbacks()` adds each device with the matching service UUID to `devices`

Then, a user can call my BT exchange method for any of these devices, by being a client to the server briefly. I call a function which opens a connection to the server, does some information exchange, and then attempts to kill the BT connections and restart the server

```boolean doBluetoothExchange(BLEAdvertisedDevice device) {
  // Connect to server
  pClient  = BLEDevice::createClient();
  BLEAddress* pAddress = new BLEAddress(device.getAddress());
  pClient->connect(*pAddress);
  ... do stuff ...
  pClient -> disconnect();
  BLEDevice::deinit(false);
  startServer();
  return true;
}```

However, my code for some reason is hanging in `startServer()` on the lines

```pServer = BLEDevice::createServer();
  pService = pServer->createService(SERVICE_UUID);```

I thought that maybe I could get away with no re-initializing pServer and pService by putting them in a `(if pServer == nullptr)` block, but then what was happening was `startServer()` would finish after a BT exchange, but a new scan would immediately return and not find anything.

What is the correct way to do what I am trying to do here?

chegewara, 2019-04-22:
Why do you need to use init and deinit?

MutatedGamer, 2019-04-22:
@chegewara you're right... the  reason I decided I had to do this was because without it, after connecting as a client the device was undiscoverable from other ESP32s. Turns out I just had to do another `pServer->startAdvertising();` after disconnecting to start up the server again.

On a different note: I am sometimes getting the following error when I try to do `BLERemoteCharacteristic* pRemoteCharacteristic = pRemoteService->getCharacteristic(readServerKerbUUID);`:

```
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c", line 268, function: heap_caps_free
abort() was called at PC 0x4015286b on core 0

Backtrace: 0x4008dac4:0x3ffcfc70 0x4008dcf5:0x3ffcfc90 0x4015286b:0x3ffcfcb0 0x4008470b:0x3ffcfce0 0x4008a445:0x3ffcfd00 0x4000bec7:0x3ffcfd20 0x400d6c32:0x3ffcfd40 0x400d76a9:0x3ffcfd80 0x400d5691:0x3ffcfda0 0x400d6789:0x3ffcfe50 0x400ea2a5:0x3ffcfec0 0x400e6292:0x3ffcff10 0x40092739:0x3ffcff40
```

This only happens about half the time. Other times it works perfectly. The full debug log looks like:

```
[D][BLERemoteService.cpp:168] retrieveCharacteristics(): >> getCharacteristics() for service: 23014420-636a-11e9-a923-1681be663d3e
[D][BLERemoteService.cpp:199] retrieveCharacteristics(): Found a characteristic: Handle: 42, UUID: 230146f0-636a-11e9-a923-1681be663d3e
[D][BLERemoteCharacteristic.cpp:43] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 42 0x42, uuid: 230146f0-636a-11e9-a923-1681be663d3e
[D][BLERemoteCharacteristic.cpp:256] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 230146f0-636a-11e9-a923-1681be663d3e
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[D][BLERemoteCharacteristic.cpp:300] retrieveDescriptors(): << retrieveDescriptors(): Found 0 descriptors.
[D][BLERemoteCharacteristic.cpp:51] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:199] retrieveCharacteristics(): Found a characteristic: Handle: 44, UUID: e4b54230-e9cd-4725-be4e-2164f3e08a35
[D][BLERemoteCharacteristic.cpp:43] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 44 0x44, uuid: e4b54230-e9cd-4725-be4e-2164f3e08a35
[D][BLERemoteCharacteristic.cpp:256] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: e4b54230-e9cd-4725-be4e-2164f3e08a35
[E][BLERemoteCharacteristic.cpp:280] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[D][BLERemoteCharacteristic.cpp:300] retrieveDescriptors(): << retrieveDescriptors(): Found 0 descriptors.
[D][BLERemoteCharacteristic.cpp:51] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[D][BLERemoteService.cpp:215] retrieveCharacteristics(): << getCharacteristics()
[D][BLERemoteCharacteristic.cpp:404] readValue(): >> readValue(): uuid: 230146f0-636a-11e9-a923-1681be663d3e, handle: 42 0x2a
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: ReadCharEvt (0x3ffe25d0), owner: <N/A> for readValue
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: ReadCharEvt (0x3ffe25d0), owner: readValue
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 5] ... Unknown
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c", line 268, function: heap_caps_free
abort() was called at PC 0x4015286b on core 0

Backtrace: 0x4008dac4:0x3ffcfc70 0x4008dcf5:0x3ffcfc90 0x4015286b:0x3ffcfcb0 0x4008470b:0x3ffcfce0 0x4008a445:0x3ffcfd00 0x4000bec7:0x3ffcfd20 0x400d6c32:0x3ffcfd40 0x400d76a9:0x3ffcfd80 0x400d5691:0x3ffcfda0 0x400d6789:0x3ffcfe50 0x400ea2a5:0x3ffcfec0 0x400e6292:0x3ffcff10 0x40092739:0x3ffcff40

Rebooting...
```

I saw some other issue threads with the same problem, but none of them had an exact solution. Is there a fix for this yet (or a WIP one) that I can try to use? As you might guess this is for a class project and it wold be nice to not have crashes when I demo it to my mentor later this week.

AlexTZA, 2019-05-21:
I have the same issue with the hang occurring on the BLEDevice::createServer() call the second time it is called after BLEDevice::deinit().

My reason for using this is because the BLE side uses a large amount of RAM and I do not have enough to run both BLE and WifiSecureClient with certificates at the same time. Therefore, in my application, switching the less-frequently used BLE off by default and allowing the occasional switch to BLE on (while Wifi is turned off) seems to be most workable solution. Currently, I have a manual check to only allow this switch once otherwise a device reboot is required.

mauroscianca98, 2019-10-20:
@AlexTZA same situation here and I've still not found any good solution to deinit BLE and restart it in a second moment..