nkolban/esp32-snippets #862: HID Composite Device

rupin, 2019-05-03:
Hello,
I have used starting code from issue #632 and based on suggestions there, I have managed to achieve quite a bit in my application. I searched through the issues list(both open and closed) but I could not find this or a similar issue reported. 

What I am trying to get to is to build a ESP Composite Device ( over GATT), which consists of Two Independent Joysticks at least.  These are to support two players. 

Based on https://eleccelerator.com/tutorial-about-usb-hid-report-descriptors/, it is possible to create two independent joysticks using two reports.It is also possible that the report map may have a mouse and a keyboard and it will show up additional devices. But the site description is generic, and mostly talks about a USB cable based wired connection versus a BLE HID connection. 

I have my code for perusal here https://github.com/rupin/ESPJoystick/blob/master/gamepadble632/gamepadble632.ino

You will notice I have two reports in my code 

````
const uint8_t reportMap[] = {
  0x05, 0x01,  // USAGE_PAGE (Generic Desktop)
  0x09, 0x04,  // USAGE (Game Pad)
  0xa1, 0x01,  // COLLECTION (Application)

  0x85, 0x01,  //     REPORT_ID (1)
  0x05, 0x09,  //     USAGE_PAGE (Button)
  0x19, 0x01,  //     USAGE_MINIMUM (Button 1)
  0x29, 0x10,  //     USAGE_MAXIMUM (Button 16)
  0x15, 0x00,  //     LOGICAL_MINIMUM (0)
  0x25, 0x01,  //     LOGICAL_MAXIMUM (1)
  0x95, 0x08,  //     REPORT_COUNT (8)
  0x75, 0x02,  //     REPORT_SIZE (2)
  0x81, 0x02,  //     INPUT (Data,Var,Abs)
  //0xa1, 0x00,  //     COLLECTION (Physical)
  0x05, 0x01,  //       USAGE_PAGE (Generic Desktop)
  0x09, 0x30,  //       USAGE (X)
  0x09, 0x31,  //       USAGE (Y)
  0x15, 0x00,  //       LOGICAL_MINIMUM (0)
  0x25, 0xFF,  //       LOGICAL_MAXIMUM (100)
  0x75, 0x08,  //       REPORT_SIZE (8)
  0x95, 0x02,  //       REPORT_COUNT (2)
  0x81, 0x02,  //       INPUT (Data,Var,Abs)
  // 0xc0,        //     END_COLLECTION

  0xc0,         // END_COLLECTION

  0x05, 0x01,  // USAGE_PAGE (Generic Desktop)
  0x09, 0x04,  // USAGE (Game Pad)
  0xa1, 0x01,  // COLLECTION (Application)

  0x85, 0x02,  //     REPORT_ID (2)
  0x05, 0x09,  //     USAGE_PAGE (Button)
  0x19, 0x01,  //     USAGE_MINIMUM (Button 1)
  0x29, 0x10,  //     USAGE_MAXIMUM (Button 16)
  0x15, 0x00,  //     LOGICAL_MINIMUM (0)
  0x25, 0x01,  //     LOGICAL_MAXIMUM (1)
  0x95, 0x08,  //     REPORT_COUNT (8)
  0x75, 0x02,  //     REPORT_SIZE (2)
  0x81, 0x02,  //     INPUT (Data,Var,Abs)
  //0xa1, 0x00,  //     COLLECTION (Physical)
  0x05, 0x01,  //       USAGE_PAGE (Generic Desktop)
  0x09, 0x33,  //       USAGE (X)
  0x09, 0x34,  //       USAGE (Y)
  0x15, 0x00,  //       LOGICAL_MINIMUM (0)
  0x25, 0xFF,  //       LOGICAL_MAXIMUM (100)
  0x75, 0x08,  //       REPORT_SIZE (8)
  0x95, 0x02,  //       REPORT_COUNT (2)
  0x81, 0x02,  //       INPUT (Data,Var,Abs)
  // 0xc0,        //     END_COLLECTION

  0xc0         // END_COLLECTION

};
````

Now, the reportMap has two reports numbered 1 and 2. 

I expected two joysticks to be visible on Android, but I see one only. Button presses and axis changes are all mapped to one single joystick. 

What am I doing wrong?

chegewara, 2019-05-03:
Try removing second `USAGE_PAGE` or even all 3:
```
  0x05, 0x01,  // USAGE_PAGE (Generic Desktop)
  0x09, 0x04,  // USAGE (Game Pad)
  0xa1, 0x01,  // COLLECTION (Application)
```

rupin, 2019-05-03:
I made the change. I till see one Gamepad only unfortunately.  This is a code running on a local server which lists all gamepads connected to the computer on which the webpage has been opened. 
[image]

This code uses the GamepadAPI from here https://developer.mozilla.org/en-US/docs/Web/API/Gamepad_API/Using_the_Gamepad_API

chegewara, 2019-05-09:
Ok, i found some time to make some tests. It took me about an hour or less. Here is working report map:
```
    const uint8_t report[] = {
0x05, 0x01,  // USAGE_PAGE (Generic Desktop)
  0x09, 0x04,  // USAGE (Game Pad)
  0xa1, 0x01,  // COLLECTION (Application)

  0x85, 0x01,  //     REPORT_ID (1)
  0x05, 0x09,  //     USAGE_PAGE (Button)
  0x19, 0x01,  //     USAGE_MINIMUM (Button 1)
  0x29, 0x10,  //     USAGE_MAXIMUM (Button 16)
  0x15, 0x00,  //     LOGICAL_MINIMUM (0)
  0x25, 0x01,  //     LOGICAL_MAXIMUM (1)
  0x95, 0x10,  //     REPORT_COUNT (16)
  0x75, 0x01,  //     REPORT_SIZE (1)
  0x81, 0x02,  //     INPUT (Data,Var,Abs)
  0xa1, 0x00,  //     COLLECTION (Physical)
  0x05, 0x01,  //       USAGE_PAGE (Generic Desktop)
  0x09, 0x30,  //       USAGE (X)
  0x09, 0x31,  //       USAGE (Y)
  0x15, 0x00,  //       LOGICAL_MINIMUM (0)
  0x25, 0xFF,  //       LOGICAL_MAXIMUM (100)
  0x75, 0x08,  //       REPORT_SIZE (8)
  0x95, 0x02,  //       REPORT_COUNT (2)
  0x81, 0x02,  //       INPUT (Data,Var,Abs)
   0xc0,        //     END_COLLECTION
   0xc0,        //     END_COLLECTION

  0x09, 0x04,  // USAGE (Game Pad)
  0xa1, 0x01,  // COLLECTION (Application)
  0x85, 0x02,  //     REPORT_ID (1)
  0x05, 0x09,  //     USAGE_PAGE (Button)
  0x19, 0x01,  //     USAGE_MINIMUM (Button 1)
  0x29, 0x10,  //     USAGE_MAXIMUM (Button 16)
  0x15, 0x00,  //     LOGICAL_MINIMUM (0)
  0x25, 0x01,  //     LOGICAL_MAXIMUM (1)
  0x95, 0x10,  //     REPORT_COUNT (16)
  0x75, 0x01,  //     REPORT_SIZE (1)
  0x81, 0x02,  //     INPUT (Data,Var,Abs)
  0xa1, 0x00,  //     COLLECTION (Physical)
  0x05, 0x01,  //       USAGE_PAGE (Generic Desktop)
  0x09, 0x30,  //       USAGE (X)
  0x09, 0x31,  //       USAGE (Y)
  0x15, 0x00,  //       LOGICAL_MINIMUM (0)
  0x25, 0xFF,  //       LOGICAL_MAXIMUM (100)
  0x75, 0x08,  //       REPORT_SIZE (8)
  0x95, 0x02,  //       REPORT_COUNT (2)
  0x81, 0x02,  //       INPUT (Data,Var,Abs)
   0xc0,        //     END_COLLECTION

  0xc0         // END_COLLECTION      
    };
```

rupin, 2019-05-09:
Will try this tommorow and report on how it works.

chegewara, 2019-05-27:
@rupin Any news on this?

rupin, 2019-05-27:
Sorry to have not reported. I tried this and it still shows up as one joystick. I managed to changed my JavaScript code and dropped changing any code on the Arduino's side, which worked.

closed 2019-05-27 by rupin

chegewara, 2019-05-27:
I dont know why, because it is working with my windows 10. I can recognize 2 joysticks with 16 buttons each.

rupin, 2019-05-27:
My win 10 machine doesn't have bluetooth, I tried this on an Android phone. Don't think the behavior would change. I used the html5 gamepad API tester for this, which runs on Chrome.

chegewara, 2019-05-27:
Like i said, its working with my windows 10 laptop. I would not post it if it wouldnt work.
Its very strange, because as far as i know android can accept broken report maps and parse it in case where windows 10 will just discard it and report HID driver error.

rupin, 2019-05-29:
Hey @chegewara can I get a copy of your code?

chegewara, 2019-05-29:
Hi @rupin, here is code i tested one more time(its not optimized):
```
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include "BLE2902.h"
#include "BLEHIDDevice.h"
#include "HIDTypes.h"
#include "HIDKeyboardTypes.h"

BLEHIDDevice* hid;
BLECharacteristic* input;

bool connected = false;

class MyCallbacks : public BLEServerCallbacks {
  void onConnect(BLEServer* pServer){
    connected = true;
    BLE2902* desc = (BLE2902*)input->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
    desc->setNotifications(true);
  }

  void onDisconnect(BLEServer* pServer){
    connected = false;
    BLE2902* desc = (BLE2902*)input->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
    desc->setNotifications(false);
  }
};

void taskServer(void*){

    BLEDevice::init("ESP32-gamepad");
    BLEServer *pServer = BLEDevice::createServer();
    pServer->setCallbacks(new MyCallbacks());

    hid = new BLEHIDDevice(pServer);
    input = hid->inputReport(1); // <-- input REPORTID from report map

    std::string name = "chegewara";
    hid->manufacturer()->setValue(name);

    hid->pnp(0x02, 0xe502, 0xa111, 0x0210);
    hid->hidInfo(0x00,0x02);

  BLESecurity *pSecurity = new BLESecurity();
//  pSecurity->setKeySize();
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);

const uint8_t report[] = {
0x05, 0x01,  // USAGE_PAGE (Generic Desktop)
  0x09, 0x04,  // USAGE (Game Pad)
  0xa1, 0x01,  // COLLECTION (Application)

  0x85, 0x01,  //     REPORT_ID (1)
  0x05, 0x09,  //     USAGE_PAGE (Button)
  0x19, 0x01,  //     USAGE_MINIMUM (Button 1)
  0x29, 0x10,  //     USAGE_MAXIMUM (Button 16)
  0x15, 0x00,  //     LOGICAL_MINIMUM (0)
  0x25, 0x01,  //     LOGICAL_MAXIMUM (1)
  0x95, 0x10,  //     REPORT_COUNT (16)
  0x75, 0x01,  //     REPORT_SIZE (1)
  0x81, 0x02,  //     INPUT (Data,Var,Abs)
  0xa1, 0x00,  //     COLLECTION (Physical)
  0x05, 0x01,  //       USAGE_PAGE (Generic Desktop)
  0x09, 0x30,  //       USAGE (X)
  0x09, 0x31,  //       USAGE (Y)
  0x15, 0x00,  //       LOGICAL_MINIMUM (0)
  0x25, 0xFF,  //       LOGICAL_MAXIMUM (100)
  0x75, 0x08,  //       REPORT_SIZE (8)
  0x95, 0x02,  //       REPORT_COUNT (2)
  0x81, 0x02,  //       INPUT (Data,Var,Abs)
   0xc0,        //     END_COLLECTION
   0xc0,        //     END_COLLECTION

  0x09, 0x04,  // USAGE (Game Pad)
  0xa1, 0x01,  // COLLECTION (Application)
  0x85, 0x02,  //     REPORT_ID (2)
  0x05, 0x09,  //     USAGE_PAGE (Button)
  0x19, 0x01,  //     USAGE_MINIMUM (Button 1)
  0x29, 0x10,  //     USAGE_MAXIMUM (Button 16)
  0x15, 0x00,  //     LOGICAL_MINIMUM (0)
  0x25, 0x01,  //     LOGICAL_MAXIMUM (1)
  0x95, 0x10,  //     REPORT_COUNT (16)
  0x75, 0x01,  //     REPORT_SIZE (1)
  0x81, 0x02,  //     INPUT (Data,Var,Abs)
  0xa1, 0x00,  //     COLLECTION (Physical)
  0x05, 0x01,  //       USAGE_PAGE (Generic Desktop)
  0x09, 0x30,  //       USAGE (X)
  0x09, 0x31,  //       USAGE (Y)
  0x15, 0x00,  //       LOGICAL_MINIMUM (0)
  0x25, 0xFF,  //       LOGICAL_MAXIMUM (100)
  0x75, 0x08,  //       REPORT_SIZE (8)
  0x95, 0x02,  //       REPORT_COUNT (2)
  0x81, 0x02,  //       INPUT (Data,Var,Abs)
   0xc0,        //     END_COLLECTION

  0xc0         // END_COLLECTION      
    };

    hid->reportMap((uint8_t*)report, sizeof(report));
    hid->startServices();

    BLEAdvertising *pAdvertising = pServer->getAdvertising();
    pAdvertising->setAppearance(HID_GAMEPAD);
    pAdvertising->addServiceUUID(hid->hidService()->getUUID());
    pAdvertising->start();
    hid->setBatteryLevel(7);

    ESP_LOGD(LOG_TAG, "Advertising started!");
    delay(portMAX_DELAY);
  
};

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  xTaskCreate(taskServer, "server", 20000, NULL, 5, NULL);
}

void loop() {
  
    vTaskDelay(5000);
}
```
[image]

chegewara, 2019-05-29:
And here with some values populated and sent as example:
```
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include "BLE2902.h"
#include "BLEHIDDevice.h"
#include "HIDTypes.h"
#include "HIDKeyboardTypes.h"

BLEHIDDevice* hid;
BLECharacteristic* input;
BLECharacteristic* input1;

bool connected = false;

class MyCallbacks : public BLEServerCallbacks {
  void onConnect(BLEServer* pServer){
    connected = true;
    BLE2902* desc = (BLE2902*)input->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
    desc->setNotifications(true);
    desc = (BLE2902*)input1->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
    desc->setNotifications(true);
  }

  void onDisconnect(BLEServer* pServer){
    connected = false;
    BLE2902* desc = (BLE2902*)input->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
    desc->setNotifications(false);
  }
};

void taskServer(void*){

    BLEDevice::init("ESP32-gamepad");
    BLEServer *pServer = BLEDevice::createServer();
    pServer->setCallbacks(new MyCallbacks());

    hid = new BLEHIDDevice(pServer);
    input = hid->inputReport(1); // <-- input REPORTID from report map
    input1 = hid->inputReport(2); // <-- input REPORTID from report map

    std::string name = "chegewara";
    hid->manufacturer()->setValue(name);

    hid->pnp(0x02, 0xe502, 0xa111, 0x0210);
    hid->hidInfo(0x00,0x02);

  BLESecurity *pSecurity = new BLESecurity();
//  pSecurity->setKeySize();
  pSecurity->setAuthenticationMode(ESP_LE_AUTH_BOND);

const uint8_t report[] = {
0x05, 0x01,  // USAGE_PAGE (Generic Desktop)
  0x09, 0x04,  // USAGE (Game Pad)
  0xa1, 0x01,  // COLLECTION (Application)

  0x85, 0x01,  //     REPORT_ID (1)
  0x05, 0x09,  //     USAGE_PAGE (Button)
  0x19, 0x01,  //     USAGE_MINIMUM (Button 1)
  0x29, 0x10,  //     USAGE_MAXIMUM (Button 16)
  0x15, 0x00,  //     LOGICAL_MINIMUM (0)
  0x25, 0x01,  //     LOGICAL_MAXIMUM (1)
  0x95, 0x10,  //     REPORT_COUNT (16)
  0x75, 0x01,  //     REPORT_SIZE (1)
  0x81, 0x02,  //     INPUT (Data,Var,Abs)
//  0xa1, 0x00,  //     COLLECTION (Physical)
//  0x05, 0x01,  //       USAGE_PAGE (Generic Desktop)
  0x09, 0x30,  //       USAGE (X)
  0x09, 0x31,  //       USAGE (Y)
  0x15, 0x00,  //       LOGICAL_MINIMUM (0)
  0x25, 0xFF,  //       LOGICAL_MAXIMUM (100)
  0x75, 0x08,  //       REPORT_SIZE (8)
  0x95, 0x02,  //       REPORT_COUNT (2)
  0x81, 0x02,  //       INPUT (Data,Var,Abs)
//   0xc0,        //     END_COLLECTION
   0xc0,        //     END_COLLECTION

  0x09, 0x04,  // USAGE (Game Pad)
  0xa1, 0x01,  // COLLECTION (Application)
  0x85, 0x02,  //     REPORT_ID (2)
  0x05, 0x09,  //     USAGE_PAGE (Button)
  0x19, 0x01,  //     USAGE_MINIMUM (Button 1)
  0x29, 0x10,  //     USAGE_MAXIMUM (Button 16)
  0x15, 0x00,  //     LOGICAL_MINIMUM (0)
  0x25, 0x01,  //     LOGICAL_MAXIMUM (1)
  0x95, 0x10,  //     REPORT_COUNT (16)
  0x75, 0x01,  //     REPORT_SIZE (1)
  0x81, 0x02,  //     INPUT (Data,Var,Abs)
//  0xa1, 0x00,  //     COLLECTION (Physical)
//  0x05, 0x01,  //       USAGE_PAGE (Generic Desktop)
  0x09, 0x30,  //       USAGE (X)
  0x09, 0x31,  //       USAGE (Y)
  0x15, 0x00,  //       LOGICAL_MINIMUM (0)
  0x25, 0xFF,  //       LOGICAL_MAXIMUM (100)
  0x75, 0x08,  //       REPORT_SIZE (8)
  0x95, 0x02,  //       REPORT_COUNT (2)
  0x81, 0x02,  //       INPUT (Data,Var,Abs)
//   0xc0,        //     END_COLLECTION

  0xc0         // END_COLLECTION      
    };

    hid->reportMap((uint8_t*)report, sizeof(report));
    hid->startServices();

    BLEAdvertising *pAdvertising = pServer->getAdvertising();
    pAdvertising->setAppearance(HID_GAMEPAD);
    pAdvertising->addServiceUUID(hid->hidService()->getUUID());
    pAdvertising->start();
    hid->setBatteryLevel(7);

    ESP_LOGD(LOG_TAG, "Advertising started!");
    delay(portMAX_DELAY);
  
};

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  xTaskCreate(taskServer, "server", 20000, NULL, 5, NULL);
}
uint32_t msg = 0x01010101;
uint32_t msg1;
void loop() {
  if(connected)
  {
    msg1 = esp_random();
    uint32_t msg0 = 0x0;
    input->setValue((uint8_t*)&msg,4);
    input->notify();
    input1->setValue((uint8_t*)&msg1,4);
    input1->notify();
    vTaskDelay(1000);
    input->setValue((uint8_t*)&msg0,4);
    input->notify();
    input1->setValue((uint8_t*)&msg0,4);
    input1->notify();
  }
    vTaskDelay(1000);
}
```

rupin, 2019-06-04:
Using your code, on Android, I see a Gamepad with 16 buttons and a keyboard. 

I have ordered a Bluetooth dongle which should arrive in a couple of days.