nkolban/esp32-snippets #863: Stack Smashing protect Failure - getCharacteristics();

Bisouke, 2019-05-03:
Hello everybody,

I'm actually using Neil Kolban's library ESP32_BLE_Arduino, but i have a little problem...
In my program i would to :

1. Scan BLE servers(**done**)
2. Connect to a remote BLE Server (**done**)
3. Once connected, find all services available on the server (**done**)
4. Find all characteristisc of all services (**PROBLEM**)
5. Read and write characteristics values (**not done yet**)

Once i'm connected to the server and recovered service map i try to recover all the characteristics map with the BLERemoteService::getCharacteristics(); but i have a **STACK SMASHING PROTECT FAILURE**

Someone has an idea why this function works with some of my BLE server services but not with all ?

There are some pics to make it more understandable...

**Picture 1 : code extract in the part i'm having trouble**
(this part of code is just after i get successfully connected on the server)
[image]

**Picture 2 : Serial Monitor with "stack smashing protect failure"** 
[image]

**Picture 3 : My BLE server when i'm connected on it with nRF Connect**
[image]

I would appreciate any hints or helps :)

Have a nice day,
Brice ANNETTE

chegewara, 2019-05-03:
Stack smashing error is when your task stack is too small. You have 2 options:
- increase current task stack (im guessing it is looprTask),
- create new task and use it for BLE

Bisouke, 2019-05-06:
Hello chegewara,
Thank you for your quick answer.

> increase current task stack (im guessing it is looprTask),

How can i increase the task stack ? Have it to be done with ESP-IDF?

Have a nice day,
Best regards

h2zero, 2019-05-06:
You can call ‘Serial.println(uxTaskGetStackHighWaterMark(NULL)) ‘ in your function to check how much stack space you have left. If it reports below 500 or so you’ll either have to put your function in a task with ‘xTaskCreate()’ and set it’s stack size higher, or find the main.cpp file and increase the task stack size there.

Bisouke, 2019-05-07:
Hello h2zero,
Thank you very much for your answer, i tried what you said with the `uxTaskGetStackHighWaterMark(NULL); `function but my stack size is higher than 500.

As you can see on this picture :

[image]

I tried several times and it's always much higher than 500 (it's around 6000);
Do you have other ideas ?

I'll try to use` xTaskCreate()` to be sure and i'll report you soon.

Have a nice day.

Bisouke, 2019-05-07:
So I tried what you said : **I created a new task for the function** that I had trouble with and it still deosn't work and I have to admit i'm a bit distraught.

this is the code that I used to create my new task : `xTaskCreate(getServicesAndCharacteristics, "MyFunction", 100000,NULL,1,NULL);` where getServicesAndCharacteristics is my function.

I tried with several others stacks depth values and it still doesn't work.

Have you other ideas of where it should come from ?

Thank you in advance.
Have a nice day,
Brice ANNETTE

h2zero, 2019-05-07:
@Bisouke

That’s actually what I expected, unless you had a huge sketch doing other things. The main task stack is 8192 so it’s big enough for most things. 

I’ll try to do some testing as I haven’t used the function you are before.

h2zero, 2019-05-07:
Just tested your code snippet and copied the service uuid you posted as created 12 other services and it seems to work fine. Not sure what else is happening in your code that could cause the issue, but the library seems to be fine.

Bisouke, 2019-05-13:
Ok thank you for this information h2zero.

### **I'm asking myself if there are some kind of data protection in the BLE protocol ?**

Because the device i'm interacting with (the server) is a **T**yre **P**ressure **M**oniroting **S**ystem **TPMS** for short. A device that is inside the tyre of recent cars.

The four services **Generic Access**, **Generic Attribute**, **Battery Service** and **Device Information** seems to behave correctly but not the 3 others in particular it is the characteristics of these services that make my program fail (when i do try to get the map of characteristics) 
**And it is those characteristics that contain the "sensitive" data** such as pressure, temperature, acceleration, state of the device etc...
This what bring me to this reflexion...

I'll try to interact only with the characteristics of the services Generic Access, Generic Attribute, Battery Service and Device Information to be sure.

chegewara, 2019-05-13:
Hi,
probably its because you are using arduino library with bug (maybe more than 1), in this library its already fixed:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLERemoteService.cpp#L129

Bisouke, 2019-05-13:
Ok thank you Chegewara i'll try it !

h2zero, 2019-05-13:
Good catch @Chegewara, the arduino library would do well to keep up with this repo, unfortunately that might break some folks currently running projects.

Bisouke, 2019-05-13:
Thanks to you two !

If i understood well i have to replace the files of the src folder in my computer at :
 
_C:\Users\MyName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\BLE_

by all the files *.cpp and *.h at _https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils_ ? 

I'm not very comfortable with Arduino yet so i prefer to ask you :-(

h2zero, 2019-05-13:
Yes, just download this repo and copy/ paste the files from cpp_utils folder into the the libraries/BLE folder and you’ll be using this library instead. You might have to make some adjustments to your code to accommodate this library, such as using an advertised device pointer in the connect() call instead of address. Just check the examples.

Bisouke, 2019-05-15:
> Hi,
> probably its because you are using arduino library with bug (maybe more than 1), in this library its already fixed:

Well it seems like you were right !

To resolve my problem i did what you said : **i updated the library** at :

_C:\Users\MyName\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2\libraries\BLE_

By the library at :

_https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils_

**I only replaced the files that were on both library**, i didn't add the other files present in cpp_utils because it made my code uncompilable.

And now, it works perfectly !!!
Thank you very much chegewara and h2zero for your time :)

Have a nice day.

closed 2019-05-15 by Bisouke

tromeli, 2019-05-22:
> Hi,
> probably its because you are using arduino library with bug (maybe more than 1), in this library its already fixed:
> https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLERemoteService.cpp#L129

Hi, I have a question about ‘esp32-snippets’ repository. I found some bugs in the arduino-esp32  that have been fixed in this repository. Why does not pull to arduino-esp32.