nkolban/esp32-snippets #864: Compilation error's when trying out cpp-utils

AEnimaH, 2019-05-06:
Hi!

I took the hello-world example and enabled C++ exceptions and ble via the menuconfig.
I deleted the c file and added the skeleton file helloworld.cpp from this repo.
Next I copied the files from cpp-utils to /components/cpp-utils.

When I do a make command i got the following error:
```
project_path/components/cpp_utils/HttpRequest.cpp:42:26: fatal error: hwcrypto/sha.h: No such file or directory
```
Solved this by changing 
```
#include <hwcrypto/sha.h>
```
to
```
#include <esp32/sha.h>
```
In the HttpRequest.cpp file under components/cpp-utils

But I get additional errors i can't seem to fix:
```
project_path/components/cpp_utils/BLEDevice.cpp: In static member function 'static void BLEDevice::whiteListAdd(BLEAddress)':
project_path/components/cpp_utils/BLEDevice.cpp:488:75: error: too few arguments to function 'esp_err_t esp_ble_gap_update_whitelist(bool, uint8_t*, esp_ble_wl_addr_type_t)'
  esp_err_t errRc = esp_ble_gap_update_whitelist(true, *address.getNative());  // True to add an entry.
                                                                           ^
In file included from project_path/components/cpp_utils/BLEDevice.cpp:17:0:
/home/parallels/esp/esp-idf/components/bt/bluedroid/api/include/api/esp_gap_ble_api.h:947:11: note: declared here
 esp_err_t esp_ble_gap_update_whitelist(bool add_remove, esp_bd_addr_t remote_bda, esp_ble_wl_addr_type_t wl_addr_type);
           ^
project_path/components/cpp_utils/BLEDevice.cpp: In static member function 'static void BLEDevice::whiteListRemove(BLEAddress)':
project_path/components/cpp_utils/BLEDevice.cpp:502:76: error: too few arguments to function 'esp_err_t esp_ble_gap_update_whitelist(bool, uint8_t*, esp_ble_wl_addr_type_t)'
  esp_err_t errRc = esp_ble_gap_update_whitelist(false, *address.getNative());  // False to remove an entry.
                                                                            ^
In file included from project_path/components/cpp_utils/BLEDevice.cpp:17:0:
/home/parallels/esp/esp-idf/components/bt/bluedroid/api/include/api/esp_gap_ble_api.h:947:11: note: declared here
 esp_err_t esp_ble_gap_update_whitelist(bool add_remove, esp_bd_addr_t remote_bda, esp_ble_wl_addr_type_t wl_addr_type);
           ^
```
Is this a known issue? Am I doing something wrong

h2zero, 2019-05-06:
New IDF changed the api call for whitelist functions. Check my comment [here](https://github.com/espressif/arduino-esp32/issues/2723#issuecomment-489107140) for a quick and dirty fix.

AEnimaH, 2019-05-07:
Thanks that solved it!

closed 2019-05-07 by AEnimaH