nkolban/esp32-snippets #866: BLE disconnect not working for ESP32

bihmedical, 2019-05-07:
Hi ,

I am trying to connect ESP32 as client to server. I have to connect multiple BLE ESP32 devices at the same time.
I have mentioned the procedure I am following below. But is there any other way to do this? I am just looking for a constant packet from the BLE devices.
The way I was working around was connect one client with server and the server would disconnect or time out.  For which I am using 
[https://www.esp32.com/viewtopic.php?t=3923](url)  . The code seem to have error as below

Any inputs would be very helpful!!

`
/*
 * 
 * This sketch emulates parts of a Polar H7 Heart Rate Sensor. 
 * It exposes the Heart rate and the Sensor position characteristics
 
   Copyright <2017> <Andreas Spiess>
  Permission is hereby granted, free of charge, to any person obtaining a copy of this software and associated documentation files (the "Software"),
  to deal in the Software without restriction, including without limitation the rights to use, copy, modify, merge, publish, distribute, sublicense,
  and/or sell copies of the Software, and to permit persons to whom the Software is furnished to do so, subject to the following conditions:
  The above copyright notice and this permission notice shall be included in all copies or substantial portions of the Software.
  THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
  FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
  LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM, OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER
  DEALINGS IN THE SOFTWARE.
   
   Based on Neil Kolban's example file: https://github.com/nkolban/ESP32_BLE_Arduino
 */

#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include <BLE2902.h>

uint32_t beginTimer=0;

byte flags = 0b00111110;
byte bpm;
byte heart[8] = { 0b00000001, 60, 0, 0, 0 , 0, 0, 0};         //00001110
byte hrmPos[1] = {2};

bool _BLEClientConnected = false;

#define heartRateService BLEUUID((uint16_t)0x180D)
BLECharacteristic heartRateMeasurementCharacteristics(BLEUUID((uint16_t)0x2A37), BLECharacteristic::PROPERTY_NOTIFY);
BLECharacteristic sensorPositionCharacteristic(BLEUUID((uint16_t)0x2A38), BLECharacteristic::PROPERTY_READ);
BLEDescriptor heartRateDescriptor(BLEUUID((uint16_t)0x2901));
BLEDescriptor sensorPositionDescriptor(BLEUUID((uint16_t)0x2901));

class MyServerCallbacks : public BLEServerCallbacks {
    void onConnect(BLEServer* pServer) {
      _BLEClientConnected = true;
    };

    void onDisconnect(BLEServer* pServer) {
      _BLEClientConnected = false;
    }
};

void InitBLE() {
  BLEDevice::init("ESP2");
  // Create the BLE Server
  BLEServer *pServer = BLEDevice::createServer();
  pServer->setCallbacks(new MyServerCallbacks());

  // Create the BLE Service
  BLEService *pHeart = pServer->createService(heartRateService);

  pHeart->addCharacteristic(&heartRateMeasurementCharacteristics);
  heartRateDescriptor.setValue("Rate from 0 to 200");
  heartRateMeasurementCharacteristics.addDescriptor(&heartRateDescriptor);
  heartRateMeasurementCharacteristics.addDescriptor(new BLE2902());

  pHeart->addCharacteristic(&sensorPositionCharacteristic);
  sensorPositionDescriptor.setValue("Position 0 - 6");
  sensorPositionCharacteristic.addDescriptor(&sensorPositionDescriptor);

  pServer->getAdvertising()->addServiceUUID(heartRateService);

  pHeart->start();
  // Start advertising
  pServer->getAdvertising()->start();
}

void setup() {
  Serial.begin(115200);
  Serial.println("Start");
  InitBLE();
  bpm = 1;
}

void loop() {
  if(_BLEClientConnected == true && beginTimer == 0){
    beginTimer = millis();
    Serial.println("Client connected timer started");
  }
  else if(_BLEClientConnected == true && beginTimer != 0 && (millis() - beginTimer) > 10000){
    Serial.println("Client connected for too long");
    _BLEClientConnected = false;
    pServer->disconnectClient() ;
  }
  else if(_BLEClientConnected == false && beginTimer != 0){
    beginTimer = 0;
    Serial.println("Client disconnected timer reset");
  }
  heart[1] = (byte)bpm;
  int energyUsed = 3000;
  
  heart[2] = 61;       //energyUsed - (heart[2] * 256);
  heart[3] = 57;      //energyUsed / 256;
  Serial.println(heart[1]);

  heartRateMeasurementCharacteristics.setValue(heart, 8);
  heartRateMeasurementCharacteristics.notify();

  sensorPositionCharacteristic.setValue(hrmPos, 1);
  bpm++;

  delay(5000);
}`

P:\BLE_notify_polar\BLE_notify_polar.ino: In function 'void loop()':

BLE_notify_polar:92:5: error: 'pServer' was not declared in this scope

     pServer->disconnectClient() ;

     ^

exit status 1
'pServer' was not declared in this scope

h2zero, 2019-05-08:
You need to define pServer globally, not inside a function call.

bihmedical, 2019-05-08:
Hi h2zero,

I had tried defining "BLEServer *pServer = BLEDevice::createServer();" globally after #include statements, but my esp32 was just restarting. The programming did not seem to execute void setup() as well. I might be missing something and not able to figure it out. 

[image]

h2zero, 2019-05-09:
This isn't a library issue but I'll answer anyway. You need to pServer globally, then initialize it in the setup code this way.

Put `BLEServer *pServer;` after the includes and change the line:
`BLEServer *pServer = BLEDevice::createServer();` to:
`pServer = BLEDevice::createServer();`

closed 2019-05-22 by bihmedical