nkolban/esp32-snippets #868: Esp32 Uart

sephyrah, 2019-05-10:
Hello, 

I'm trying to make connection with ESP32 and a "black box" that i only know the Baudrate-9600 and that if we connect this box with USB-RS485 to putty it shows a Menu.

So im trying to connect the ESP 32 with RS-485 but it only sends 1 byte from my array, i have this:

void app_main()
{
	init();

	static const char *TX_TASK_TAG = "TX_TASK";
	static const char *RX_TASK_TAG = "RX_TAK";
	const char *Data[10] = {0x65, 0x66, 0x61, 0x63, 0x65, 0x63, 0x30, 0x30, 0x31};
	esp_log_level_set(TX_TASK_TAG, ESP_LOG_INFO);
	esp_log_level_set(RX_TASK_TAG, ESP_LOG_INFO);

	sendData(TX_TASK_TAG, Data);
	rx_task();
}

but in the terminal it says:
TX_TASK: Wrote 1 bytes

Shouldn't it write like 8/9 bytes ? How do I know the box is receiving it?
Could it be the uart configurations?

zz-zsys, 2019-05-10:
What is the argument for the sendData function? I will assume that the declaration of the sendData function is like following:

`void sendData(const std::string& data);`

`void sendData(const uint8_t* pData, size_t length);`

`void sendData(const char* pData, size_t length);`

under this assumption you are basically just sending 'T' with one byte length (char).

sephyrah, 2019-05-10:
I'm using the sendData from the uart_async example:
int sendData(const char* logName, const char* data){
... returns txBytes;}

Yes, i suppose i'm only sending that 'T'.

I'm using most part of the example's code, like sendData and tx/rx_task, is there a way to send the Data ?

sephyrah, 2019-05-10:
> What is the argument for the sendData function? I will assume that the declaration of the sendData function is like following:
>
> `void sendData(const std::string& data);`
>
> `void sendData(const [...]
I'm using the sendData from the uart_async example:
int sendData(const char* logName, const char* data){
... returns txBytes;}

Yes, i suppose i'm only sending that 'T'.

I'm using most part of the example's code, like sendData and tx/rx_task, is there a way to send the Data ?

sephyrah, 2019-05-10:
Well, it was not need to convert Data to ASCII, it sends, all i had to do was initialize the UART for RS-485 and end the Data with .\n\r, i received the menu in hex.
Thank you so much

closed 2019-05-10 by sephyrah