nkolban/esp32-snippets #869: ble_client example crash on disconnect with adafruit huzzah esp32

chacalnoir, 2019-05-19:
I am testing with an adafruit huzzah esp32, currently running the ble_client.ino example.  The only changes I made to the example were changing the serviceUUID to match an advertized service on the BLE peripheral I am testing, changing the scan time to 30 seconds, and commenting out the pRemoteCharacteristic->writeValue line since the characteristic isn't found.  Line 74 should disconnect from the server since the characteristic is not found.  Each time this disconnect occurs, the ESP32 crashes and reboots.  I have attached the serial monitor prinout below.  I found this is true for both the Arduino library version of the code and the newest master on the git repo.  I also commented out BLEDevice::getScan()->start(0); in line 156 to make sure the issue wasn't with restarting the scan.  Has anyone else experienced this?  Are there settings for the ESP32 that I am missing?

Thank you.

BLE Advertised Device found: Name: , Address: c0:28:8d:83:90:f5, manufacturer data: c0288d8390f5, serviceUUID: 72daa6c3-29c2-6283-0c4a-2818e4d37e75
BLE Advertised Device found: Name: ACGAM R1          , Address: ff:ff:97:01:6f:c7, serviceUUID: 00001812-0000-1000-8000-00805f9b34fb
Forming a connection to ff:ff:97:01:6f:c7
 - Created client
 - Connected to server
 - Found our service
Failed to find our characteristic UUID: beb5483e-36e1-4688-b7f5-ea07361b26a8
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d1924: 05b89aac 1014c1a2 95252011
Core 1 register dump:
PC      : 0x400d192a  PS      : 0x00060f30  A0      : 0x800d70b8  A1      : 0x3ffca300  
A2      : 0x3ffc4fbe  A3      : 0x3ffc60a4  A4      : 0x00000020  A5      : 0x80000020  
A6      : 0x00000000  A7      : 0x3ffba2c4  A8      : 0x800d192a  A9      : 0x3ffca2b0  
A10     : 0x00000000  A11     : 0x19ba7153  A12     : 0x19ba7153  A13     : 0x3ffc60a4  
A14     : 0x3ffe1a10  A15     : 0x00a80010  SAR     : 0x00000010  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff  

Backtrace: 0x400d192a:0x3ffca300 0x400d70b5:0x3ffca350 0x40090461:0x3ffca370

Rebooting...

chegewara, 2019-05-19:
`Failed to find our characteristic UUID: beb5483e-36e1-4688-b7f5-ea07361b26a8`
Check characteristic UUID, or see this:
https://github.com/nkolban/esp32-snippets/issues/399#issuecomment-493703302

chacalnoir, 2019-05-19:
Thank you, @chegewara .  You're right - the issue was with the characteristic retrieval.  The timing of when it was crashing looked like it was later in the program than an issue with the characteristic.  I still have crashes when retrieving characteristics, but I was able to run the program far enough to find the characteristic UUID that I wanted, and adding that valid UUID prevents the crash.

closed 2019-05-19 by chacalnoir