nkolban/esp32-snippets #870: Problem with ble callbacks.

ArTyGill, 2019-05-22:
On esp32 I have a bluetooth server, it has one service configured and two characteristics in it.
When sending a large message (60 characters) to one of the characteristics, the callback arrives at all characteristics.

I am use NRF CONNECT, mtu is set to 512 bytes.
ESP32 dev module.

zz-zsys, 2019-05-22:
hi @ArTyGill,

can you provide a simple example code for testing? Did you use the latest code from esp32-snippets?

ArTyGill, 2019-05-22:
hi @zz-zsys 

I am using the latest arduino ide.
But he made some changes in the library in order to send bluetooth to specific customers.

I have example in issue arduino-esp32.
https://github.com/espressif/arduino-esp32/issues/2796

ArTyGill, 2019-05-22:
My changes:
###BLECharacteristic.cpp

@@ -142,10 +142,21 @@ BLEDescriptor* BLECharacteristic::getDescriptorByUUID(BLEUUID descriptorUUID) {
  */
 uint16_t BLECharacteristic::getHandle() {
 	return m_handle;
 } // getHandle
 
+/**
+ * @brief Get the conn_id of write to the characteristic.
+ * @return conn_id of write to the characteristic.
+ */
+
+uint16_t BLECharacteristic::getWriteID(){
+
+	return m_writeId;
+
+} // getWriteId
+
 void BLECharacteristic::setAccessPermissions(esp_gatt_perm_t perm) {
 	m_permissions = perm;
 }
 
 esp_gatt_char_prop_t BLECharacteristic::getProperties() {
@@ -278,10 +289,11 @@ void BLECharacteristic::handleGATTServerEvent(
 			if (param->write.handle == m_handle) {
 				if (param->write.is_prep) {
 					m_value.addPart(param->write.value, param->write.len);
 				} else {
 					setValue(param->write.value, param->write.len);
+					m_writeId = param->write.conn_id;
 				}
 
 				log_d(" - Response to write event: New value: handle: %.2x, uuid: %s",
 						getHandle(), getUUID().toString().c_str());
 
@@ -472,11 +484,11 @@ void BLECharacteristic::indicate() {
  * @brief Send a notify.
  * A notification is a transmission of up to the first 20 bytes of the characteristic value.  An notification
  * will not block; it is a fire and forget.
  * @return N/A.
  */
-void BLECharacteristic::notify(bool is_notification) {
+void BLECharacteristic::notify(int32_t connection, bool is_notification) {
 	log_v(">> notify: length: %d", m_value.getValue().length());
 
 	assert(getService() != nullptr);
 	assert(getService()->getServer() != nullptr);
 
@@ -501,22 +513,44 @@ void BLECharacteristic::notify(bool is_notification) {
 		if (p2902 != nullptr && !p2902->getIndications()) {
 			log_v("<< indications disabled; ignoring");
 			return;
 		}
 	}
-	for (auto &myPair : getService()->getServer()->getPeerDevices(false)) {
-		uint16_t _mtu = (myPair.second.mtu);
-		if (m_value.getValue().length() > _mtu - 3) {
-			log_w("- Truncating to %d bytes (maximum notify size)", _mtu - 3);
+	size_t connections = getService()->getServer()->getConnectedCount();
+	if(connection < 0){
+		for (auto &myPair : getService()->getServer()->getPeerDevices(false)) {
+			uint16_t _mtu = (myPair.second.mtu);
+			if (m_value.getValue().length() > _mtu - 3) {
+				log_w("- Truncating to %d bytes (maximum notify size)", _mtu - 3);
+			}
+
+			size_t length = m_value.getValue().length();
+			if(!is_notification)
+				m_semaphoreConfEvt.take("indicate");
+			esp_err_t errRc = ::esp_ble_gatts_send_indicate(
+					getService()->getServer()->getGattsIf(),
+					myPair.first,
+					getHandle(), length, (uint8_t*)m_value.getValue().data(), !is_notification); // The need_confirm = false makes this a notify.
+			if (errRc != ESP_OK) {
+				log_e("<< esp_ble_gatts_send_ %s: rc=%d %s",is_notification?"notify":"indicate", errRc, GeneralUtils::errorToString(errRc));
+				m_semaphoreConfEvt.give();
+				return;
+			}
+			if(!is_notification)
+				m_semaphoreConfEvt.wait("indicate");
 		}
+	}
+	else
+	{
+
 
 		size_t length = m_value.getValue().length();
 		if(!is_notification)
 			m_semaphoreConfEvt.take("indicate");
 		esp_err_t errRc = ::esp_ble_gatts_send_indicate(
 				getService()->getServer()->getGattsIf(),
-				myPair.first,
+				connection,
 				getHandle(), length, (uint8_t*)m_value.getValue().data(), !is_notification); // The need_confirm = false makes this a notify.
 		if (errRc != ESP_OK) {
 			log_e("<< esp_ble_gatts_send_ %s: rc=%d %s",is_notification?"notify":"indicate", errRc, GeneralUtils::errorToString(errRc));
 			m_semaphoreConfEvt.give();
 			return;

ArTyGill, 2019-05-22:
BLECharacteristic.h

@@ -62,11 +62,11 @@ public:
 	BLEUUID        getUUID();
 	std::string    getValue();
 	uint8_t*       getData();
 
 	void indicate();
-	void notify(bool is_notification = true);
+	void notify(int32_t connection, bool is_notification = true);
 	void setBroadcastProperty(bool value);
 	void setCallbacks(BLECharacteristicCallbacks* pCallbacks);
 	void setIndicateProperty(bool value);
 	void setNotifyProperty(bool value);
 	void setReadProperty(bool value);
@@ -80,10 +80,11 @@ public:
 	void setWriteProperty(bool value);
 	void setWriteNoResponseProperty(bool value);
 	std::string toString();
 	uint16_t getHandle();
 	void setAccessPermissions(esp_gatt_perm_t perm);
+	uint16_t getWriteID();
 
 	static const uint32_t PROPERTY_READ      = 1<<0;
 	static const uint32_t PROPERTY_WRITE     = 1<<1;
 	static const uint32_t PROPERTY_NOTIFY    = 1<<2;
 	static const uint32_t PROPERTY_BROADCAST = 1<<3;
@@ -98,10 +99,11 @@ private:
 	friend class BLECharacteristicMap;
 
 	BLEUUID                     m_bleUUID;
 	BLEDescriptorMap            m_descriptorMap;
 	uint16_t                    m_handle;
+	uint16_t 					m_writeId;
 	esp_gatt_char_prop_t        m_properties;
 	BLECharacteristicCallbacks* m_pCallbacks;
 	BLEService*                 m_pService;
 	BLEValue                    m_value;
 	esp_gatt_perm_t             m_permissions = ESP_GATT_PERM_READ | ESP_GATT_PERM_WRITE;

ArTyGill, 2019-05-22:
and FS.h

 {
 
 #define FILE_READ       "r"
 #define FILE_WRITE      "w"
 #define FILE_APPEND     "a"
+#define FILE_READ_WRITE "r+"
 
 class File;
 
 class FileImpl;
 typedef std::shared_ptr<FileImpl> FileImplPtr;

ArTyGill, 2019-05-28:
Did anyone have such a problem?