nkolban/esp32-snippets #871: ESP32 No Name in Scan

JPLR10, 2019-05-24:
I have an application(Arduino) where I scan for BLE iBeacons and others, the Beacon name is obviously important. I used your library for the implementation and it worked perfectly, but for some reason it stopped getting the Name, (Complete Local Name).
I even reloaded your BLE_scan example and below is the result:
Advertised Device: Name: , Address: 24:4b:03:82:21:1c, manufacturer data: 750042040180ae244b0382211c264b0382211b01080000000000 
**Advertised Device: Name: , Address: c5:6b:f7:e4:0f:64, serviceUUID: 713d0000-503e-4c75-ba94-3148f18d941e 
Advertised Device: Name: , Address: f3:e1:e8:6d:12:2b, serviceUUID: 713d0000-503e-4c75-ba94-3148f18d941e** 
Advertised Device: Name: , Address: 7a:dc:c5:13:76:72, manufacturer data: 4c00100503189a8f22 
Advertised Device: Name: , Address: ea:67:34:f4:ad:d4, manufacturer data: 4c000215fda50693a4e24fb1afcfc6eb07647825273cb9c6c3 
**Advertised Device: Name: , Address: c9:3c:3d:7d:b4:23, serviceUUID: 713d0000-503e-4c75-ba94-3148f18d941e** 
Advertised Device: Name: , Address: 22:d1:a3:5a:66:1f, manufacturer data: 060001092002b2fb59ebd8902fdd1fe841ed349b0733df77d5b943fbeb 
**Advertised Device: Name: , Address: da:cf:f3:77:6f:92, serviceUUID: 713d0000-503e-4c75-ba94-3148f18d941e** 
Advertised Device: Name: , Address: f9:6e:b0:a2:0f:9f, manufacturer data: 4c000215fda50693a4e24fb1afcfc6eb0764782528cfb9c6c3 
**Advertised Device: Name: , Address: c1:12:a8:b5:f5:3c, serviceUUID: 713d0000-503e-4c75-ba94-3148f18d941e** 
Advertised Device: Name: , Address: f1:d6:22:35:32:a0, manufacturer data: 4c000215fda50693a4e24fb1afcfc6eb07647825273cb9c6c3 
Advertised Device: Name: 20010011510000070134, Address: 24:6f:28:d4:63:b2, txPower: 3 
Advertised Device: Name: 10000012 
Devices found: 13 
The ones in BOLD are iBeacons. The bottom two are simple beacons I built for testing, it picks their names up without a problem.

I would highly appreciate it if you could offer some help.

diegoavellanedat17, 2019-06-14:
Hello, Did you solve it ? I didn't find any solution to the same problem, so I tried to save the advertised Device and save the MAC in a vector and forget about Names, but I think is not the best solution.

linxcow, 2019-07-02:
I also have the same problem getName(); returns empty string.
Does anyone have a solution?

onelittlegoat, 2020-01-26:
Any progress on this one?  I just installed the library and it works well so far except all ble device names are null.  :(.

chegewara, 2020-01-29:
Its hard to do anything if i cant reproduce issue:
```
03:27:33.927 -> Device: 
03:27:33.927 -> TX power => -7
03:27:33.927 -> RSSI => -96
03:27:33.927 -> Name: Chege
03:27:34.226 -> Device: 
03:27:34.226 -> TX power => -7
03:27:34.226 -> RSSI => -85
03:27:34.226 -> Name: Chege
03:27:34.493 -> Device: 
03:27:34.493 -> TX power => -7
03:27:34.493 -> RSSI => -85
03:27:34.493 -> Name: Chege
03:27:34.792 -> Device: 
03:27:34.792 -> TX power => -7
03:27:34.792 -> RSSI => -86
03:27:34.792 -> Name: Chege
03:27:35.092 -> Device: 
03:27:35.092 -> TX power => -7
03:27:35.092 -> RSSI => -84
03:27:35.092 -> Name: Chege
03:27:35.658 -> Device: 
03:27:35.658 -> TX power => -7
03:27:35.658 -> RSSI => -89
03:27:35.658 -> Name: Chege
03:27:36.233 -> Device: 
03:27:36.233 -> TX power => -7
03:27:36.233 -> RSSI => -91
03:27:36.233 -> Name: Chege
03:27:36.529 -> Device: 
03:27:36.529 -> TX power => -7
03:27:36.529 -> RSSI => -81
03:27:36.529 -> Name: Chege
```
My peer device is android smartphone running nRF connect with advertising.
Here is scan callback:
```
class ScanResults: public BLEAdvertisedDeviceCallbacks {
  void onResult(BLEAdvertisedDevice device)
  {
    if(device.haveName())
    {
      Serial.printf("Device: \nTX power => %d\nRSSI => %d\n", device.getTXPower(), device.getRSSI());
      Serial.printf("Name: %s\n", device.getName().c_str());
    }
  }
};
```
I can only say that i am running this code on platformio with arduino + esp-idf, wifi and mqtt (code omitted).

sylque, 2024-01-08:
A BLE peripheral can advertise its name using either a 'long name' or a 'short name' record in either an 'advertising' or a 'scan response' packet. Peripherals need to choose one of the four possible combinations. For example, a peripheral based on the Arduino BLE 33 board usually uses a "long name" record in a "scan response" packet.

If you're developing code for a central device using the **[ESP32 BLE Arduino](https://www.arduino.cc/reference/en/libraries/esp32-ble-arduino/)** library, three potential issues may arise:

### 1. This library doesn't support "short name" records

It means that any peripheral that advertises its name through a "short name" record will appear with an empty name.

Until this is fixed by the team (@chegewara ), you can easily fix it yourself in your local library. Locate the file `BLEAdvertisedDevice.cpp` and add `case ESP_BLE_AD_TYPE_NAME_SHORT:` before line 296.

The initial code is:
```c++
			switch(ad_type) {
				case ESP_BLE_AD_TYPE_NAME_CMPL: {   // Adv Data Type: 0x09
					setName(std::string(reinterpret_cast<char*>(payload), length));
					break;
				} // ESP_BLE_AD_TYPE_NAME_CMPL
````
and should become:
````c++
			switch(ad_type) {
				case ESP_BLE_AD_TYPE_NAME_SHORT: 
				case ESP_BLE_AD_TYPE_NAME_CMPL: {   // Adv Data Type: 0x09
					setName(std::string(reinterpret_cast<char*>(payload), length));
					break;
				} // ESP_BLE_AD_TYPE_NAME_CMPL
````

### 2. You forgot to use setActiveScan(true)

`BLEDevice::getScan()->setActiveScan(true)` tells the library to ask peripherals for "scan response" packets. If you don't call this function, any peripheral that advertises its name through a "scan response" packet (such as the Arduino BLE 33) will appear with an empty name.

### 3. You are using the default scan interval and window

In this library, the default scan interval and window are both of 10ms. It seems this is too small to allow for asking for (or receiving) "scan response" packets. Use at least:
````c++
BLEDevice::getScan()->setInterval(40);
BLEDevice::getScan()->setWindow(40);
````

Hope this helps.

solopilot, 2025-06-21:
FYI, I am scanning for BLE ads from a Tesla and I see the correct name show up in about 1 in 10 callbacks.  There seems to be a timing issue in the code.  Average RSSI is about -97 (weak).

Cabrio-Bob, 2026-05-10:
In my case I really need the `ESP_BLE_AD_TYPE_NAME_SHORT` added. My TPMS sensors only reveal themselves by a short name.
Since Arduinodroid doesn't let me change pre-installed libraries I am stuck.