nkolban/esp32-snippets #872: u8g2 Example for Heltec ESP32

kschoos, 2019-05-25:
Hi,
I had problems getting my Heltec ESP32 working with your I2C test code for the u8g2 display library. It was a really simple problem: The reset pin needed to be set in the u8g2 configuration.

I was curious if I could open a pull-request for my example configuration, so that future people like me may not need so much time debugging?

Best regards

TheEarl1, 2020-03-05:
I would be interested in your pull request!

Just setting the reset pin is not working for me. It appears, from the Heltec library that the reset GPIO is inverted from what the HAL is expecting. Did you need to modify the hal init function?

mabi, 2020-03-11:
> I would be interested in your pull request!
> 
> Just setting the reset pin is not working for me. It appears, from the Heltec library that the reset GPIO is inverted from what the HAL is expecting. Did you need to modify the hal init function?

@TheEarl1 I got it working by setting the reset pin
u8g2_esp32_hal.reset =16
and calling the i2c setup 
u8g2_Setup_ssd1306_**i2c**_128x64_noname_f(

TheEarl1, 2020-04-20:
I was missing the i2c setup, and was calling the spi setup. The reset GPIO is not inverted from what the i2c setup expects. 

thanks @mabi