nkolban/esp32-snippets #874: BLE client gets stuck while calling BLEClient::connect()

TSOgang, 2019-05-27:
First of all, hello and thank you for your libraries and excuse me for this probably very simple question.

Very recently I have begun learning BLE and the esp32 and my goal for the moment is to successully establish a BLE connection between an esp32 (as client) and the nRF52 (as server). I believe the nRF52 works, since I can connect to it using my phone and control the value of characteristics in it. However, I can't get the esp32 to connect to it successully. It is able to scan and detect nearby devices and find the nRF52's adress no problem, but when a connection attempt is made using the connect() function, the program seems to wait indefinitely and no connection is ever established. While I have done some research on semaphores, I am still quite unfamilliar with them.

I wish to understand where it gets stuck in the connect() fuction, why it does that and how can I make it work.
Here is my code (I am using the Arduino framework in PlatformIO):

```
#include <Arduino.h>

#include "BLEDevice.h"

static BLEUUID serviceUUID("6e400001-b5a3-f393-e0a9-e50e24dcca9e");
static BLEUUID characteristicUUID("0000004f-0000-1000-8000-00805f9b34fb");
String My_BLE_Address = "d1:c7:bd:79:c4:05";

static BLERemoteCharacteristic* pRemoteCharacteristic;
BLEScan* pBLEScan;
BLEScanResults foundDevices;
static BLEAddress *Server_BLE_Address;
String Scanned_BLE_Address;

boolean paired = false;      // Flag verifying if we are paired with our device
boolean foundDevice = false; // Flag verifying if we found our device

bool connectToserver (BLEAddress pAddress){
    
    BLEClient*  pClient  = BLEDevice::createClient();
    Serial.println(" - Created client");
     
     
    // Connect to the BLE Server.
    pClient->connect(pAddress); // Stuck here...
    Serial.println(" - Connected to server");

    // Obtain a reference to the service we are after in the remote BLE server.
    BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService != nullptr)
    {
      Serial.println(" - Found our service");
      return true;
    }
    else
    return false;

    // Obtain a reference to the characteristic in the service of the remote BLE server.
    pRemoteCharacteristic = pRemoteService->getCharacteristic(characteristicUUID);
    if (pRemoteCharacteristic != nullptr)
      Serial.println(" - Found our characteristic");

      return true;
}

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks 
{ 
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.printf("Scan Result: %s \n", advertisedDevice.toString().c_str());
      if(foundDevice == false){   // Flag verifying if we found our device
      Server_BLE_Address = new BLEAddress(advertisedDevice.getAddress()); // Update Server_BLE_Address if we didn't
      Scanned_BLE_Address = Server_BLE_Address->toString().c_str();
      }
      if(Scanned_BLE_Address == My_BLE_Address) // Compares the scanned adress to what we are looking for
      {
        foundDevice = true; // Found our device
      }
      Serial.println("Comparing to our server adress: ");
      Serial.println(My_BLE_Address);
    }
};

void setup() {
    Serial.begin(115200); //Start serial monitor 
    Serial.println("ESP32 BLE Client program"); //Intro message 

    BLEDevice::init("ESP32");
    pBLEScan = BLEDevice::getScan(); //create new scan
    pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks()); //Call the class that is defined above 
    pBLEScan->setActiveScan(false); //active scan uses more power, but get results faster
}

void loop() {

  foundDevices = pBLEScan->start(3); //Scan for 3 seconds to find the server
  uint8_t devicesCount = foundDevices.getCount(); 
  Serial.printf("%d devices found \n", devicesCount);
  while (devicesCount >= 1 && foundDevice == true) // We found our server
  {
    if (foundDevice == true && paired == false) // We're not connected yet
    {
      Serial.println("Found our Device... connecting to Server as client \n");
       if (connectToserver(*Server_BLE_Address)) // Attempt a connection to the server   ...Stuck here...
      {
      paired = true;
      Serial.println("********************CONNECTED TO SERVER************************");
     
      break;
      }
      else
      {
      Serial.println("Pairing failed");
      foundDevice = false;
      break;
      }
    }
    devicesCount = 0; // Reset the number of devices
  }
  pBLEScan->stop();
  delay(5000);
}
```
And here is what the serial monitor looks like:
```
ESP32 BLE Client program
Scan Result: Name: , Address: 3f:43:89:02:09:99, manufacturer data: 0600010920020273111e440a64443a2bbdb913d98c83605f724bf88dd6
Comparing to our server adress:
d1:c7:bd:79:c4:05
Scan Result: Name: , Address: 21:5f:63:6c:59:18, manufacturer data: 0600010920024cbbbc219822bf9d445e8c12280b7f0e218a9c491fd485
Comparing to our server adress:
d1:c7:bd:79:c4:05
Scan Result: Name: , Address: 22:08:60:11:95:b3, manufacturer data: 06000109200267b9b0dc5ac0c049f892ec1bb09f3afed1bac30914b6fb
Comparing to our server adress:
d1:c7:bd:79:c4:05
Scan Result: Name: , Address: 3b:e3:63:51:43:55, manufacturer data: 060001092002578fe17a471e0a3fff4cb07172df27a10fc4835298acd4
Comparing to our server adress:
d1:c7:bd:79:c4:05
Scan Result: Name: nRF52, Address: d1:c7:bd:79:c4:05, txPower: 4
Comparing to our server adress:
d1:c7:bd:79:c4:05
Scan Result: Name: , Address: 04:b3:68:61:52:c6, manufacturer data: 060001092002810c02a9395aa4bc0d6ca4ee830ff5dc8f1dfc49e15e74
Comparing to our server adress:
d1:c7:bd:79:c4:05
Scan Result: Name: , Address: 74:97:07:46:24:ca, manufacturer data: 4c001005131cc48f22
Comparing to our server adress:
d1:c7:bd:79:c4:05
Scan Result: Name: , Address: 08:c0:21:ff:af:48, manufacturer data: 060001092002585e2df55dd581bf04830fe41d2f2d0cd131b433f79c76
Comparing to our server adress:
d1:c7:bd:79:c4:05
Scan Result: Name: , Address: f4:5c:89:a7:90:36, manufacturer data: 4c0010050b1c604ec2
Comparing to our server adress:
d1:c7:bd:79:c4:05
Scan Result: Name: , Address: 52:33:f9:76:9e:a1, manufacturer data: 4c000c0e00f2e83ef80545a417a96ba8e194
Comparing to our server adress:
d1:c7:bd:79:c4:05
Scan Result: Name: , Address: 7c:87:5c:cc:fd:c8, manufacturer data: 4c0010050318008367
Comparing to our server adress:
d1:c7:bd:79:c4:05
11 devices found
Found our Device... connecting to Server as client

 - Created client
```

Do you think you can help me?
Thanks in advance.

PotatoSauceVFX, 2019-05-30:
Same issue here. I noticed that after a reset on the client this happens, after resetting once or twice again it reconnects.

chegewara, 2019-05-30:
It can be few reasons, but if i would have to guess its because in arduino version there is still few bugs. My advice is to use this library instead, or to find someone who will update arduino version.

DasHammer, 2019-06-02:
I am having the same issue.

TSOgang, 2019-06-03:
Please excuse for my silence, I have stopped working on the esp-32 (a little bit out of frustration) to concentrate on the other chip im working on. This morning I wanted to try again from the beginning and simply copy pasted the client example in the "ESP32 BLE Arduino" library (that I have tried before without success) and changed up the Uuids to match my other chip's Uuids.

I gave it a go and it...connected successfully without trouble... 
My best guess would be that my problem was on the server side, and since I worked on it, I must have corrected the problem without realising it.
I am still unsure as to why my program got stuck in the connect() function (and I see im not the only one), so I won't close this issue just yet.

If there is one thing I know it's that the problem dosen't seem to come from the libraries (since it worked), but I can't know for sure.

majstor12, 2019-06-03:
Hi!
I'm also using Arduino to program ESP32. This issue is very frustrating. I don't have solution for this problem, but I have noticed that it occurs when there are lot of scanned devices.
 It looks that code stucks on that line: 
`BLERemoteService* pRemoteService = pClient->getService(serviceUUID); `
in Client example.

Any ideas how to fix this?

chegewara, 2019-06-04:
@majstor12 yes, use this repository instead of arduino library. In arduino library is some bug that has been fixed here long time ago (few other bugs too).

arsab, 2019-06-23:
Do we expect to have these changes/improvements propagated to arduino library and how?
we see following update having made significant changes, but could not find which issues it is fixing:
https://github.com/nkolban/esp32-snippets/commit/e6d9ff2b9ec8adbccd7d629e85fb419c9d56a9e4

YeonwooSung, 2019-06-29:
@chegewara Might be a silly question, how can I use this repository instead of arduino library? Can you please teach me how to do that?

chegewara, 2019-06-29:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/ArduinoBLE.md

YeonwooSung, 2019-06-30:
@chegewara Thanks :)

jiangminghe, 2019-08-11:
Hi, I have got the same problem
// -------------------------------
pClient->connect(pAddress); // Stuck here...
//--------------------------------
I solved this problem by modifying the FreeRTOS.cpp file.
You can see this command in line 77 of FreeRTOS.cpp
xSemaphoreTake(m_semaphore, portMAX_DELAY);
As the portMAX_DELAY value is 0xffffffffUL, maybe the code will wait for about 50 days to connect to the BLE device.
So that, I changed the code like this and my code didn't get stuck anymore.
xSemaphoreTake(m_semaphore, 60000);

Thanks.

thomastech, 2019-10-28:
> So that, I changed the code like this and my code didn't get stuck anymore.
> xSemaphoreTake(m_semaphore, 60000);

Thanks for sharing your hang workaround. Instead of 60000 I'm using 15000; The smaller value works well for me (shorter "hang" times).
- Thomas

jiangminghe, 2019-10-28:
Thanks for your notice, Thomas.

thomastech, 2019-10-31:
@jiangminghe During validation I found another xSemaphoreTake() that caused an endless hang in my client application. This required another workaround patch at line 198: I changed the timeout to 15 seconds, as follows:
`rc = ::xSemaphoreTake(m_semaphore, 15000UL) == pdTRUE;`

I recognize that these edits are workarounds, not permanent patches. But I will use them until the semaphore issues are resolved. 

- Thomas

chegewara, 2019-11-01:
Patch that is fixing semaphores issue has been added to arduino-esp32 repository some time ago.
Because code in BLEClient is just a bit different the same patch didnt work here, or maybe i was not lucky.

thomastech, 2019-11-02:
I'm using the latest arduino-esp32 repository (last patch was from about 30 days ago). If there are more recent semaphore fixes then I would definitely want to try them out.
- Thomas

chegewara, 2019-11-02:
Sorry, it was the latest one, in this case there is somewhere else semaphore that causing stuck.

vcmorini, 2020-01-05:
Anyone could implement a **watchdog** for this case?
I am using Arduino IDE and my watchdog, created by Ticker lib, does not get called.
Thks :)

Prathik-Jain, 2020-01-07:
I tried replacing the libraries as suggested by @chegewara , I see no change. The program still gets stuck.

ufanders, 2020-01-25:
Hi, I am having this issue as well:

`bool connectToServer() {

    BLEClient*  pClient  = BLEDevice::createClient();
    pClient->setClientCallbacks(new MyClientCallback());

    // Connect to the remote BLE Server.
    Serial.printf("Connecting to %s.\n", serverAddress);
    //Serial.println(myDevice->getAddress().toString().c_str());
    pClient->connect(BLEAddress(serverAddress)); //myDevice);  // if you pass BLEAdvertisedDevice instead of address, it will be recognized type of peer device address (public or private)

//**** MANY TIMES, DOES NOT EXECUTE CODE BELOW ****//

    // Obtain a reference to the service we are after in the remote BLE server.
    pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == NULL) {
      Serial.println("Service not found.");
      M5.Lcd.println("Service not found.");
      M5.update();
      pClient->disconnect();
      return false;
    }
    Serial.println("Service found.");
    M5.Lcd.println("Service found.");
    M5.update();
}`

Any new updates, @chegewara?

chegewara, 2020-01-26:
It is possible that calling this line:
`pClient->connect(BLEAddress(serverAddress)); //myDevice); `
not necessarily ends with connecting to peer device. I would suggest to add additional check after this line:
```
if(!pClient->isConnected())
    return false;
```

ufanders, 2020-01-28:
Ok thanks I'll try the timed semaphore fix.

BojanJurca, 2020-02-11:
You can not connect while scanning. Stop scanning first.

ravsten, 2020-03-03:
While reading #757 I have found that `connect` call can actually take two arguments - `BLEAddress` as the first one and `esp_ble_addr_type_t type` as a second (optional) argument that defaults to `BLE_ADDR_TYPE_PUBLIC`.
However, as mentioned, "most if not all smartphones, tablets and laptops are using random addressing" and `esp_ble_addr_type_t type` has to be set to `BLE_ADDR_TYPE_RANDOM` (https://github.com/nkolban/esp32-snippets/issues/757#issuecomment-448336410).
Worked for me!

chegewara, 2020-03-03:
@ravsten There is overloaded connect function, which will figure out address type, but you need to pass advertised device instead of address.

ravsten, 2020-03-04:
> @ravsten There is overloaded connect function, which will figure out address type, but you need to pass advertised device instead of address.

To avoid misleading anyone I want to mention that I am using an Arduino built-in BLE ESP32 library so I guess that is why I pass BLEAddress instead of BLEDevice. Correct @chegewara ?

chegewara, 2020-03-04:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEClient.h#L36

vcmorini, 2020-03-07:
> Ok thanks I'll try the timed semaphore fix.

Hello, could you share with us how to do it? :)

xsaco07, 2020-05-28:
Hi I'm having the same issue here. I'm trying to connect to ESP32 and the Client is stuck in connect(myAdvDevice) line. Is there any update here?
I'm using the Arduino library.
The semaphores fix is the only option?

Thanks.

xsaco07, 2020-06-07:
I edited the FreeRTOS.cpp file changing the `xSemaphoreTake()` parameter to 15000UL as mentioned in the comments and I'm not getting the result expected yet. It still gets stuck on the connect(myAdvDecice) function call.

thomastech, 2020-06-07:
Check the file again and confirm you have changed **ALL**  the occurrences of xSemaphoreTake()  that used the portMAX_DELAY to 15000UL. If you've already done that then I doubt your issue is due to the semaphore timeout hang issue.
- Thomas

Huntercover, 2020-09-27:
Hello, I am facing the same stuck whenever trying to connect to the fourth server. Is there any chance to connect to more than 3 servers?
btw: I will try to use 2 ESP32 to connect to 6 BLE servers, hopefully that will work for me.
Regards Andreas

LK-Simon, 2021-09-05:
**Do not attempt to Connect to a BLE Server before the call to `BLEScan->start(...);` returns!**

You need to initiate the scan, on your `onResult` Callback method, you need to determine if it is the device in which you are interested. If it is, you need to store a reference to the `BLEAdvertisedDevice`

Once the scan call returns (as stated in the first line of this response) you can *then* connect to this `BLEAdvertisedDevice` and it will work.

MertGurdogann, 2021-09-06:
Hi There, i have same problem in connecting to peripheral. It's always stuck on connect command line. I also have already try Mr. Andreas Polar Receiver example but that has same stuck too. I use ESP32 pico kit. 

Here's my demo code. Does anyone have a solution for this issue? I will be grateful if you could help me. 

```
#include` "BLEDevice.h"
// The remote service we wish to connect to.
static  BLEUUID serviceUUID(BLEUUID((uint16_t)0x180D));
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID(BLEUUID((uint16_t)0x2A37));
static BLEAddress *pServerAddress;
static BLEScan* pBLEScan;
static BLEClient*  pClient;
```
```
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks
{
    void onResult(BLEAdvertisedDevice advertisedDevice)
    {
      Serial.print("BLE Advertised Device found: ");
      Serial.println(advertisedDevice.toString().c_str());
      pServerAddress = new BLEAddress(advertisedDevice.getAddress());

      if (advertisedDevice.haveServiceUUID() && advertisedDevice.getServiceUUID().equals(serviceUUID))
      {
        Serial.println("=============== Device found! =============== ");
        Serial.print("Name: ");
        Serial.print(advertisedDevice.getName().c_str());
        Serial.print("\tAddress: ");
        Serial.println(pServerAddress->toString().c_str());
        advertisedDevice.getScan()->stop();
        //       pClient->connect(*pServerAddress);   I try those method but i couldn't succsessful on connect 
        pClient->connect(&advertisedDevice);
        Serial.println(" - Connected to server");
      }

      /******** This releases the memory when we're done. ********/
      delete pServerAddress;
    }
}; // MyAdvertisedDeviceCallbacks
```
```
void setup()
{
  Serial.begin(115200);
  Serial.println("Starting Arduino BLE Client application...");
  BLEDevice::init("");
  pClient  = BLEDevice::createClient();
  Serial.println(" - Created client");
  pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true);
  pBLEScan->start(30);
}
void loop()
{
  if (pClient->isConnected()) Serial.println("connect");
  else Serial.println("non connect");
  while (1);
}
```

And Here's my output from Serial monitor.

```
Starting Arduino BLE Client application...
 - Created client
BLE Advertised Device found: Name: 36374-65, Address: c4:2a:21:cf:6e:c2, appearance: 833, manufacturer data: ffff6b00168e41400001020c, serviceUUID: 0000180d-0000-1000-8000-00805f9b34fb
=============== Device found! =============== 
Name: 36374-65	Address: c4:2a:21:cf:6e:c2
non connect
```

chegewara, 2021-09-06:
There is second parameter in connect function:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEClient.h#L37
for that reason there is another connect function, which should automatically detect address type:
https://github.com/espressif/arduino-esp32/blob/master/libraries/BLE/src/BLEClient.h#L36

AndreasHobby, 2021-09-06:
@chegewara: and this solves the stuck?

In the meanwhile and as a workaround I am using the other CORE for any BLE connect tasks.

chegewara, 2021-09-06:
Im not sure it will solve, but it was the reason in many other cases. Thats why i had to write that second connect function.

LK-Simon, 2021-09-06:
If anyone wants a confirmed-working example of BLE connections for the ESP32 boards... I have a repository as part of my Automatic Discovery & Pairing articles. https://github.com/Flowduino/ESP32_ESP-Now_Pairing
It's only really intended to facilitate Automated Discovery (I'm using BLE for Discovery, then ESP-Now for actual communication) but this code is functioning just fine.

You might be able to figure out what you need to change in your own respective codebases from my code in that repository.

Enjoy!

MertGurdogann, 2021-09-07:
@LK-Simon Thank you I solved the stuck problem and made an example about it. Those who have problems with Connect can try this example.

[https://github.com/MertGurdogann/BLEconnectExample/blob/704ed8f278c6da4ff9f46c14c49cd087fa31f9c1/BLEconnect/BLEconnect.ino](url)

Here is example output from Serial Monitor
```
Advertised Device: Name: , Address: 6e:46:76:f6:42:3c, manufacturer data: 4c0010067e1e3f5fd2f5, txPower: 8 
Advertised Device: Name: Decathlon Dual HR, Address: ed:13:79:f4:72:01, appearance: 833, serviceUUID: 0000180d-0000-1000-8000-00805f9b34fb 
Advertised Device: Name: 36374-65, Address: c4:2a:21:cf:6e:c2, appearance: 833, manufacturer data: ffff6b00168e41400001020c, serviceUUID: 0000180d-0000-1000-8000-00805f9b34fb 
Found Devices
0.   Device Name:36374-65
1.   Device Name:Decathlon Dual HR
Connecting
Connection Successfull
Connected 	Decathlon Dual HR
Connecting
Connection Successfull
Connected 	36374-65
```

Best regards.

chegewara, 2021-09-07:
From the code i am guesing that another `connect()` function helped.

JoFRT, 2021-10-14:
Hello,
Many thanks to your contribution in this thread, I understood 2 important things for my sketch.
(So now my sketches will be not stuck anymore and I understood that my 2nd device as a random address configuration).
So now I can get a connection with my 2nd device too. But I cannot connect to a service : the ESP32 reboots because of this error :
`>> Guru Meditation Error: Core  0 panic'ed (Double exception).`
With my 1st device (static address configuration) everything is perfect : I can read and write characteristics.
So by comparison I saw that with my 2nd device I have those 2 lines in addition just before the reset :
```
>> [ 11002][D][BLEClient.cpp:493] handleGAPEvent(): BLEClient ... handling GAP event!
>> custom gap event handler, event: 20
```
I am wandering why this 2nd exception(gap event) occurs !
Below I give you the log (Log with Debug config only because I do not have the Verbose’s in my arduino v1.8.15, and I do not know to add it  ).
After that I give you also the backtrace details.
Could you please help me with this issue ?

```
Starting Arduino BLE Client application...
custom gap event handler, event: 2 
custom gap event handler, event: 7 
[   749][D][BLEAdvertisedDevice.cpp:472] setRSSI(): - setRSSI(): rssi: -82
[   749][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x01 (), length: 1, data: 05
[   753][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x02 (), length: 2, data: 0018
[   762][D][BLEAdvertisedDevice.cpp:500] setServiceUUID(): - addServiceUUID(): serviceUUID: 00001800-0000-1000-8000-00805f9b34fb
[   773][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x09 (), length: 8, data: 522d444343393437
[   783][D][BLEAdvertisedDevice.cpp:461] setName(): - setName(): name: R-DCC947
[   790][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0xff (), length: 8, data: 600154104be04638
[   800][D][BLEAdvertisedDevice.cpp:449] setManufacturerData(): - manufacturer data: 600154104be04638
[   809][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0xff (), length: 29, data: 600154104be04638010a044b000301e500190414000000000000000000
[   822][D][BLEAdvertisedDevice.cpp:449] setManufacturerData(): - manufacturer data: 600154104be04638010a044b000301e500190414000000000000000000
BLE Advertised Device found: Name: R-DCC947, Address: a4:c1:38:46:e0:4b, manufacturer data: 600154104be04638010a044b000301e500190414000000000000000000, serviceUUID: 00001800-0000-1000-8000-00805f9b34fb
custom gap event handler, event: 3 
(…)
[  3823][D][BLEScan.cpp:105] handleGAPEvent(): Ignoring 2b:45:52:00:e0:26, already seen it.
custom gap event handler, event: 3 
[  3919][D][BLEScan.cpp:105] handleGAPEvent(): Ignoring a4:c1:38:46:e0:4b, already seen it.
custom gap event handler, event: 3 
[  3919][D][BLEScan.cpp:105] handleGAPEvent(): Ignoring 2b:45:52:00:e0:26, already seen it.
custom gap event handler, event: 3 
[  4122][D][BLEScan.cpp:105] handleGAPEvent(): Ignoring 2b:45:52:00:e0:26, already seen it.
custom gap event handler, event: 3 
[  4183][D][BLEScan.cpp:105] handleGAPEvent(): Ignoring 2b:45:52:00:e0:26, already seen it.
custom gap event handler, event: 3 
[  4322][D][BLEScan.cpp:105] handleGAPEvent(): Ignoring 2b:45:52:00:e0:26, already seen it.
custom gap event handler, event: 3 
[  4339][D][BLEScan.cpp:105] handleGAPEvent(): Ignoring 2b:45:52:00:e0:26, already seen it.
custom gap event handler, event: 3 
[  4353][D][BLEAdvertisedDevice.cpp:472] setRSSI(): - setRSSI(): rssi: -43
[  4354][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x01 (), length: 1, data: 0a
[  4358][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x07 (), length: 16, data: 00c7c44ee36c51a7334be8ed5a0eb803
[  4369][D][BLEAdvertisedDevice.cpp:500] setServiceUUID(): - addServiceUUID(): serviceUUID: 03b80e5a-ede8-4b33-a751-6ce34ec4c700
[  4380][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x09 (), length: 20, data: 4e5558204d494748545920504c5547204d494449
[  4392][D][BLEAdvertisedDevice.cpp:461] setName(): - setName(): name: NMPM
BLE Advertised Device found: Name: NMPM, Address: cb:4e:fd:d7:02:d5, serviceUUID: 03b80e5a-ede8-4b33-a751-6ce34ec4c700
Found our device!  address: C_fct: Leaving "Setup" function...
C_fct: Start "loop" function...
Wait the end of searching device, 3s...
custom gap event handler, event: 3 
custom gap event handler, event: 3 
custom gap event handler, event: 3 
custom gap event handler, event: 3 
custom gap event handler, event: 3 
custom gap event handler, event: 18 
custom gap event handler, event: 18 
Wait 3s more
Forming a connection to cb:4e:fd:d7:02:d5
 - Created client
Server address type: 1
[ 10433][I][BLEDevice.cpp:622] addPeerDevice(): add conn_id: 0, GATT role: client
[ 10437][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[... 39 lines cut ...]
```

```
Decoding stack results
0x401181ca: btc_transfer_context at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/btc/core/btc_task.c line 248
0x400ece4b: btc_gattc_cback at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/btc/profile/std/gatt/btc_gattc.c line 194
0x4011d327: bta_gattc_search_service at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/bta/gatt/bta_gattc_cache.c line 1115
[... cut ...]
0x4010411a: gatt_le_data_ind at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/stack/gatt/gatt_main.c line 587
0x401112e1: l2c_rcv_acl_data at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/stack/l2cap/l2c_main.c line 275
0x400ffada: btu_hci_msg_process at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/host/bluedroid/stack/btu/btu_task.c line 144
0x40119f1b: osi_thread_run at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/bt/common/osi/thread.c line 67
```

chegewara, 2021-10-14:
> Guru Meditation Error: Core 0 panic'ed (Double exception).

This is very nasty error, because its more or less telling that you have error in code somewhere (maybe in library) and this error itself is causing to generate logs or something which is causing another error.
Usually (or maybe always) it means very bad memory corruption bug.

`custom gap event handler, event: 20 `
This is `UPDATE_CONN_PARAMS` event.

I dont know your code and purpose of it, but is it possible that 2 devices you are trying to connect to have 2 completely different set of services and characteristics?
Did you try to connect to each device separately?

JoFRT, 2021-10-14:
Hello Chegewara,
Thanks for replying.
In fact I use exactly the same sketch for only 1 device connection. I change the UUID service, UUID characteristic and the device address in the sketch depending of the device I want to connect with.
Both device are totally different.

JoFRT, 2021-10-15:
After a lot of trial my sketch works only with `"https://github.com/espressif/arduino-esp32"`, with the "snippets" I have to many error on delcarations.
I finally found how to get the Verbose details so I share it below with the associated backtrace; Hoping that means something for you.
I have also found a sketch more complex but with a similar device `"https://github.com/max22-/ESP32-BLE-MIDI/blob/master/examples/02-Basic-Midi-Client/02-Basic-Midi-Client.ino"`. Unfortunately I have the same issue.

After that I give you for the 1st device (without any problem).

2nd device:
```
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1240
load:0x40078000,len:13012
load:0x40080400,len:3648
entry 0x400805f8
[⸮⸮mum⸮⸮⸮⸮2-hal-cpu.c:211] setCpuFrequencyMhz(): PLL: 480 / 2 = 240 Mhz, APB: 80000000 Hz
Starting Arduino BLE Client application...
[   695][V][BLEScan.cpp:385] start(): >> start(duration=10)
[   696][V][FreeRTOS.cpp:192] take(): Semaphore taking: name: ScanEnd (0x3ffdff98), owner: <N/A> for start
[   699][V][FreeRTOS.cpp:204] take(): Semaphore taken:  name: ScanEnd (0x3ffdff98), owner: start
[   708][V][BLEScan.cpp:417] start(): << start()
[   712][V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: ScanEnd (0x3ffdff98), owner: start for start
[   711][V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 2 0x02
[   711][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[   738][V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
custom gap event handler, event: 2 
[   756][V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 7 0x07
[   756][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[   765][V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
custom gap event handler, event: 7 
[   783][V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[   783][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[   792][V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[   799][D][BLEAdvertisedDevice.cpp:472] setRSSI(): - setRSSI(): rssi: -90
[   806][V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
[   806][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x01 (), length: 1, data: 1a
[   821][V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xff
[   821][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0xff (), length: 20, data: 7500021811a1f553681f5728ffd074b7a1c2fb29
[   839][D][BLEAdvertisedDevice.cpp:449] setManufacturerData(): - manufacturer data: 7500021811a1f553681f5728ffd074b7a1c2fb29
BLE Advertised Device found: Name: , Address: 5e:d9:8b:c2:7b:f9, manufacturer data: 7500021811a1f553681f5728ffd074b7a1c2fb29
custom gap event handler, event: 3 
[   871][V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[   871][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[   880][V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[   887][D][BLEAdvertisedDevice.cpp:472] setRSSI(): - setRSSI(): rssi: -47
[   894][V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
[   894][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x01 (), length: 1, data: 0a
[   909][V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x7
[   909][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x07 (), length: 16, data: 00c7c44ee36c51a7334be8ed5a0eb803
[   926][D][BLEAdvertisedDevice.cpp:500] setServiceUUID(): - addServiceUUID(): serviceUUID: 03b80e5a-ede8-4b33-a751-6ce34ec4c700
[   938][V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x9
[   938][D][BLEAdvertisedDevice.cpp:292] parseAdvertisement(): Type: 0x09 (), length: 20, data: 4e5558204d494748545920504c5547204d494449
[   955][D][BLEAdvertisedDevice.cpp:461] setName(): - setName(): name: NUX MIGHTY PLUG MIDI
BLE Advertised Device found: Name: NUX MIGHTY PLUG MIDI, Address: cb:4e:fd:d7:02:d5, serviceUUID: 03b80e5a-ede8-4b33-a751-6ce34ec4c700
Found our device!  address: [   984][V][BLEScan.cpp:440] stop(): >> stop()
[   985][V][FreeRTOS.cpp:146] give(): Semaphore giving: name: ScanEnd (0x3ffdff98), owner: start
[   990][V][BLEScan.cpp:452] stop(): << stop()
[   990][V][FreeRTOS.cpp:80] wait(): << wait: Semaphore released: name: ScanEnd (0x3ffdff98), owner: <N/A>
[   994][V][BLEScan.cpp:440] stop(): >> stop()
C_fct: Leaving "Setup" function...
[  1008][V][FreeRTOS.cpp:146] give(): Semaphore giving: name: ScanEnd (0x3ffdff98), owner: <N/A>
C_fct: Start "loop" function...
[  1020][V][BLEScan.cpp:452] stop(): << stop()
custom gap event handler, event: 3 
[  1030][V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[  1030][V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
[  1046][V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[... 301 lines cut ...]
[ 10631][V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 6
[ 10631][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[... cut ...]
[ 10796][V][BLERemoteCharacteristic.cpp:263] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 00002a01-0000-1000-8000-00805f9b34fb
[ 10809][E][BLERemoteCharacteristic.cpp:287] retrieveDescriptors(): esp_ble_gattc_get_all_descr: Unknown
[ 10818][V][BLERemoteCharacteristic.cpp:307] retrieveDescriptors(): << retrieveDescriptors(): Found 0 descriptors.
[ 10828][V][BLERemoteCharacteristic.cpp:46] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[ 10837][V][BLERemoteService.cpp:209] retrieveCharacteristics(): << getCharacteristics()
 - Found our characteristic
(...)
```

JoFRT, 2021-10-16:
Yes it is not the only services : here are the others :
```
I 13:23:05.175 Services discovered (through NRF Connect)
V 13:23:05.180 Generic Access (0x1800)
- Device Name [R W] (0x2A00)
- Peripheral Preferred Connection Parameters [R] (0x2A04)
Unknown Service (00006666-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [E N R SW W WNR] (00008888-0000-1000-8000-00805f9b34fb)
   Client Characteristic Configuration (0x2902)
Unknown Service (00007777-0000-1000-8000-00805f9b34fb)
- Unknown Characteristic [R SW W WNR] (00008877-0000-1000-8000-00805f9b34fb)
Unknown Service (03b80e5a-ede8-4b33-a751-6ce34ec4c700)
- Unknown Characteristic [N R W WNR] (7772e5db-3868-4112-a1a9-f2669d106bf3)
```

Yes I have a second esp32, and there is no problem with the client with this server sketch. I give you the log below.

Because of the perfect connection, not sure that giving you my sketch will help, but I put just below just in case :
```
/**
 * A BLE client example that is rich in capabilities.
 */

#include "BLEAdvertisedDevice.h"
#include "BLEDevice.h"
#include "SerialDebug.h"

//Nux
// Service & Characteristic for switching mode
static BLEUUID serviceUUID("03b80e5a-ede8-4b33-a751-6ce34ec4c700");
static BLEUUID    charUUID("7772e5db-3868-4112-a1a9-f2669d106bf3");

/*
//Itag (black)
static BLEUUID serviceUUID("1800");
static BLEUUID    charUUID("2A00");
*/

static BLEAddress *pServerAddress;
static boolean doConnect = false;
static boolean connected = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
BLEAdvertisedDevice _advertisedDevice;
BLEScan* pBLEScan;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
    Serial.print("Notify callback for characteristic ");
    Serial.print(pBLERemoteCharacteristic->getUUID().toString().c_str());
    Serial.print(" of data length ");
    Serial.println(length);
}

bool connectToServer(BLEAddress pAddress) {
    Serial.println("Wait the end of searching device, 3s..."); 
    delay(3000);
    Serial.println("Wait 3s more");  delay(3000);
    Serial.print("Forming a connection to ");  Serial.println(pAddress.toString().c_str());
    
    BLEClient*  pClient  = BLEDevice::createClient();
    Serial.println(" - Created client");
    
    Serial.println("Wait 1s.");  delay(1000);

    // Connect to the remove BLE Server.
    //Original line:
      ///pClient->connect(pAddress);  
[... 356 lines cut ...]
[ 10111][V][BLERemoteService.cpp:34] BLERemoteService(): << BLERemoteService()
custom gattc event handler, event: 7 
[ 10128][V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 6
[ 10128][D][BLEDevice.cpp:148] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[ 10139][V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 6
[ 10139][V][BLEUtils.cpp:1283] dumpGattClientEvent(): GATT Event: Unknown
[ 10154][V][BLEUtils.cpp:951] gattClientEventTypeToString(): Unknown GATT Client event type: 6
[ 10154][D][BLEClient.cpp:178] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[ 10172][V][FreeRTOS.cpp:146] give(): Semaphore giving: name: SearchCmplEvt (0x3ffdf350), owner: getServices
custom gattc event handler, event: 6 
[ 10181][V][FreeRTOS.cpp:80] wait(): << wait: Semaphore released: name: SearchCmplEvt (0x3ffdf350), owner: <N/A>
[ 10194][V][BLEClient.cpp:465] getServices(): << getServices
[ 10200][V][BLEClient.cpp:425] getService(): << getService: found the service with uuid: 03b80e5a-ede8-4b33-a751-6ce34ec4c700
 - Found our service
[ 10221][V][BLERemoteService.cpp:162] retrieveCharacteristics(): >> getCharacteristics() for service: 03b80e5a-ede8-4b33-a751-6ce34ec4c700
[ 10225][D][BLERemoteService.cpp:193] retrieveCharacteristics(): Found a characteristic: Handle: 42, UUID: 7772e5db-3868-4112-a1a9-f2669d106bf3
[ 10238][V][BLERemoteCharacteristic.cpp:36] BLERemoteCharacteristic(): >> BLERemoteCharacteristic: handle: 42 0x42, uuid: 7772e5db-3868-4112-a1a9-f2669d106bf3
[ 10251][V][BLERemoteCharacteristic.cpp:263] retrieveDescriptors(): >> retrieveDescriptors() for characteristic: 7772e5db-3868-4112-a1a9-f2669d106bf3
[ 10265][D][BLERemoteCharacteristic.cpp:293] retrieveDescriptors(): Found a descriptor: Handle: 43, UUID: 00002902-0000-1000-8000-00805f9b34fb
[ 10277][V][BLERemoteCharacteristic.cpp:307] retrieveDescriptors(): << retrieveDescriptors(): Found 1 descriptors.
[ 10287][V][BLERemoteCharacteristic.cpp:46] BLERemoteCharacteristic(): << BLERemoteCharacteristic
[ 10296][V][BLERemoteService.cpp:209] retrieveCharacteristics(): << getCharacteristics()
 - Found our characteristic 

```

JoFRT, 2021-10-16:
I have just one thing to say: many many many and many thanks to you Chegewara 👍 .
Finally the sketch works perfectly after the addition of one delay.
But let's start by the beginning...
> I have done the changes you suggested, and then I can see that the led of the device informing of bluetooth connection switched on a few seconds later.
> So in my second trial I have enlarged the the delay you precognized to 10s (to be sure but 2-3s is enough).
> Then the miracle appears in front of me: the connection passed!
> I have tried then to modify a characteristic and it works perfectly.
I am wandering if the is a way to monitor when the device will be ready to accept the server connection. But for it is ok like that for me!.
So many thanks to you and also for the others for repling on issues for the #874 item and for the other items.

JpEncausse, 2023-06-27:
Hello, I'm stuck on the same issue with an [ESP-32 TTGO Display](https://fr.aliexpress.com/item/33050639690.html) I try to implement many explanation but without success.

Has I understand the connect() function require a timeout not to hang forever ? But why it is not in standard ? How do I add it ? 

```

bool connected = false;
static BLEAdvertisedDevice* myDevice;

bool createClient(){
  Serial.println("createClient()");
    if (connected){ return true; }

    BLEClient*  pClient  = BLEDevice::createClient();
    pClient->setClientCallbacks(new MyClientCallback());

    if (!pClient->connect(myDevice)){
      Serial.println("Failed to connect to server");
      return false;
    }

    BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr) {
      Serial.print("Failed to find our service UUID");
      pClient->disconnect();
      return false;
    }

    BLERemoteCharacteristic* pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
    if (pRemoteCharacteristic == nullptr) {
      Serial.print("Failed to find our characteristic UUID");
      pClient->disconnect();
      return false;
    }

    if (pRemoteCharacteristic->canNotify()) {
      pRemoteCharacteristic->registerForNotify(notifyCallback);
    } else {
      Serial.println("Failed to register for notifications");
      pClient->disconnect();
      return false;
    }
    
    Serial.println("Connect to server");
    return true;
}

class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      
      if (advertisedDevice.haveServiceUUID()){
        for (size_t i = 0; i < advertisedDevice.getServiceUUIDCount(); i++){
          BLEUUID service = advertisedDevice.getServiceUUID(i);
          if (service.toString() == serviceUUID){
            Serial.println(advertisedDevice.toString().c_str());
            /* read manufacturer data */
            BLEDevice::getScan()->stop();
            myDevice = new BLEAdvertisedDevice(advertisedDevice);
            connected = createClient();
          }
        }
      }
    }
};

  BLEDevice::init("");
  pBLEScan = BLEDevice::getScan(); //create new scan
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true); //active scan uses more power, but get results faster
  pBLEScan->setInterval(100);
  pBLEScan->setWindow(99);  // less or equal setInterval value

```

raydenvoldeskine, 2025-03-15:
Colleagues, I read the whole thread and tried suggestions. 
I have PlatformIO project and I am using BLE lib from espressif repo - BLE lib version 3.x.x (installed manually, as default install installs older version 2.x.x for some reason). 

My connect() call never returns still:

```
   client_ = std::unique_ptr<BLEClient>(BLEDevice::createClient());
    if (!client_->connect(bleDevice_.get())) {
        log("BLEPeripherial: Failed to connect to device.");
        return;
    }
    log("BLEPeripherial: Connected");
```

If i try to modify FreeRTOS semaphore timeouts as suggested after timeout it proceeds but it looks like a "dirty hack" for me - why should it work? So connect proceed (after timeout time) but then getService() fails - I have no idea to know is it because of hack or not. 

I also do stop discovery before connecting. 

Anything I could have missed? I'm quite puzzled because it is a BLE supporting board and discovery etc works perfectly, then why the main connect() method does not work? I'm sure my device is available for connect - at the same time I see it with android BLE app and can connect.

chegewara, 2025-03-17:
As you can see this is very old topic. In general this library is a bit outdated and have some not resolved issues.
It is highly advised to use nimble library which is very easy and straightforward to migrate from this one.