nkolban/esp32-snippets #876: Cannot get any service from device

ASL07, 2019-06-03:
Hi, I am trying to read data from the xiaomi mijia temeprature sensor.

I used to be able to read data before, but now something strange happens and I am not able to get any of the device's services.

I can connect to the device correctly but when I call pXiaomiTempSensorRemoteService = pBleClient->getService(XIAOMI_TEMP_SERVICE_UUID);  
it always returns nullptr.

I post the log of the connection below but I need some help interpreting it.
Thanks

```
.[0;36m[D][BLEDevice.cpp:62] createClient(): >> createClient.[0m
.[0;36m[D][BLEDevice.cpp:68] createClient(): << createClient.[0m

Starting xiaomi_temperature_sensor_read_temperature test
.[0;36m[D][BLEClient.cpp:147] disconnect(): >> disconnect().[0m
.[0;36m[D][BLEClient.cpp:153] disconnect(): << disconnect().[0m
Connecting
.[0;36m[D][BLEClient.cpp:104] connect(): >> connect(4c:65:a8:d4:13:78).[0m
.[0;32m[I][BLEDevice.cpp:596] addPeerDevice(): add conn_id: 0, GATT role: client.[0m
.[0;36m[D][FreeRTOS.cpp:164] take(): Semaphore taking: name: RegEvt (0x3ffded20), owner: <N/A> for connect.[0m
.[0;36m[D][FreeRTOS.cpp:173] take(): Semaphore taken:  name: RegEvt (0x3ffded20), owner: connect.[0m
.[0;37m[V][FreeRTOS.cpp:69] wait(): >> wait: Semaphore waiting: name: RegEvt (0x3ffded20), owner: connect for connect.[0m
.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 0.[0m
.[0;36m[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown.[0m
.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 0.[0m
.[0;37m[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown.[0m
.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 0.[0m
.[0;36m[D][BLEClient.cpp:166] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown.[0m
.[0;37m[V][FreeRTOS.cpp:119] give(): Semaphore giving: name: RegEvt (0x3ffded20), owner: connect.[0m
.[0;37m[V][FreeRTOS.cpp:85] wait(): << wait: Semaphore released: name: RegEvt (0x3ffded20), owner: <N/A>.[0m
.[0;36m[D][FreeRTOS.cpp:164] take(): Semaphore taking: name: OpenEvt (0x3ffdfba8), owner: <N/A> for connect.[0m
.[0;36m[D][FreeRTOS.cpp:173] take(): Semaphore taken:  name: OpenEvt (0x3ffdfba8), owner: connect.[0m
.[0;37m[V][FreeRTOS.cpp:69] wait(): >> wait: Semaphore waiting: name: OpenEvt (0x3ffdfba8), owner: connect for connect.[0m
.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 40.[0m
.[0;36m[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown.[0m
.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 40.[0m
.[0;37m[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown.[0m
.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 40.[0m
.[0;36m[D][BLEClient.cpp:166] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown.[0m
.[0;36m[D][BLEDevice.cpp:580] updatePeerDevice(): update conn_id: 4, GATT role: client.[0m
.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 2.[0m
.[0;36m[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown.[0m
.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 2.[0m
.[0;37m[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown.[0m
.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 2.[0m
.[0;36m[D][BLEClient.cpp:166] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown.[0m
.[0;37m[V][FreeRTOS.cpp:119] give(): Semaphore giving: name: OpenEvt (0x3ffdfba8), owner: connect.[0m
.[0;37m[V][FreeRTOS.cpp:85] wait(): << wait: Semaphore released: name: OpenEvt (0x3ffdfba8), owner: <N/A>.[0m
.[0;36m[D][BLEClient.cpp:137] connect(): << connect(), rc=1.[0m
.[0;36m[D][BLEClient.cpp:147] disconnect(): >> disconnect().[0m
.[0;36m[D][BLEClient.cpp:153] disconnect(): << disconnect().[0m
Address: 4c:65:a8:d4:13:78.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 5.[0m
Get service
.[0;36m[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown.[0m
.[0;36m[D][BLEClient.cpp:384] getService(): >> getService: uuid: 226c0000-6476-4566-7562-66734470666d.[0m
.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 5.[0m
.[0;36m[D][BLEClient.cpp:421] getServices(): >> getServices.[0m
.[0;37m[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown.[0m
.[0;36m[D][BLEClient.cpp:79] clearServices(): >> clearServices.[0m
.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 5.[0m
.[0;36m[D][BLEClient.cpp:86] clearServices(): << clearServices.[0m
.[0;36m[D][BLEClient.cpp:166] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown.[0m
.[0;36m[D][FreeRTOS.cpp:164] take(): Semaphore taking: name: SearchCmplEvt (0x3ffdfc08), owner: <N/A> for getServices.[0m
.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41.[0m
.[0;36m[D][FreeRTOS.cpp:173] take(): Semaphore taken:  name: SearchCmplEvt (0x3ffdfc08), owner: getServices.[0m
.[0;36m[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown.[0m
.[0;37m[V][FreeRTOS.cpp:69] wait(): >> wait: Semaphore waiting: name: SearchCmplEvt (0x3ffdfc08), owner: getServices for getServices.[0m
.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41.[0m
.[0;37m[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown.[0m
.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41.[0m
.[0;36m[D][BLEClient.cpp:166] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown.[0m
.[0;32m[I][BLEDevice.cpp:607] removePeerDevice(): remove: 0, GATT role client.[0m
.[0;37m[V][FreeRTOS.cpp:119] give(): Semaphore giving: name: RssiCmplEvt (0x3ffdfc68), owner: <N/A>.[0m
.[0;37m[V][FreeRTOS.cpp:119] give(): Semaphore giving: name: SearchCmplEvt (0x3ffdfc08), owner: getServices.[0m
.[0;37m[V][FreeRTOS.cpp:85] wait(): << wait: Semaphore released: name: SearchCmplEvt (0x3ffdfc08), owner: <N/A>.[0m
.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 1.[0m
.[0;36m[D][BLEClient.cpp:438] getServices(): << getServices.[0m
.[0;36m[D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown.[0m
.[0;36m[D][BLEClient.cpp:402] getService(): << getService: not found.[0m
.[0;37m[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 1.[0m
.[0;36m[D][BLEClient.cpp:147] disconnect(): >> disconnect().[0m
.[0;37m[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown.[0m
.[0;36m[D][BLEClient.cpp:153] disconnect(): << disconnect().[0m
Temperature: 0.00
.[0;36m[D][BLEClient.cpp:147] disconnect(): >> disconnect().[0m
.[0;36m[D][BLEClient.cpp:153] disconnect(): << disconnect().[0m
```

chegewara, 2019-06-04:
Hi
its because you are having problems with connecting to sensor. You are trying to retrieve services after disconnecting:
```
[D][BLEClient.cpp:147] disconnect(): >> disconnect()
[D][BLEClient.cpp:153] disconnect(): << disconnect()
...
[D][BLEClient.cpp:384] getService(): >> getService: uuid: 226c0000-6476-4566-7562-66734470666d
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 5
[D][BLEClient.cpp:421] getServices(): >> getServices
```

ASL07, 2019-06-10:
My mistake,
@chegewara Thanks for your help, greatly appreciated

closed 2019-06-10 by ASL07