nkolban/esp32-snippets #877: Notifying a string array larger than 20 Bytes

borstan, 2019-06-07:
Hello guys I want to notify my data into 20 bytes chunks so it fits to lowest MTU possible (23). 

I tried to use the code of thread no #798 but he sends only **string** but I want to send an **array** of strings. (accelerometer readings that I connected into one string array - **myConcatenation** and in the android app I will split them by delimiter.)

Here is the code:

    char myConcatenation[] = "";
    sprintf(myConcatenation, 
     "%3.2f,%3.2f,%3.2f|%3.2f,%3.2f,%3.2f|%3.2f,%3.2f,%3.2f|%3.2f,%3.2f,%3.2f|", ax,  ay, az, gx, gy, gz, mx, my, mz, yaw, pitch, roll); //connect yaw, pitch, roll into string
     size_t chars = sizeof(myConcatenation);
     size_t packetSize = 20;

     // assume I have this characteristic already setup
    if (chars >= packetSize) { //added =
    auto packets = buildPackets(myConcatenation, packetSize);
    for (auto packet : packets) {
    pCharacteristic->setValue(packet);
    pCharacteristic->notify();
    }
    }
     else {
    pCharacteristic->setValue(myConcatenation);
    pCharacteristic->notify();
    }

This is how is the function defined: completely by #798:

    std::vector<std::string> buildPackets(std::string data, size_t packetSize) {
    // split up the packet
    size_t partialPacketSize = packetSize - 4;

    // calculate how many packets
    size_t packetCount = data.length() / partialPacketSize;
    packetCount = data.length() % partialPacketSize == 0 ? packetCount : packetCount + 1;

    std::vector<std::string> packets;
    // construct each packet
    for (int i = 0; i < packetCount; i++) {
    // start of packet
    std::string packet = "##";
    packet += (char)packetCount;
    packet += (char)(i + 1);
    std::string part = i == packetCount - 1 ? data.substr(i * partialPacketSize) : data.substr(i * 
    partialPacketSize, partialPacketSize);
    packet.append(part);

    // add to vector of packets
    packets.push_back(packet);
    }

    return packets;
   }

Do you know what changes should I do in the code to achieve what I want? 

Thank you.

dreed-ist, 2019-06-25:
+1 this Question. BTW, it's not the first time I have seen the limit.

chegewara, 2019-06-26:
> +1 this Question. BTW, it's not the first time I have seen the limit.

And you will see it a lot, because its bluetooth LE specification.