nkolban/esp32-snippets #879: POLAR H10 connection problem.

afnan, 2019-06-10:
Hi,
Thank you for providing us with amazing BLE libraries. I am working to connect ESP32 with POLAR H10. The code that I have worked with POLAR H7 but does not seem to work with POLAR H10. I do not know why as the BLE UUIDs are same. Following is my code and log.

Code

`
static void my_gap_event_handler(esp_gap_ble_cb_event_t  event, esp_ble_gap_cb_param_t* param) {
  ESP_LOGW(LOG_TAG, "custom gap event handler, event: %d", (uint8_t)event);
}

static void my_gattc_event_handler(esp_gattc_cb_event_t event, esp_gatt_if_t gattc_if, esp_ble_gattc_cb_param_t* param) {
  ESP_LOGW(LOG_TAG, "custom gattc event handler, event: %d", (uint8_t)event);
}

bool connectToServer(BLEAddress pAddress) {
  Serial.print("Forming a connection to ");
  Serial.println(pAddress.toString().c_str());

  pClient  = BLEDevice::createClient();
  Serial.println(" - Created client");

  // Connect to the remove BLE Server.
  pClient->connect(pAddress, BLE_ADDR_TYPE_RANDOM);
  Serial.println(" - Connected to server");
  /**************CONNECT TO HR SERVICE******************/
  // Obtain a reference to the HR service of the remote BLE server.
  BLERemoteService* pRemote_HR_Service = pClient->getService(service_HR_UUID);

  // Obtain a reference to the BATT service of the remote BLE server.
  BLERemoteService* pRemote_BATT_Service = pClient->getService(service_BATT_UUID);

  // Obtain a reference to the HR characteristic in the service of the remote BLE server.
  pRemote_HR_Characteristic = pRemote_HR_Service->getCharacteristic(char_HR_UUID);

  // Obtain a reference to the BATT characteristic in the service of the remote BLE server.
  pRemote_BATT_Characteristic = pRemote_BATT_Service->getCharacteristic(char_BATT_UUID);

  // Set notification
  pRemote_HR_Characteristic->registerForNotify(notifyCallback);
  return true;
}

/**
   Scan for BLE servers and find the first one that advertises the service we are looking for.
*/
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    /**
        Called for each advertising BLE server.
    */
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.print("BLE Advertised Device found: ");
      Serial.println(advertisedDevice.toString().c_str());

      // We have found a device, let us now see if it contains the service we are looking for.
      if (advertisedDevice.haveServiceUUID() && advertisedDevice.getServiceUUID().equals(service_HR_UUID)) {
        Serial.println("Found our device!  Stop Scanning: ");
        advertisedDevice.getScan()->stop();
        Serial.print("Found our device!  address: ");
        Serial.println(advertisedDevice.getAddress().toString().c_str());
        pServerAddress = new BLEAddress(advertisedDevice.getAddress());
        Serial.println("Obtained the Server Object");
        doConnect = true;
        return;
      } // Found our server
    } // onResult
}; // MyAdvertisedDeviceCallbacks

void BLESetup() {
  BLEDevice::init("");
  BLEDevice::setCustomGapHandler(my_gap_event_handler);
[... 468 lines cut ...]
[V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[W][BLEHelper.ino:3] my_gap_event_handler(): custom gap event handler, event: 3
[V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[W][BLEHelper.ino:3] my_gap_event_handler(): custom gap event handler, event: 3
[V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[W][BLEHelper.ino:3] my_gap_event_handler(): custom gap event handler, event: 3
[V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[W][BLEHelper.ino:3] my_gap_event_handler(): custom gap event handler, event: 3
[V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 3 0x03
[V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[W][BLEHelper.ino:3] my_gap_event_handler(): custom gap event handler, event: 3
[V][BLEUtils.cpp:1818] gapEventToString(): gapEventToString: Unknown event type 18 0x12
[V][BLEUtils.cpp:1050] dumpGapEvent(): Received a GAP event: Unknown event type
[V][BLEUtils.cpp:1265] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
[W][BLEHelper.ino:3] my_gap_event_handler(): custom gap event handler, event: 18
`

chegewara, 2019-06-10:
This is last useful info from your logs:
`Serial.println("Obtained the Server Object");`
All logs after that are:
`ESP_GAP_BLE_SCAN_RESULT_EVT` == 0x03 (3)
or
`ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT` == 0x12(18)

There is no info you even are trying to connect your device.

Besides that, there may be option (i dont know) that H10 is requiring pairing/authenticating and H7 dont.

afnan, 2019-06-10:
> This is last useful info from your logs:
> `Serial.println("Obtained the Server Object");`
> All logs after that are:
> `ESP_GAP_BLE_SCAN_RESULT_EVT` == 0x03 (3)
> or
> `ESP_GAP_BLE_SCAN_STOP_COMPLETE_EVT` == [...]

Yes you are correct. I am unable to figure out why `Serial.println("Scan Stopped.");` line is never called.  after `setAdvertisedDeviceCallbacks` is set. 

What could be the good approach to move forward to debug it

chegewara, 2019-06-10:
This whole function is strange:
```
void BLESetup() {
  BLEDevice::init("");
  BLEDevice::setCustomGapHandler(my_gap_event_handler);
  BLEDevice::setCustomGattcHandler(my_gattc_event_handler);
  // Retrieve a Scanner and set the callback we want to use to be informed when we
  // have detected a new device.  Specify that we want active scanning and start the
  // scan to run for 30 seconds.
  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setActiveScan(true);
  Serial.println("Scan Stopped.");
  pBLEScan->start(30);

  // If the flag "doConnect" is true then we have scanned for and found the desired
  // BLE Server with which we wish to connect.  Now we connect to it.  Once we are
  // connected we set the connected flag to be true.
  if (doConnect == true) {
    Serial.println("Attempting to establish connection to the server.");
    if (connectToServer(*pServerAddress)) {
      Serial.println("We are now connected to the BLE Server.");
      connected = true;
    } else {
      Serial.println("We have failed to connect to the server; there is nothin more we will do.");
    }
    doConnect = false;
  }
}
```
It looks like it is and is not called from loop.
This code should be in loop:
```
if (doConnect == true) {
    Serial.println("Attempting to establish connection to the server.");
    if (connectToServer(*pServerAddress)) {
      Serial.println("We are now connected to the BLE Server.");
      connected = true;
    } else {
      Serial.println("We have failed to connect to the server; there is nothin more we will do.");
    }
    doConnect = false;
  }
```
and all before that cant be in loop (maybe except `pBLEScan->start(30);`, but not straightforward).

afnan, 2019-06-10:
> This whole function is strange:
>
> ```
> void BLESetup() {
>   BLEDevice::init("");
>   BLEDevice::setCustomGapHandler(my_gap_event_handler);
>   BLEDevice::setCustomGattcHandler(my_gattc_event_handler);
>   // [...]

I upgraded https://github.com/SensorsIot/Bluetooth-BLE-on-Arduino-IDE/blob/master/Polar_Receiver/Polar_Receiver.ino code into more functions and thing is `doConnect` is true but the program never makes it to that point. 

I copied the code from the link and tried and has this dump

afnan, 2019-06-10:
Finally This code works and I cant know WHY? by the way its your code I found at another post.
```

/**
   A BLE client example that is rich in capabilities.
   There is a lot new capabilities implemented.
   author unknown
   updated by chegewara
*/

#include "BLEDevice.h"
//#include "BLEScan.h"

// The remote service we wish to connect to.
static BLEUUID serviceUUID(BLEUUID((uint16_t)0x180D));
// The characteristic of the remote service we are interested in.
static BLEUUID    charUUID(BLEUUID((uint16_t)0x2A37));

static boolean doConnect = false;
static boolean connected = false;
static boolean doScan = false;
static BLERemoteCharacteristic* pRemoteCharacteristic;
static BLEAdvertisedDevice* myDevice;

static void notifyCallback(
  BLERemoteCharacteristic* pBLERemoteCharacteristic,
  uint8_t* pData,
  size_t length,
  bool isNotify) {
  Serial.print("Notify callback for characteristic ");
//  for (int i = 0; i < length; i++) {
//    Serial.print(pData[i]);
//    Serial.print(" ");
//  }
//  Serial.println();

  uint8_t HR_val;
  if (pData[0] == 16) {
    if (pData[1] > 10 && pData[1] < 254) {                    // HR must be between 10 and 254 bpm
      HR_val = pData[1];
      //HR_contact = true;
    } else HR_val = 0;
  } else {
    HR_val = 0;
    //HR_contact = false;
  }
  Serial.println(HR_val);
}

class MyClientCallback : public BLEClientCallbacks {
    void onConnect(BLEClient* pclient) {
      connected = true;
      Serial.println("onConnect");
    }

    void onDisconnect(BLEClient* pclient) {
      connected = false;
      Serial.println("onDisconnect");
    }
};

bool connectToServer() {
  Serial.print("Forming a connection to ");
  Serial.println(myDevice->getAddress().toString().c_str());

  BLEClient*  pClient  = BLEDevice::createClient();
  Serial.println(" - Created client");

  pClient->setClientCallbacks(new MyClientCallback());

  // Connect to the remove BLE Server.
  pClient->connect(myDevice);  // if you pass BLEAdvertisedDevice instead of address, it will be recognized type of peer device address (public or private)
  Serial.println(" - Connected to server");

  // Obtain a reference to the service we are after in the remote BLE server.
  BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
  if (pRemoteService == nullptr) {
    Serial.print("Failed to find our service UUID: ");
    Serial.println(serviceUUID.toString().c_str());
    pClient->disconnect();
    return false;
  }
  Serial.println(" - Found our service");

  // Obtain a reference to the characteristic in the service of the remote BLE server.
  pRemoteCharacteristic = pRemoteService->getCharacteristic(charUUID);
  if (pRemoteCharacteristic == nullptr) {
    Serial.print("Failed to find our characteristic UUID: ");
    Serial.println(charUUID.toString().c_str());
    pClient->disconnect();
    return false;
  }
  Serial.println(" - Found our characteristic");

  // Read the value of the characteristic.
  if (pRemoteCharacteristic->canRead()) {
    std::string value = pRemoteCharacteristic->readValue();
    Serial.print("The characteristic value was: ");
    Serial.println(value.c_str());
  }

  if (pRemoteCharacteristic->canNotify())
    pRemoteCharacteristic->registerForNotify(notifyCallback);

  connected = true;
  return true;
}
/**
   Scan for BLE servers and find the first one that advertises the service we are looking for.
*/
class MyAdvertisedDeviceCallbacks: public BLEAdvertisedDeviceCallbacks {
    /**
        Called for each advertising BLE server.
    */
    void onResult(BLEAdvertisedDevice advertisedDevice) {
      Serial.print("BLE Advertised Device found: ");
      Serial.println(advertisedDevice.toString().c_str());

      // We have found a device, let us now see if it contains the service we are looking for.
      if (advertisedDevice.haveServiceUUID() && advertisedDevice.isAdvertisingService(serviceUUID)) {

        BLEDevice::getScan()->stop();
        myDevice = new BLEAdvertisedDevice(advertisedDevice);
        doConnect = true;
        doScan = true;

      } // Found our server
    } // onResult
}; // MyAdvertisedDeviceCallbacks

void setup() {
  Serial.begin(115200);
  Serial.println("Starting Arduino BLE Client application...");
  BLEDevice::init("");

  // Retrieve a Scanner and set the callback we want to use to be informed when we
  // have detected a new device.  Specify that we want active scanning and start the
  // scan to run for 5 seconds.
  BLEScan* pBLEScan = BLEDevice::getScan();
  pBLEScan->setAdvertisedDeviceCallbacks(new MyAdvertisedDeviceCallbacks());
  pBLEScan->setInterval(1349);
  pBLEScan->setWindow(449);
  pBLEScan->setActiveScan(true);
  pBLEScan->start(5, false);
} // End of setup.

// This is the Arduino main loop function.
void loop() {

  // If the flag "doConnect" is true then we have scanned for and found the desired
  // BLE Server with which we wish to connect.  Now we connect to it.  Once we are
  // connected we set the connected flag to be true.
  if (doConnect == true) {
    if (connectToServer()) {
      Serial.println("We are now connected to the BLE Server.");
    } else {
      Serial.println("We have failed to connect to the server; there is nothin more we will do.");
    }
    doConnect = false;
  }

  // If we are connected to a peer BLE Server, update the characteristic each time we are reached
  // with the current time since boot.
  if (connected) {
    String newValue = "Time since boot: " + String(millis() / 1000);
    Serial.println("Setting new characteristic value to \"" + newValue + "\"");

    // Set the characteristic's value to be the array of bytes that is actually a string.
    pRemoteCharacteristic->writeValue(newValue.c_str(), newValue.length());
  } else if (doScan) {
    BLEDevice::getScan()->start(0);  // this is just eample to start scan after disconnect, most likely there is better way to do it in arduino
  }

  delay(1000); // Delay a second between loops.
} // End of loop
```

chegewara, 2019-06-10:
There may be many reasons. You have to check logs and compare with this list:
https://github.com/chegewara/esp32-ble-wiki/issues/5

afnan, 2019-06-11:
> There may be many reasons. You have to check logs and compare with this list:
> [chegewara/esp32-ble-wiki#5](https://github.com/chegewara/esp32-ble-wiki/issues/5)

I have nailed it down to `pServerAddress`. If I connect to it does not work but if i go with your approach and pass myDevice object it works.

chegewara, 2019-06-11:
Im not sure it was like that from beginning or it was changed at some point in esp-idf, but devices with RPA it is required to pass what type of address it is (with public address probably too, but it is default value).

closed 2019-06-12 by afnan