nkolban/esp32-snippets #882: BLEDevice.h: No such file or directory

pana216, 2019-06-12:
Hey guys,

I just installed the eclipse IDE and make it work with c++. Now I want to test the SampleServer.cpp example and running into the following error: 

[...]
.../SampleServer.cpp:4:23: fatal error: BLEDevice.h: No such file or directory
compilation terminated.
/home/pana/esp/esp-idf/make/component_wrapper.mk:289: recipe for target 'SampleServer.o' failed
make[1]: *** [SampleServer.o] Error 1
/home/pana/esp/esp-idf/make/project.mk:582: recipe for target 'component-main-build' failed
make: *** [component-main-build] Error 2

Using Eclipse IDE und linux virtual toolbox with an ESP32 DEV-C Kit v4.
Copied the main.cpp and the SampleServer.cpp in the project folder /main and then run make menuconfig saved it and then make flash monitor resolving to this error.

Cheers and thank you in advance!

pana216, 2019-06-12:
Okay sometimes it's useful to read the whole information given:
[https://github.com/nkolban/esp32-snippets/tree/master/cpp_utils#compiling-the-c-classes](url)
Issue closed

closed 2019-06-12 by pana216

rexzz, 2019-08-13:
ty. if it had been a snake it would have bit me.