nkolban/esp32-snippets #883: How can I manually start and stop advertising?

TSOgang, 2019-06-12:
Hello all, 
I wish to know if it is possible to manually stop advertising a BLE server (without a connection being established). I can start advertising by using:
```
BLEDevice::startAdvertising();
```

In my code it looks like this:

```
BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
pAdvertising->addServiceUUID(PAIRING_SERVICE_UUID);
BLEDevice::startAdvertising();
```
This works, but I haven't found a way to stop it without actually connecting to the device. I have tried this function:
```
pAdvertising->stop();
```
But no success. Can anyone help?
Thanks in advance.

chegewara, 2019-06-12:
Hi, 
there is small issue in library that is causing autostart advertising. Delete this line:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEAdvertising.cpp#L496

TSOgang, 2019-06-12:
Thank you for the fast response.
I deleted the line, but the issue still persists. In case you want the full sketch, here it is (it's a VERY lightly modified version of your server sketch):

```
#define PAIRING_SERVICE_UUID "aa11aa11-bb22-cc33-dd44-ee55ee55ee55"
#define CHARACTERISTIC_UUID  "beb5483e-36e1-4688-b7f5-ea07361b26a8"

class MyServerCallback : public BLEServerCallbacks {
  void onConnect(BLEServer* pserver) {
    Serial.println("Connected");
  }

  void onDisconnect(BLEServer* pserver) {
    Serial.println("Disconnected");
  }
};

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("");
  BLEServer *pServer = BLEDevice::createServer();
  BLEService *pService = pServer->createService(PAIRING_SERVICE_UUID);
  BLECharacteristic *pCharacteristic = pService->createCharacteristic(
                                         CHARACTERISTIC_UUID,
                                         BLECharacteristic::PROPERTY_READ |
                                         BLECharacteristic::PROPERTY_WRITE
                                       );

  uint16_t tempData = 0xAA;
  pCharacteristic->setValue(tempData);

  pService->start();
  // BLEAdvertising *pAdvertising = pServer->getAdvertising();  // this still is working for backward compatibility
  BLEAdvertising *pAdvertising = BLEDevice::getAdvertising();
  pAdvertising->addServiceUUID(PAIRING_SERVICE_UUID);
  pAdvertising->setScanResponse(true);
  pAdvertising->setMinPreferred(0x06);  // functions that help with iPhone connections issue
  pAdvertising->setMinPreferred(0x12);
  BLEDevice::startAdvertising();
  Serial.println("Characteristic defined!");

  //delay(3000);

  pAdvertising->stop();
  BLEDevice::getAdvertising()->stop();

  Serial.println("Adv stopped?");

  pServer->setCallbacks(new MyServerCallback());
}

void loop() {
  // put your main code here, to run repeatedly:
  delay(2000);
}
```

chegewara, 2019-06-13:
You can use only one of them:
```
 pAdvertising->stop();
  BLEDevice::getAdvertising()->stop();
```
but even if you use both it shouldnt cause any issue in app.

TSOgang, 2019-06-17:
Thanks a lot. I think I got it to work. On another note, I have a small question to ask. It's about the 
"getValue()" function in BLECharacteristic (server side). From what I can see it is possible to write different types of values in the characteristic:
```
	void setValue(uint8_t* data, size_t size);
	void setValue(std::string value);
	void setValue(uint16_t& data16);
	void setValue(uint32_t& data32);
	void setValue(int& data32);
	void setValue(float& data32);
	void setValue(double& data64);
``` 
However, getValue() can only return the value of a characteristic as an "std::string" type and nothing else : 
```
std::string    getValue();
```
If I want to read this value as a hex value (for example), I could try to create a function that converts this string into a hex, but if there are other simpler solutions, I could really use it.

Thanks in advance.

chegewara, 2019-06-17:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLECharacteristic.cpp#L191

TSOgang, 2019-06-18:
Thanks! it was a lot simpler than I thought. I will close this issue now.

closed 2019-06-18 by TSOgang