nkolban/esp32-snippets #886: Sending responses from BLE client to server.

DCooper-nz, 2019-06-20:
I am wanting to have some two way communication between 2 esp32s using ble. I have been able to run the Uart code on one esp32 operating as a server and have another receive data connected as a client. I need the client to be able to send responses to the server. I can have two way communication between the Uart server and my phone just fine but I don't how to either, modify the Uart code to connect as a client to another esp32 or modify the client code to allow it to send responses back to the server. Has anyone done this and could point to me to some code I could learn from?

Cheers
Dave

DCooper-nz, 2019-06-20:
I should add more detail, I can have the server send data to the client via a 'notify' and the client can send to the server with 'writeValue' and that all works great but how do you do both. Obviously the Uart example has separate characteristics for sending and receiving, is it possible to do the same in the client example. It seems the code is somewhat different for each in how it sets up the characteristics so I am not sure how to modify the client to add another one if its even possible. I don't require data going in both directions simultaneously, its more an command and acknowledge deal so is it possible to send switch between sending and receiving on the same characteristic?

chegewara, 2019-06-20:
Hi,
using UART over BLE is not very intuitive. With normal UART you have cross connected lines. In BLE it looks something like this:
```
           UUID(6E400002)  RX ------------------ TX  UUID(6E400002)
server                                                                 client
           UUID(6E400003)  TX ------------------ RX  UUID(6E400003)
```

DCooper-nz, 2019-06-20:
I have that bit sorted, its coding the second characteristic into the client code. The Uart and client examples are different in the way the characteristics are set up and its not obvious to me how to add a second characteristic into the client example. I had just hoped someone might have an example where they had already done this.

chegewara, 2019-06-20:
```
BLERemoteService* pRemoteService = pClient->getService(serviceUUID);
    if (pRemoteService == nullptr) {
      Serial.print("Failed to find our service UUID: ");
      Serial.println(serviceUUID.toString().c_str());
      return false;
    }
    Serial.println(" - Found our service");

    // Obtain a reference to the characteristic in the service of the remote BLE server.
    pRemoteCharacteristic1 = pRemoteService->getCharacteristic(charUUID1);

    // Obtain a reference to the characteristic in the service of the remote BLE server.
    pRemoteCharacteristic2 = pRemoteService->getCharacteristic(charUUID2);
    
```

DCooper-nz, 2019-06-21:
Thanks, so it just that easy, I guess I was thinking there must be more to it than that. 

Cheers
Dave