nkolban/esp32-snippets #888: Is there a way to remove the ble bonding info - ESP32 in ArduinoIDE ?

elias-hatami, 2019-08-05:
Hello
Is there a way to remove the ble bonding info in ArduinoIDE ?
I have a little experience with esp32.
I want a short and complete example.
Thank you
--------------------------------------------------
//example my code//

content2 = Serial.readStringUntil('\r');
if (content2=="RRR"){

??? remove bonded devices ???

}

-------------------------------------------------

chegewara, 2019-08-09:
I dont think there is any API in this library, you will have to use esp-idf functions to do it.

elias-hatami, 2019-08-10:
Is there a short and complete example of this on esp-idf?
I would like to follow the "if" in the example above

Thank you

chegewara, 2019-08-10:
https://github.com/espressif/esp-idf/blob/6517c5033aab27afe491e433d6beeba93dfb1aa7/examples/bluetooth/bluedroid/ble/gatt_security_server/main/example_ble_sec_gatts_demo.c#L273-L284

elias-hatami, 2019-08-11:
I couldn't use this method to clear the memory address (binding mode) or connection address (connection mode) !!!

chegewara, 2019-08-11:
Why?

elias-hatami, 2019-08-11:
I actually want to unpair from the destination device.

thanks again

elias-hatami, 2019-08-11:
Is it possible to do this?

Rob58329, 2025-01-28:
If anyone want's to know how to list and/or remove/delete a BLE Bond/Pair in Arduino, see https://github.com/espressif/arduino-esp32/issues/1458