nkolban/esp32-snippets #889: Versions of esp32-snippets and ESP-IDF

BobaTrek, 2019-08-06:
What is the version compatibility of esp32-snippets to versions of ESP-IDF?

I downloaded the Master from the esp-idf github and get numerous compilation errors when I compiled it with the Master from esp32-snippets, mostly regarding ringbuf_type_t being deprecated.

esp-idf has releases 3.0. 3.1, 3.2, and 3.3.  The Master is currently version 4.0-dev.

Which work with esp32-snippets?

(Someday it would be great if this code base was merged into esp-idf!!!)

chegewara, 2019-08-09:
This library should work with all esp-idf v3.x i think. Library has not been updated for some time, so its possible it wont work with esp-idf master.

mws-rmain, 2019-10-15:
esp-idf commit [8a597b8](https://github.com/espressif/esp-idf/commit/8a597b865f45a4161a0df775276b195d4d9825a2#diff-bf807558c71a6798e443f9772c2be955) changed 'ringbuf_type_t' to 'RingbufferType_t'.
Make this change in FreeRTOS.h:61 & FreeRTOS.cpp:232 and esp32-snippets should compile.

jmporter34, 2022-05-02:
@mws-rmain Thanks for the 'ringbuf_type_t' tip!