nkolban/esp32-snippets #890: ESP32 example BT_DISCOVERY has error ESP_ERR_INVALID_ARG

rexzz, 2019-08-15:
1. trying to run copy of esp-idf\examples\bluetooth\bt_discovery. 

NOW WORKING AFTER I manually edited SDK.  BUT this is a note if others trying this out or if someone wants to update "README" for bt_discovery.(Maybe  mention editing SDK).  AND IF SOMEONE wants to look at my SDK changes and tell me if something is wrong, I'd appreciate.  
1. trying to run copy of esp-idf\examples\bluetooth\bt_discovery.  Using idf.py.  menuconfig and build seem to run.
2. I get error "app_main enable controller failed: ESP_ERR_INVALID_ARG"
3. BT choices in menuconfig in two pictures
4. "Menuconfig" presents slighty different choices than those in the bt_discovery\README.rst.
   (nowhere can I see where to not choose "release DRAM from classic BT Controller" as README suggests)
5. so I look at esp-dif\examples\bluetooth\bt_discover\sdkconfig.defaults
and found which was different than what menuconfig had generated.
CONFIG_BT_ENABLED=y
CONFIG_BTDM_CONTROLLER_MODE_BLE_ONLY=   ns
CONFIG_BTDM_CONTROLLER_MODE_BR_EDR_ONLY=y 
CONFIG_BTDM_CONTROLLER_MODE_BTDM=
CONFIG_CLASSIC_BT_ENABLED=y  
CONFIG_A2DP_ENABLE=n     
CONFIG_GATTS_ENABLE=n   
CONFIG_GATTC_ENABLE=n   
CONFIG_BLE_SMP_ENABLE=n   

6.  So I changed SDK for my copy. Most of the relvant bits are below.

CONFIG_BT_ENABLED=y
# CONFIG_BTDM_CONTROLLER_MODE_BLE_ONLY is not set
CONFIG_BTDM_CONTROLLER_MODE_BR_EDR_ONLY=y
# CONFIG_BTDM_CONTROLLER_MODE_BTDM is not set
CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN=2
CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_SYNC_CONN=0
CONFIG_BTDM_CONTROLLER_BLE_MAX_CONN_EFF=0
CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_ACL_CONN_EFF=2
CONFIG_BTDM_CONTROLLER_BR_EDR_MAX_SYNC_CONN_EFF=0
CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE_0=y
# CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE_1 is not set
CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE=0
CONFIG_BTDM_CONTROLLER_HCI_MODE_VHCI=y
# CONFIG_BTDM_CONTROLLER_HCI_MODE_UART_H4 is not set
CONFIG_BTDM_CONTROLLER_MODEM_SLEEP=y
CONFIG_BTDM_MODEM_SLEEP_MODE_ORIG=y
# CONFIG_BTDM_MODEM_SLEEP_MODE_EVED is not set
CONFIG_BTDM_LPCLK_SEL_MAIN_XTAL=y
CONFIG_BLUEDROID_ENABLED=y
CONFIG_BLUEDROID_PINNED_TO_CORE_0=y
# CONFIG_BLUEDROID_PINNED_TO_CORE_1 is not set
CONFIG_BLUEDROID_PINNED_TO_CORE=0
CONFIG_BTC_TASK_STACK_SIZE=8000
# CONFIG_BLUEDROID_MEM_DEBUG is not set
CONFIG_CLASSIC_BT_ENABLED=y
# CONFIG_A2DP_ENABLE is not set
CONFIG_BT_SPP_ENABLED=y

    
[image]
[image]
[image]

chegewara, 2019-08-15:
@rexzz You want to use classic BT, but this library is to work with BLE. So what is your problem?

chegewara, 2019-08-15:
I think that in esp-idf version you are using some menuconfig values changed its names. Most likely this one:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEDevice.cpp#L350
`CLASSIC_BT_ENABLED` is now `CONFIG_CLASSIC_BT_ENABLED`

rexzz, 2019-08-21:
thanks chegewara.  You are right. I realize the library is 95% BLE. Still that fragment for was for classic BT and I got it to work (after tweaking a few things).

hassance3, 2020-05-05:
@rexzz it will be helpful if you show us what things you tweak to make it work!.

rexzz, 2020-05-24:
hi @hassance3.  sorry not to respond earlier.  give me a couple weeks to find what I did last year. it's definitely NOT pretty.

rexzz, 2020-06-21:
hey @hassance3 - I'm still not sure what I was up to last year. (I found something but I couldn't get it to work). Meanwhile take a look at https://github.com/AntorFr/ClassicBTScan .  I was following him last year. I notice he updated recently.  I haven't tried as I've gone in 'a different direction'

hassance3, 2020-06-22:
@rexzz Thank you, it's work, my building script copy the wrong sdkconfig, so after cloning new IDF, using command line to build it's work, for Classic and BLE.

Thank you.