nkolban/esp32-snippets #897: BLE reconect problem after 9-10 min connection

SwargHUB, 2019-09-02:
Hello. Thanks for the awesome library.
I have encountered such a problem. My device (ble client) connects to the ble server once every 5 seconds, the rest of the time it sleeps. Each time before going to bed, I disconnect, and after waking up I call the connection.
After 9-10 minutes of work, my application freezes. It only responds to interrupts.
What could be the problem?
_Sleep replaced by delay (5000), the problem remains._

`BLEClient*  pClient;    
static BLEAdvertisedDevice* myDevice;        
boolean connected;
...
/*When first connecting*/
pClient  = BLEDevice::createClient();
                        
...
/*In the start Loop*/
pClient->connect(myDevice);
...
/*In the end Loop*/
 if (connected) pClient->disconnect(); 
 esp_sleep_enable_timer_wakeup(5000000); 
 esp_light_sleep_start(); `
...

chegewara, 2019-09-03:
Hi,
it all depends which library you are using. Few weeks ago i posted PR in arduino-esp32 version that should fix problem with semaphores(its not updated in this repo), which usually caused app stuck.

It can be also problem with memory leak, but without logs its hard to say.

SwargHUB, 2019-09-04:
Thanks for the quick response.
The application freezes immediately after disconnect. There were no backtraces. Are there additional logging systems that can help track this problem?

Where can I find the updated library?

SwargHUB, 2019-09-04:
Thanks for the quick response
Where can I find the updated library?

Advanced log:
```
Free bytes on the heap: 162168
@[D][BLEClient.cpp:103] connect(): >> connect(cc:78:ab:13:5f:80)
[I][BLEDevice.cpp:600] addPeerDevice(): add conn_id: 28, GATT role: client
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: RegEvt (0x3ffe0720), owner: <N/A> for connect
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: RegEvt (0x3ffe0720), owner: connect
[V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: RegEvt (0x3ffe0720), owner: connect for connect
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 0
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: RegEvt (0x3ffe0720), owner: connect
[V][FreeRTOS.cpp:86] wait(): << wait: Semaphore released: name: RegEvt (0x3ffe0720), owner: connect
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: OpenEvt (0x3ffe0a30), owner: <N/A> for connect
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: OpenEvt (0x3ffe0a30), owner: connect
[V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: OpenEvt (0x3ffe0a30), owner: connect for connect
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 40
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 40
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 40
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:584] updatePeerDevice(): update conn_id: 4, GATT role: client
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 2
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 2
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 2
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: OpenEvt (0x3ffe0a30), owner: connect
[V][FreeRTOS.cpp:86] wait(): << wait: Semaphore released: name: OpenEvt (0x3ffe0a30), owner: <N/A>
[D][BLEClient.cpp:136] connect(): << connect(), rc=1
Read and Parse Answer
[D][BLERemoteCharacteristic.cpp:404] readValue(): >> readValue(): uuid: f0001131-0451-4000-b000-000000000000, handle: 30 0x1e
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: ReadCharEvt (0x3ffe22e0), owner: <N/A> for readValue
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: ReadCharEvt (0x3ffe22e0), owner: readValue
[V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: ReadCharEvt (0x3ffe22e0), owner: readValue for readValue
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 18
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 18
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 18
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 3
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 3
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 3
[D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][FreeRTOS.cpp:120] give(): Semaphore giving: name: ReadCharEvt (0x3ffe22e0), owner: readValue
[V][FreeRTOS.cpp:86] wait(): << wait: Semaphore released: name: ReadCharEvt (0x3ffe22e0), owner: readValue
[D][BLERemoteCharacteristic.cpp:432] readValue(): << readValue(): length: 22
331(time in seconds from the start);Request params
[D][BLERemoteCharacteristic.cpp:550] writeValue(): >> writeValue(), length: 6
[D][FreeRTOS.cpp:165] take(): Semaphore taking: name: WriteCharEvt (0x3ffe367c), owner: <N/A> for writeValue
[D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: WriteCharEvt (0x3ffe367c), owner: writeValue
[V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: WriteCharEvt (0x3ffe367c), owner: writeValue for writeValue
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 4
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[... cut ...]
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 5
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 1
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 1
[V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
```

chegewara, 2019-09-04:
Updated library is now part of arduino-esp32 v1.0.3:
https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE

chegewara, 2019-09-04:
Im not sure about light sleep together with BLE, maybe @me-no-dev can help with it.

SwargHUB, 2019-09-05:
Thank's.
No problem sleeping. I replace sleep with delay, the problem remains