nkolban/esp32-snippets #898: How to Read setValue(std::string value) on the BLE receiver?

pana216, 2019-09-02:
Hello there,

I am notifying my client (iOS App) for a new value on the characteristic, I set with the setValue(std::string value) method.

I have a u32int value voltage1 and convert it into a string:
`std::string Voltage1AsString((char*)&voltage1, 4);`

Then set the value for the characteristic:
`Signal1_Characteristic->setValue(Voltage1AsString);`

How am I supposed to "convert" or read the characteristic value on the client side? 
I do not expect swift code, but a general way on how to handle this data would be nice, since I am very nooby at c/c++

Thank you in advance!

chegewara, 2019-09-03:
Hi @pana216 
you can send value in 2 different ways:
1. send raw integer value `Signal1_Characteristic->setValue((uint8_t*)&voltage1, 4);`
2. send as string, its similar way you are doing it
Then on client side you will receive:
1. 4 bytes representing integer value, you have to do nothing but read it and eventually change bytes order (but most likely just read it)
2. use https://en.cppreference.com/w/cpp/string/byte/atoi

pana216, 2019-09-05:
With the help of stackoverflow, google and your answer I came to following conclusion (for everybody who works with iOS as client, a must read, makes things very easy):

Go to the `didUpdateValueFor characteristic` method from CoreBLE and get the characteristic.value by 
`guard let characteristicData = characteristic.value else {return -1}`
`let byteArray = [UInt8](characteristicData)`

now if you print the byteArray depending on what value you set to the characteristic in this case we say 'Signal1_Characterisitc->setValue("1840")` you recieve an array of bytes (here [48,7,0,0])

and with the following calculation we can get the Int value back:
`let value = (Int(byteArray[0]) << 0)  + (Int(byteArray[1]) << 8)
    + (Int(byteArray[2]) << 16)
    + (Int(byteArray[3]) << 24)`

closed 2019-09-05 by pana216