nkolban/esp32-snippets #899: error:undefined reference to `app_main'

SophistCedar, 2019-09-04:
system: window10.  compiler: mingw32,
i want to test the demo mpu6050.c program in my esp32board whose directory is esp32-snippets/hardware/accelerometers/mpu6050.c

Firstly, i have compiled the official hello_world program successfully and the result is as follows:
AR build/xtensa-debug-module/libxtensa-debug-module.a
LD build/hello-world.elf
esptool.py v2.6
To flash all build output, run 'make flash' or:
python /d/ESP32/esp-idf-v3.2.2/components/esptool_py/esptool/esptool.py --chip esp32 --port COM1 --baud 115200 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 40m --flash_size detect 0x1000 /C/Users/Administrator/esp/hello_world/build/bootloader/bootloader.bin 0x10000 /C/Users/Administrator/esp/hello_world/build/hello-world.bin 0x8000 /C/Users/Administrator/esp/hello_world/build/partitions_singleapp.bin

secondly, I just replace the whole code in the hello_world_main.c file with the complete code in above directory and an error occurs to me: 
AR build/main/libmain.a
LD build/hello-world.elf
C:/Users/Administrator/esp/hello_world/build/esp32\libesp32.a(cpu_start.o):(.lit                                                                                                                eral.main_task+0x18): undefined reference to `app_main'
C:/Users/Administrator/esp/hello_world/build/esp32\libesp32.a(cpu_start.o): In f                                                                                                                unction `main_task':
D:/ESP32/esp-idf-v3.2.2/components/esp32/cpu_start.c:489: undefined reference to                                                                                                                 `app_main'
collect2.exe: error: ld returned 1 exit status
make: *** [D:/ESP32/esp-idf-v3.2.2/make/project.mk:461：/C/Users/Administrator/e                                                                                                                sp/hello_world/build/hello-world.elf] 错误 1

the main problem is undefined reference to `app_main' and I also refer to another same problem whose cause is not copying the component.mk file into the main subfolder, but the component.mk has existed in the main subfolder.

I donnot know how to deal with it, plz help me , thanks in advance.

chegewara, 2019-09-04:
` undefined reference to `app_main'`
https://github.com/nkolban/esp32-snippets/blob/master/skeletons/cpp/main_helloworld.cpp

SophistCedar, 2019-09-04:
i have read the above link, and it contains some skeletons for main files for C++ applications.
but i donnot know what to do with it.
sorry to ask again, could you please tell me detailed solutions cause i am a beginner in esp32, 
thanks again.

chegewara, 2019-09-04:
Just copy hello world project from esp-idf or use this one:
https://github.com/espressif/esp-idf-template

then replace `main.c` with `main_helloworld.cpp`

SophistCedar, 2019-09-04:
sorry to say that, probably it my description problem.
i can compile hello_world project successfully, and i replace the original code with the code in mpu6050.c
but it occurs error when i compile the code in  mpu6050.c

SophistCedar, 2019-09-04:
i want to compile the mpu6050.c  in my computer

chegewara, 2019-09-04:
Your description is perfectly fine.
You cant replace original code with mpu6050 because you need entry function `app_main` in your app, which does not exist in mpu6050.c. Then from `app_main` you can call your mpu6050 code.

SophistCedar, 2019-09-04:
then what i can do to integrate entry function app_main in my app,

chegewara, 2019-09-04:
With one exception, because i assumed you want to use MPU6050.cpp library (ignore main_helloworld.cpp), read what i wrote, learn how to work with esp-idf...
What you are asking is basics and you can find many tutorials and vids on youtube and is not related to this library/repository.

SophistCedar, 2019-09-04:
Thank you very much, appreciate your help.
I will come back to write the solution and close issue when i solve it.

SophistCedar, 2019-09-05:
@chegewara, thank you for your kind help and suggestion, now i solve it.
firstly, i add the app main function in the end of  to call the subfunction task_mpu6050(), and it compiles successfully, but no result printed when i monitor the serial. the code is as follows:
void app_main()
{ 
	task_mpu6050(NULL);
}
secondly, i add a sentence code in line 109 in the mpu6050.h and it works well and show printed data from sensor, the code is as follows:
printf("accel_x = %d  accel_y = %d accel_z = %d\n" , (int)accel_x, (int)accel_y , (int)accel_z);

SophistCedar, 2019-09-05:
thanks for your great support again, have a good day!

closed 2019-09-05 by SophistCedar