nkolban/esp32-snippets #900: Esp32 Crash -> reboot ble on loop

Floriantoine, 2019-09-05:
# Resume 
### here is a very simplified version of my code, the goal being to be enable and disable the BL  to economise the Esp battery.
### It works but a Guru Meditation this product after X times

# Code 
```
#define DISPLAY_SSD1306Wire
#include <Arduino.h>
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

#define SERVICE_UUID        "012bd80f-14f6-42be-a45c-a62836a4fa3f"
#define CHARACTERISTIC_UUID "065de41b-79fb-479d-b592-47caf39bfccb"

BLEServer* pServer = NULL;
BLECharacteristic* pCharacteristic = NULL;

#define NOTIFY_CHARACTERISTIC

bool State = false;
bool oldState = false;

class ServerState : public BLEServerCallbacks
{
    void onConnect(BLEServer* pServer)
    {
      State = true;
    }

    void onDisconnect(BLEServer* pServer)
    {
      State = false;
    }
};

void setup()
{
  Serial.begin(115200);
  Serial.println("Test BLE init");

  Serial.println("Test BLE init server");

  BLEDevice::init("MonESP32");

  pServer = BLEDevice::createServer();
  pServer->setCallbacks(new ServerState());

  BLEService *pService = pServer->createService(SERVICE_UUID);

  pCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID,BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_WRITE  | BLECharacteristic::PROPERTY_NOTIFY | BLECharacteristic::PROPERTY_INDICATE);

  pService->start();
  pServer->getAdvertising()->start();

  Serial.println("Test BLE start advertising");
  Serial.println("Test BLE wait connection");
}

void loop()
{
    uint8_t loop = 0;
    while (loop < 1000) {
      loop++;
      pCharacteristic->setValue(&loop, 1);
      delay(10);
      Serial.printf("BLE notify : %d\n", loop);

      BLEDevice::deinit();
      delay(10);
      BLEDevice::init("MonESP32");

      pServer = BLEDevice::createServer();
      pServer->setCallbacks(new ServerState());
      BLEService *pService = pServer->createService(SERVICE_UUID);

      pCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID,BLECharacteristic::PROPERTY_READ | BLECharacteristic::PROPERTY_WRITE  | BLECharacteristic::PROPERTY_NOTIFY | BLECharacteristic::PROPERTY_INDICATE);

      pService->start();

      pServer->getAdvertising()->start();

      Serial.println("Test BLE start advertising");
      Serial.println("Test BLE wait connection");
    }
}
```
# Error
```
[...]
BLE notify : 24
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x4000c46c  PS      : 0x00060e30  A0      : 0x8011b8e2  A1      : 0x3ffdde30  
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x0000001c  A5      : 0x00000000  
A6      : 0x401a6c30  A7      : 0x00000001  A8      : 0x800dd00e  A9      : 0x3ffdddd0  
A10     : 0x3ffe2fe8  A11     : 0x00000044  A12     : 0x00000000  A13     : 0x00000000  
A14     : 0x00001800  A15     : 0x00000084  SAR     : 0x0000001e  EXCCAUSE: 0x0000001d  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x4000c46c:0x3ffdde30 0x4011b8df:0x3ffdde40 0x4011c1c1:0x3ffdde60 0x400f6ee9:0x3ffdde80 0x400f6f23:0x3ffddea0 0x40090461:0x3ffdded0

Rebooting...
[...]
```
### this error this product has the line "BLEDevice::init("MonESP32");" the 24th time.
Beware it depends on the number of characters, with 24 characters this being the 3rd time on line " pCharacteristic = pService->createCharacteristic(CHARACTERISTIC_UUID,BLECharacteristic::PROPERTY_READ | BLECharacteristic[...]);
".

chegewara, 2019-09-05:
Hi, 
it looks like memory leak issue.
Before deinit, did you delete all characteristics and services?

Floriantoine, 2019-09-05:
I test right now and I tell you

Floriantoine, 2019-09-05:
> Hi,
> it looks like memory leak issue.
> Before deinit, did you delete all characteristics and services?
> From this snippet i can see you did not, but each time you create new server and new characteristics.

I test right now and I tell you

Floriantoine, 2019-09-05:
@chegewara 
#### using delete or just service-> stop () and server-> removeService ()?

Floriantoine, 2019-09-05:
I have ADD
```
pService->stop();
pServer->removeService(pService);

delete pService;
delete pServer;
```
it improves I went from 25 to 35 times but still a crash

chegewara, 2019-09-05:
Still memory leak here and there. For example:
`pServer->setCallbacks(new ServerState());`
Every time you create new instance of this class.

It is also possible there is some memory leak in library, so try to delete pAdvertising too.