nkolban/esp32-snippets #902: ESP32 Crash on Android on IPhone it works

coltc50, 2019-09-08:
Hello,

I'm new at arduino and I don't know how to fix my problem. I asked at the arduino forum but didn't get a answer,

I try to get this script to work:
https://github.com/jstiefel/esp32_komoot_ble

When I try on IPhone everything works. But I don't have an IPhone and on Android the script is make a guru meditation after connect.

I tried a lot but didn't found the problem.

Here a Log:
`22:22:39.779 -> Scan again...
22:22:39.813 -> [V][BLEScan.cpp:195] start(): >> start(duration=5)
22:22:39.813 -> [D][FreeRTOS.cpp:189] take(): Semaphore taking: name: ScanEnd (0x3ffde244), owner: <N/A> for start
22:22:39.813 -> [D][FreeRTOS.cpp:198] take(): Semaphore taken:  name: ScanEnd (0x3ffde244), owner: start
22:22:39.813 -> [V][BLEScan.cpp:227] start(): << start()
22:22:39.813 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 2 0x02
22:22:39.813 -> [V][FreeRTOS.cpp:63] wait(): >> wait: Semaphore waiting: name: ScanEnd (0x3ffde244), owner: start for start
22:22:39.813 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
22:22:39.863 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
22:22:39.863 -> [D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
22:22:39.863 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 7 0x07
22:22:39.863 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
22:22:39.863 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
22:22:39.863 -> [D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
22:22:40.020 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
22:22:40.020 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
22:22:40.020 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
22:22:40.020 -> [D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
22:22:40.020 -> [D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -70
22:22:40.020 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0xff
22:22:40.020 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0xff (), length: 29, data: 0600010920029c6e758cd010f137d3aa01a2cffd485b1e4f07b7e3c652
22:22:40.061 -> [D][BLEAdvertisedDevice.cpp:395] setManufacturerData(): - manufacturer data: 0600010920029c6e758cd010f137d3aa01a2cffd485b1e4f07b7e3c652
22:22:40.061 -> BLE Advertised Device found: Name: , Address: 3a:ee:5b:43:b1:a7, manufacturer data: 0600010920029c6e758cd010f137d3aa01a2cffd485b1e4f07b7e3c652
22:22:40.061 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
22:22:40.061 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
22:22:40.112 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
22:22:40.112 -> [D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
22:22:40.112 -> [D][BLEScan.cpp:97] handleGAPEvent(): Ignoring 3a:ee:5b:43:b1:a7, already seen it.
22:22:40.230 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
22:22:40.230 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
22:22:40.230 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
22:22:40.230 -> [D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
22:22:40.230 -> [D][BLEScan.cpp:97] handleGAPEvent(): Ignoring 3a:ee:5b:43:b1:a7, already seen it.
22:22:40.230 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
22:22:40.230 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
22:22:40.278 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
22:22:40.278 -> [D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
22:22:40.278 -> [D][BLEScan.cpp:97] handleGAPEvent(): Ignoring 3a:ee:5b:43:b1:a7, already seen it.
22:22:41.168 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 3 0x03
22:22:41.168 -> [V][BLEUtils.cpp:1049] dumpGapEvent(): Received a GAP event: Unknown event type
22:22:41.168 -> [V][BLEUtils.cpp:1264] dumpGapEvent(): *** dumpGapEvent: Logger not coded ***
22:22:41.168 -> [D][BLEClient.cpp:458] handleGAPEvent(): BLEClient ... handling GAP event!
22:22:41.168 -> [D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -75
22:22:41.168 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x1
22:22:41.168 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 02
22:22:41.244 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x7
22:22:41.244 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x07 (), length: 16, data: 6c43b31d170fb2a2a84f2fd928e1c171
22:22:41.244 -> [D][BLEAdvertisedDevice.cpp:447] setServiceUUID(): - addServiceUUID(): serviceUUID: 71c1e128-d92f-4fa8-a2b2-0f171db343
22:22:41.244 -> [V][BLEUtils.cpp:746] advTypeToString():  adv data type: 0x9
22:22:41.244 -> [D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x09 (), length: 13, data: 485541574549204d6174652039
22:22:41.292 -> [D][BLEAdvertisedDevice.cpp:407] setName(): - setName(): name: HUAWEI Mate 9
22:22:41.292 -> BLE Advertised Device found: Name: HUAWEI Mate 9, Address: 6e:bb:db:72:4c:3f, serviceUUID: 71c1e128-d92f-4fa8-a2b2-0f171db343
22:22:41.292 -> Found our device!
22:22:41.292 -> [V][BLEScan.cpp:250] stop(): >> stop()
22:22:41.292 -> [V][FreeRTOS.cpp:143] give(): Semaphore giving: name: ScanEnd (0x3ffde244), owner: start
22:22:41.292 -> [V][BLEScan.cpp:262] stop(): << stop()
22:22:41.292 -> [V][FreeRTOS.cpp:77] wait(): << wait: Semaphore released: name: ScanEnd (0x3ffde244), owner: <N/A>
22:22:41.292 -> [V][BLEUtils.cpp:1817] gapEventToString(): gapEventToString: Unknown event type 18 0x12
[... 128 lines cut ...]
22:22:43.194 -> A10     : 0x0000004b  A11     : 0x00000062  A12     : 0x00000001  A13     : 0x00000000  
22:22:43.256 -> A14     : 0x00000000  A15     : 0xffffffac  SAR     : 0x00000004  EXCCAUSE: 0x0000001c  
22:22:43.256 -> EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffd  
22:22:43.256 -> 
22:22:43.256 -> Backtrace: 0x400d7f80:0x3ffc8890 0x400d8072:0x3ffc8900 0x400d208e:0x3ffc8960 0x400d23e7:0x3ffc89b0 0x400d24c3:0x3ffc89d0 0x400d38ed:0x3ffc8a10 0x400dd125:0x3ffc8af0 0x4008e5f1:0x3ffc8b10
22:22:43.256 -> 
22:22:43.256 -> Rebooting...
`

And here the output from ExceptionDecoder:
`PC: 0x400d7efc: BLERemoteDescriptor::toString[abi:cxx11]() at C:\Users\User\Documents\Arduino\hardware\Espressif\esp32\libraries\BLE\src\BLERemoteDescriptor.cpp line 117
EXCVADDR: 0x00000000

Decoding stack results
0x400d7efc: BLERemoteDescriptor::toString[abi:cxx11]() at C:\Users\User\Documents\Arduino\hardware\Espressif\esp32\libraries\BLE\src\BLERemoteDescriptor.cpp line 117
0x400d7fee: BLERemoteDescriptor::writeValue(unsigned char*, unsigned int, bool) at C:\Users\User\Documents\Arduino\hardware\Espressif\esp32\libraries\BLE\src\BLERemoteDescriptor.cpp line 132
0x400d2082: connectToServer() at C:\Users\User\AppData\Local\Temp\arduino_build_877457\sketch\Connection.cpp line 91
0x400d23c7: connect() at C:\Users\User\AppData\Local\Temp\arduino_build_877457\sketch\Connection.cpp line 163
0x400d24a3: scan() at C:\Users\User\AppData\Local\Temp\arduino_build_877457\sketch\Connection.cpp line 108
0x400d3817: setup() at C:\Users\User\Documents\Arduino\esp32_komoot_ble/esp32_komoot_ble.ino line 65
0x400dd08f: loopTask(void*) at C:\Users\User\Documents\Arduino\hardware\Espressif\esp32\cores\esp32\main.cpp line 14
0x4008e5f1: vPortTaskWrapper at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/port.c line 143
`

What's the problem and how I can fix this?

chegewara, 2019-09-09:
`getDescriptor(): << getDescriptor: Not found `

coltc50, 2019-09-09:
Thank you. But how can I fix this?

When I comment these two lines out it didn't crash but then it doesn't notify...

    //pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)notificationOn, 2, true);
   
    //pRemoteCharacteristic->registerForNotify(notifyCallback);

chegewara, 2019-09-10:
Fix android app.

coltc50, 2019-09-10:
That I cannot do because it's the official Komoot App.

Is there no work around I can do on ESP32 side?

Maybe hardcoding address or something?

chegewara, 2019-09-10:
Like i said earlier, this means android app.

What you could do is to comment out those lines:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLERemoteCharacteristic.cpp#L465-L466

and this one in your app:
```
//pRemoteCharacteristic->getDescriptor(BLEUUID((uint16_t)0x2902))->writeValue((uint8_t*)notificationOn, 2, true);
```

coltc50, 2019-09-10:
It works, it works... Thank you so much. No problem to break IPhone because I didn't have one... 
How can I donate? Thank you again!

Now I can make me thoughts about a housing.

Great. You made my day!

closed 2019-09-10 by chegewara

chegewara, 2019-09-10:
Im glad it worked.

coltc50, 2019-10-09:
Sorry to post again but since a few days it doesn't work anymore. Now it stuck anytime. Don't know witch update caused this.

Here Log:
`19:44:38.915 ->  - Found our characteristic
19:44:38.915 -> CanRead : 1
19:44:38.915 -> CanWrite : 0
19:44:38.915 -> CanNotify : 1
19:44:38.915 -> CanIndicate : 0
19:44:38.915 -> [D][BLERemoteCharacteristic.cpp:444] registerForNotify(): >> registerForNotify(): Characteristic: uuid: 503dd605-9bcb-4f6e-b235-270a57483026, handle: 56 0x38, props: broadcast: 0, read: 1, write_nr: 0, write: 0, notify: 1, indicate: 0, auth: 0
19:44:38.950 -> [D][FreeRTOS.cpp:165] take(): Semaphore taking: name: RegForNotifyEvt (0x3ffed968), owner: <N/A> for registerForNotify
19:44:38.950 -> [D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: RegForNotifyEvt (0x3ffed968), owner: registerForNotify
19:44:38.950 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 38
19:44:38.950 -> [D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
19:44:38.985 -> [D][BLERemoteCharacteristic.cpp:329] getDescriptor(): >> getDescriptor: uuid: 00002902-0000-1000-8000-00805f9b34fb
19:44:38.985 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 38
19:44:38.985 -> [D][BLERemoteCharacteristic.cpp:337] getDescriptor(): << getDescriptor: Not found
19:44:38.985 -> [V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
19:44:38.985 -> [V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: RegForNotifyEvt (0x3ffed968), owner: registerForNotify for registerForNotify
19:44:39.019 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 38
19:44:39.019 -> [D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
19:44:39.019 -> [V][FreeRTOS.cpp:120] give(): Semaphore giving: name: RegForNotifyEvt (0x3ffed968), owner: registerForNotify
19:44:39.019 -> [W][Connection.cpp:182] my_gattc_event_handler(): custom gattc event handler, event: 38
19:44:39.019 -> [V][FreeRTOS.cpp:86] wait(): << wait: Semaphore released: name: RegForNotifyEvt (0x3ffed968), owner: registerForNotify
19:44:39.054 -> [D][BLERemoteCharacteristic.cpp:484] registerForNotify(): << registerForNotify()
19:44:39.054 -> 
19:44:40.042 -> Loop
19:44:41.051 -> Loop
19:44:42.025 -> Loop
19:44:43.030 -> Loop
19:44:43.406 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 15
19:44:43.406 -> [D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
19:44:43.406 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 15
19:44:43.406 -> [V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
19:44:43.406 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 15
19:44:43.406 -> [D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
19:44:43.453 -> [I][BLEClient.cpp:171] gattClientEventHandler(): SERVICE CHANGED
19:44:43.453 -> [W][Connection.cpp:182] my_gattc_event_handler(): custom gattc event handler, event: 15
19:44:44.069 -> Loop
19:44:44.069 -> Connection lost or navigation turned off! Reconnecting...
19:44:44.069 -> [D][BLEClient.cpp:146] disconnect(): >> disconnect()
19:44:44.069 -> [D][BLEClient.cpp:152] disconnect(): << disconnect()
19:44:44.069 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 5
19:44:44.069 -> [D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
19:44:44.069 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 5
19:44:44.069 -> [V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
19:44:44.139 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 5
19:44:44.139 -> [D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
19:44:44.139 -> [W][Connection.cpp:182] my_gattc_event_handler(): custom gattc event handler, event: 5
19:44:44.139 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
19:44:44.139 -> [D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
19:44:44.139 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
19:44:44.174 -> [V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
19:44:44.174 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
19:44:44.174 -> [D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
19:44:44.174 -> [I][BLEDevice.cpp:607] removePeerDevice(): remove: 0, GATT role client
19:44:44.174 -> [V][FreeRTOS.cpp:120] give(): Semaphore giving: name: RssiCmplEvt (0x3ffeb454), owner: <N/A>
19:44:44.174 -> [V][FreeRTOS.cpp:120] give(): Semaphore giving: name: SearchCmplEvt (0x3ffeb3f4), owner: <N/A>
19:44:44.209 -> [W][Connection.cpp:182] my_gattc_event_handler(): custom gattc event handler, event: 41
19:44:44.209 -> 
19:44:44.209 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 1
19:44:44.209 -> [D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[... cut ...]
19:45:14.541 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 1
19:45:14.541 -> [D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
19:45:14.541 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 1
19:45:14.574 -> [V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
19:45:14.574 -> [W][Connection.cpp:182] my_gattc_event_handler(): custom gattc event handler, event: 1
`

chegewara, 2019-10-10:
Make sure you are using library updated with this PR (on arduino only), or code will stuck on semaphore:
https://github.com/espressif/arduino-esp32/pull/2974
```
19:44:44.347 -> Reconnecting to 75:xx:xx:xx:e0:93
19:44:44.347 -> [D][BLEClient.cpp:103] connect(): >> connect(75:xx:xx:xx:e0:93)
19:44:44.347 -> [I][BLEDevice.cpp:596] addPeerDevice(): add conn_id: 1, GATT role: client
19:44:44.347 -> [D][FreeRTOS.cpp:165] take(): Semaphore taking: name: RegEvt (0x3ffeb140), owner: <N/A> for connect
19:44:44.347 -> [D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: RegEvt (0x3ffeb140), owner: connect
19:44:44.347 -> [V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: RegEvt (0x3ffeb140), owner: connect for connect
19:44:44.347 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 0
19:44:44.381 -> [D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
19:44:44.381 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 0
19:44:44.381 -> [V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
19:44:44.381 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 0
19:44:44.381 -> [D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
19:44:44.381 -> [V][FreeRTOS.cpp:120] give(): Semaphore giving: name: RegEvt (0x3ffeb140), owner: connect
19:44:44.415 -> [W][Connection.cpp:182] my_gattc_event_handler(): custom gattc event handler, event: 0
19:44:44.415 -> [V][FreeRTOS.cpp:86] wait(): << wait: Semaphore released: name: RegEvt (0x3ffeb140), owner: connect
19:44:44.415 -> [D][FreeRTOS.cpp:165] take(): Semaphore taking: name: OpenEvt (0x3ffeb394), owner: <N/A> for connect
19:44:44.415 -> [D][FreeRTOS.cpp:174] take(): Semaphore taken:  name: OpenEvt (0x3ffeb394), owner: connect
19:44:44.415 -> [V][FreeRTOS.cpp:70] wait(): >> wait: Semaphore waiting: name: OpenEvt (0x3ffeb394), owner: connect for connect
19:45:14.472 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
19:45:14.472 -> [D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
19:45:14.472 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
19:45:14.472 -> [V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
19:45:14.472 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 41
19:45:14.472 -> [D][BLEClient.cpp:165] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
19:45:14.506 -> [I][BLEDevice.cpp:607] removePeerDevice(): remove: 1, GATT role client
19:45:14.506 -> [V][FreeRTOS.cpp:120] give(): Semaphore giving: name: RssiCmplEvt (0x3ffeb454), owner: <N/A>
19:45:14.506 -> [V][FreeRTOS.cpp:120] give(): Semaphore giving: name: SearchCmplEvt (0x3ffeb3f4), owner: <N/A>
19:45:14.506 -> [W][Connection.cpp:182] my_gattc_event_handler(): custom gattc event handler, event: 41
19:45:14.506 -> 
19:45:14.506 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 2
19:45:14.506 -> [D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
19:45:14.541 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 2
19:45:14.541 -> [V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
19:45:14.541 -> [W][Connection.cpp:182] my_gattc_event_handler(): custom gattc event handler, event: 2
19:45:14.541 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 1
19:45:14.541 -> [D][BLEDevice.cpp:150] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
19:45:14.541 -> [V][BLEUtils.cpp:952] gattClientEventTypeToString(): Unknown GATT Client event type: 1
19:45:14.574 -> [V][BLEUtils.cpp:1284] dumpGattClientEvent(): GATT Event: Unknown
19:45:14.574 -> [W][Connection.cpp:182] my_gattc_event_handler(): custom gattc event handler, event: 1
```

coltc50, 2019-10-10:
@chegewara 
Again big thanks to you. It seems I had an old library in. Deleted all double and now its working. Thank you again.