nkolban/esp32-snippets #904: CLIENT SYSTEM IS NOT GOING TO SLEEP MODE

GKCHDEV, 2019-09-16:
Hi,

I am implementing the HID keyboard using ESP32 through ESP-IDF for that i am using the below library
https://github.com/asterics/esp32_mouse_keyboard

It is working fine but when my BLE is connected ,client system is not going to sleep mode it is waking automatically after few seconds.

i am requesting you could you please help on this

chegewara, 2019-09-28:
Hi,
i think sleep mode is not implemented in HID, but you can try to stop sending notifications.

In HID keyboard there are OUTPUT and INPUT report characteristic, one is to send key codes the other is to receive info from client, for example info about CAPS LOCk etc. It is possible that client is sending data to let keyboard know when is entering sleep mode (but im not sure). Try to catch all messages on that characteristic.

GKCHDEV, 2019-11-03:
Hi thanks for your reply we are stopping the notifications  when client is disconnecting using below functions even though it is waking automatically i guess because of advertising how can i stop the advertising please help me.

 if(activateKeyboard)
    {
      desc = (BLE2902*) inputKbd->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
      desc->setNotifications(false);
      kbd->stop();
    }
    
    if(activateMouse)
    {
      desc = (BLE2902*) inputMouse->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
      desc->setNotifications(false);
      mouse->stop();
    }
    
    if(activateJoystick)
    {
      desc = (BLE2902*) inputJoystick->getDescriptorByUUID(BLEUUID((uint16_t)0x2902));
      desc->setNotifications(false);
      joystick->stop();
    }

GKCHDEV, 2019-11-05:
i want to stop the advertising when client is going to sleep mode because of advertising my client system is not going to sleep mode  mode .as well as i want connect automatically when client is wakes please help me on this