nkolban/esp32-snippets #906: compile problem

marrog86, 2019-09-20:
Hello, 

Wanted to try out your work, but i had problems compiling it. 
idf.py --version
ESP-IDF v4.1-dev-281-g96b96ae24

cpp_utils/Socket.cpp:63:23: error: '::lwip_accept_r' has not been declared
  int clientSockFD = ::lwip_accept_r(m_sock,  (struct sockaddr*) &client_addr, &sin_size);

any idea what i could try?

chegewara, 2019-09-27:
Most likely some API changed since esp-idf v3.x and you will have to change this library if you want to use it with esp-idf v4.x.

stephenrjohnson, 2019-12-30:
Is there any alternative to this for the esp32, as the snippets repo is becoming abandoned ware

chegewara, 2019-12-30:
You can fork repo and apply required changes in code and eventually create PR with new branch.
Remember that esp-idf v3.3 is still last stable release.