nkolban/esp32-snippets #907: Sending Dynamic Advertise Data

hazee007, 2019-09-21:
Hello guy 
i have ble gateway(not arduino) and i want to use esp32 as an ibeacon to transmit sensor data to the gateway without connecting to the gateway. the gateway just act as a ble scanner the get uuid, major, minor and so on, what i want is i want the gateway to read the sensor data also.

is there anyways to achieve this with esp32.
Thank you

chegewara, 2019-09-27:
I know you are not working with arduino but in this case example from link can be also used:
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_iBeacon/BLE_iBeacon.ino#L50-L68

Just use this API to change advertising on the fly.

hazee007, 2019-09-28:
> I know you are not working with arduino but in this case example from link can be also used:
> https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/tests/BLETests/Arduino/BLE_iBeacon/BLE_iBeacon.ino#L50-L68
> 
> Just use this API to change advertising on the fly.

i was able to achieve this using this solution you provided but now i am facing a different problem,
i want to advertise sensor data using the ble on esp32.
i am using mpu6050 to get 6 axis value to get the x,y,z value of both accelerometer and gyroscope as well as a step count.
i got some help from [https://github.com/nkolban/esp32-snippets/issues/375](url) on making the advertising data dynamic to send the data but the issue here is that the advertisement data is 31 byte long(size) is there anyway i can increase this or another way around it..
here is my code below 
`#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>
#include "Wire.h"
#include "SPI.h"
#include "SD.h"

#define MPU6050_ACCEL_XOUT_H       0x3B   // R 
#define MPU6050_PWR_MGMT_1         0x6B   // R/W
#define MPU6050_PWR_MGMT_2         0x6C   // R/W
#define MPU6050_WHO_AM_I           0x75   // R

float gyro_xx;
float gyro_yy;
float gyro_zz;
float gyro_x;
float gyro_y;
float gyro_z;
float accel_x;
float accel_y;
float accel_z;

#define MPU6050_I2C_ADDRESS 0x68
typedef union accel_t_gyro_union
{
  struct
  {
    uint8_t x_accel_h;
    uint8_t x_accel_l;
    uint8_t y_accel_h;
    uint8_t y_accel_l;
    uint8_t z_accel_h;
    uint8_t z_accel_l;
    uint8_t t_h;
    uint8_t t_l;
    uint8_t x_gyro_h;
    uint8_t x_gyro_l;
    uint8_t y_gyro_h;
    uint8_t y_gyro_l;
    uint8_t z_gyro_h;
    uint8_t z_gyro_l;
  } reg;
  struct
  {
    int16_t x_accel;
    int16_t y_accel;
    int16_t z_accel;
    int16_t temperature;
    int16_t x_gyro;
    int16_t y_gyro;
    int16_t z_gyro;
  } value;
};

// Use the following global variables and access functions to help store the overall
// rotation angle of the sensor
unsigned long last_read_time;
float         last_x_angle;  // These are the filtered angles
float         last_y_angle;
float         last_z_angle; 
[... 294 lines cut ...]
    return (-20);

  n = Wire.write(pData, size);  // write data bytes
  if (n != size)
    return (-21);

  error = Wire.endTransmission(true); // release the I2C-bus
  if (error != 0)
    return (error);

  return (0);         // return : no error
}

int MPU6050_write_reg(int reg, uint8_t data)
{
  int error;

  error = MPU6050_write(reg, &data, 1);

  return (error);
}

`

Thanks...

chegewara, 2019-09-28:
No, you cant increase it, because its bluetooth specification. What you can do is split your data and half send in advertising data, the other half send in scan response data.

hazee007, 2019-10-03:
> No, you cant increase it, because its bluetooth specification. What you can do is split your data and half send in advertising data, the other half send in scan response data.

Can  you please a reference as to how i can be able to achieve this... my current code is 
```
#include <BLEDevice.h>
#include <BLEUtils.h>
#include <BLEServer.h>

BLEAdvertisementData advert;
BLEAdvertising *pAdvertising;

int i = 0;

//manufacturer code (0x02E5 for Espressif)
int man_code = 0x02E5;

//function takes String and adds manufacturer code at the beginning 
void setManData(String c, int c_size, BLEAdvertisementData &adv, int m_code) {
  
  String s;
  char b2 = (char)(m_code >> 8);
  m_code <<= 8;
  char b1 = (char)(m_code >> 8);
  s.concat(b1);
  s.concat(b2);
  s.concat(c);
  adv.setManufacturerData(s.c_str());
  
}

void setup() {
  Serial.begin(115200);
  Serial.println("Starting BLE work!");

  BLEDevice::init("MyESP32");
  BLEServer *pServer = BLEDevice::createServer();

  pAdvertising = pServer->getAdvertising();
  advert.setName("ESP32-new");
  pAdvertising->setAdvertisementData(advert);
  pAdvertising->start();
}

void loop() {
  i++;
  String a = String(i);
  
  BLEAdvertisementData scan_response;
  setManData(a, a.length() , scan_response, man_code);

  pAdvertising->stop();
  pAdvertising->setScanResponseData(scan_response);
  pAdvertising->start();
 
  delay(2000);
}
```
Thanks