nkolban/esp32-snippets #908: Updating data on server and client

usman1515, 2019-09-26:
I am trying to create a server and client esp32. Im trying to send temperature and humidity data accquired via dht11 from server to a client. I have configured the server and the client and the dht11. 
First, I'm not sure whether to add the dht11 on the server or the client side? Any new and improved approach for solving this problem is welcomed.

Second, I tried calling this function in `void loop()`  on server side but it gives an error.
`temperaturecharacteristic->setValue(String(dht11_temp));`
`humiditycharacteristic->setValue(String(dht11_humid));`

Third, I found this function on the client side. Can it be used to update temp on the client side.
`temperaturecharacteristic->writeValue(dht11_temp.c_str(), dht11_temp.length());`
`humiditycharacteristic->writeValue(dht11_humid.c_str(), dht11_humid.length());`

Thanks

chegewara, 2019-09-27:
1. dht11 is a sensor, sensors usually are working as BLE server
2. `temperaturecharacteristic->setValue(String(dht11_temp));` if i remember correct there is no function that allows to use arduino String as parameter

usman1515, 2019-09-29:
Yes you are correct. The string function gives an error. I was able to fix it by using `std::string`.
One problem still persists. I attached dht11 on the client side. It updates the temperature and humidity characteristics at regular delays without any erros. Code attached below.
### void loop code implemented on client when connection established
```
if (connected)
  {
    h = dht.readHumidity();
    tc = dht.readTemperature();       //°C

    //Check if any reads failed and exit early (to try again).
    if (isnan(h) || isnan(tc) || isnan(tf)) {
      Serial.println(F("Failed to read from DHT sensor!"));
      return;
    }
    else {
      //print_dht11(tc, tf, h, hic, hif);
      
      String Humidvalue = "Humidity: " + String(h) + " %";
      Serial.println("New characteristic value:" + Humidvalue);
      //Set characteristic value to be array of bytes that is actually a string.
      HumidChar->writeValue(Humidvalue.c_str(), Humidvalue.length());

      String Ctempvalue = "Temperature: " + String(tc) + " °C";
      Serial.println("New characteristic value:" + Ctempvalue);
      //Set characteristic value to be array of bytes that is actually a string.
      CtempChar->writeValue(Ctempvalue.c_str(), Ctempvalue.length());
    }
  }
```
But it doesnt update the same characteristic on the server side. Any idea on what i might be doing wrong and how to fix it.
### void loop code implemented on server when connection established 
```
if (deviceConnected)
  {
    //when client transmits temperature value
    std::string Ctempstr = Ctempchar->getValue();
    //Ctempstr = "Temperature: " + Ctempstr.c_str() + " °C";
    Serial.print("Characteristic value: ");
    Serial.println(Ctempstr.c_str());
    Ctempchar->setValue(Ctempstr);
    Ctempchar->notify();
    delay(20);

    //when client transmits humidity value
    std::string Humidstr = Humidchar->getValue();
    //Humidstr = "Humidity: " + Humidstr.c_str() + " °C";
    Serial.print("Characteristic value:");
    Serial.println(Humidstr.c_str());
    Humidchar->setValue(Humidstr);
    Humidchar->notify();
    delay(20);

    //CtempCharacteristicCallbacks::onWrite(Ctempchar);
    //HumidCharacteristicCallbacks::onWrite(Humidchar);
  }
```

chegewara, 2019-09-29:
Im not sure i understand what you are trying to do. 
You should set temp and humid values on server and send notify, and client should only register for notify and receive values. 
What you do now is write to server from client, then replace that value on server and send notify with new values.

closed 2019-10-04 by usman1515