nkolban/esp32-snippets #910: BLE_client disconnecting from Nordic UART

HeadTek, 2019-09-26:
Hopefully this is the correct place for my question.
And by the way, great examples and great work. New to this and still learning but learning a lot looking through these sketches and info. Have a question about what could be causing my immediate and constant disconnects from Nordic UART

UUIDs 
static BLEUUID serviceUUID("6e400001-b5a3-f393-e0a9-e50e24dcca9e");
static BLEUUID    charUUID("6e400001-b5a3-f393-e0a9-e50e24dcca9e");

Serial.Prints
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:928
ho 0 tail 12 room 4
load:0x40078000,len:8740
load:0x40080400,len:5800
entry 0x4008069c
Starting Arduino BLE Client application...
BLE Advertised Device found: Name: iDevice_C118330107202, Address: d6:0b:17:92:b4:ba, serviceUUID: 6e400001-b5a3-f393-e0a9-e50e24dcca
Forming a connection to d6:0b:17:92:b4:ba
 - Created client
onDisconnect
 - Connected to server
---------------------------------------------------
Verbose

Starting Arduino BLE Client application...
[D][BLEAdvertisedDevice.cpp:418] setRSSI(): - setRSSI(): rssi: -92
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x01 (), length: 1, data: 05
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x09 (), length: 19, data: 69466c65785f43313138333330313037323032
[D][BLEAdvertisedDevice.cpp:407] setName(): - setName(): name: iFlex_C118330107202
[D][BLEAdvertisedDevice.cpp:247] parseAdvertisement(): Type: 0x07 (), length: 16, data: 9ecadc240ee5a9e093f3a3b50100406e
[D][BLEAdvertisedDevice.cpp:447] setServiceUUID(): - addServiceUUID(): serviceUUID: 6e400001-b5a3-f393-e0a9-e50e24dcca
BLE Advertised Device found: Name: iDevice_C118330107202, Address: d6:0b:17:92:b4:ba, serviceUUID: 6e400001-b5a3-f393-e0a9-e50e24dcca
Forming a connection to d6:0b:17:92:b4:ba
[V][BLEDevice.cpp:66] createClient(): >> createClient
[V][BLEDevice.cpp:72] createClient(): << createClient
[V][BLEClient.cpp:96] connect(): >> connect(d6:0b:17:92:b4:ba)
[I][BLEDevice.cpp:599] addPeerDevice(): add conn_id: 0, GATT role: client
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[D][BLEClient.cpp:158] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown

[V][BLEClient.cpp:129] connect(): << connect(), rc=0
[I][BLEDevice.cpp:610] removePeerDevice(): remove: 0, GATT role client
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEClient.cpp:377] getService(): >> getService: uuid: 6e400001-b5a3-f393-e0a9-e50e24dcca
[D][BLEDevice.cpp:154] gattClientEventHandler(): gattClientEventHandler [esp_gatt_if: 4] ... Unknown
[V][BLEClient.cpp:413] getServices(): >> getServices
[V][BLEClient.cpp:71] clearServices(): >> clearServices
[V][BLEClient.cpp:78] clearServices(): << clearServices
----------------------------------------------------------------------
Also, using Nordic nRF won't keep connection either. 

Any help on this would be greatly appreciated. I've spent a lot of time on this and not making any traction.

Thanks,
Dan

chegewara, 2019-09-27:
Why do you have the same UUID for service and characteristic?
```
static BLEUUID serviceUUID("6e400001-b5a3-f393-e0a9-e50e24dcca9e");
static BLEUUID charUUID("6e400001-b5a3-f393-e0a9-e50e24dcca9e");
```

HeadTek, 2019-09-28:
@chegewara I am also trying to understand why this is the case for Nordic UART devices but I have yet to find a resource to explain why this is the case. But as for BLE UART devices they only have 2 characteristics TX and RX for serial transmit and receive. These two characteristics have the same UUID and those UUIDs are the same as the service UUID. These are the exact same UUIDs across all Nordic UART BLE devices (as far as i've seen so far). See attached image. 
I think this might be part the reason for the ESP32_BLE_Client to not be working with Nordic UART. 
Not sure if you  are aware of the work done by ThingEngineer. If not, take a look here 
[https://github.com/ThingEngineer/ESP32_BLE_client_uart](url)
ThingEngineer gave credit to this sketch as the basis of the modified sketch to support UART
This looked like exactly what i needed but i get the same disconnects of the Service. I see some slight changes in how the objects are referenced in that sketch so i came back to this BLE_Client sketch to see if going back to the basics might shed some light on this issue. 
[image]

chegewara, 2019-09-28:
If you will look closer then you will see that all 3 UUIDs are different.

HeadTek, 2019-09-28:
@chegewara OMG, My eyes are going with my age. I should have paid more attention the image attached here as it is much larger than on my phone. lol. 
Referring back to ThingEngineer's sketch has the three correct UUIDs as the image but still continue to disconnect or just not connect to the service at all. Even the screenshot from my last post, it was hard to get the shot on my phone because the nRF app wouldn't establish a connection but only for brief second. I had to keep hitting connect and almost instantly click for screenshot. Possible this is a security issue on this device?

chegewara, 2019-09-28:
It is hard to say, because those logs does not say anything.
You can always check nRF logs, there should be disconnect reason.

HeadTek, 2019-09-28:
Below is the UART log. sorry i only have a screen shot of log of nRF Toolbox set to auto-connect but this is what is scrolling contently with ERROR: (0x13) GATT CONN TERMINATE PEER USER. It'll reconnect and keep throwing the error.
[image]

chegewara, 2019-09-28:
For some reason esp32 is causing disconnection (clean disconnection). Please check your code.

italocjs, 2023-11-14:
I am having the exact same issue, BLE UART works for an hour or so, then disconnect and does not allow any other device to connect.  There is no error message on the terminal.   Has anyone found an possible cause?

italocjs, 2023-12-13:
A month later i managed to write a workaround to this by checking if my device has disconnected and trying to connect again,  which is not an optimal way,  something is wrong with our example code.

I also have another issue, any package above 20 bytes makes the BLE disconnect without any notice or reason