nkolban/esp32-snippets #911: BLE Client: getCharacteristics doesn't get any values

skye0402, 2019-09-28:
Dear experts,

I try to connect to a Decawave Module which uses a Nordic BLE Chip nRF52832. I can connect successfully with the Android nRF app.
Now the [client app](https://github.com/skye0402/ic2019esp32/blob/master/ESP32_BLE_Client_Decawave_test03) that I use only gets to the point saying:

> ESP32 BLE Server program
> Scan Result: Name: , Address: 38:a0:d6:a3:ca:c5, manufacturer data: 06000109210a63c058508cf454594f4e353635373930343941 
> Scan Result: Name: , Address: c7:69:d1:f4:79:4e 
> Found Device :-)... connecting to Server as client
>  - Created client
>  - Connected to fitnessband
>  - Found our service
> ********************LED turned ON************************

However, it doesn't connect to the characteristic and so I tried to read all available characteristics (nRF shows many, maybe 20). But it seems this variable is null.

Any expert here that can check if I made a coding mistake? I try now since days to connect and can't manage. I see that the blue LED of the server goes on which means BLE is being connected to, but then goes off again.

Thank you for your help!

chegewara, 2019-09-28:
There is additional parameter in connect() function. Try to use `BLE_ADDR_TYPE_RANDOM`.
https://github.com/nkolban/esp32-snippets/blob/master/cpp_utils/BLEClient.h#L37

skye0402, 2019-09-29:
Thank you very much for that hint. 
`pClient->connect(pAddress, BLE_ADDR_TYPE_RANDOM);`

Is this how you meant it? Now the program stops after output of line

> ESP32 BLE Server program
> Scan Result: Name: , Address: 25:7e:20:f9:bf:92, manufacturer data: 06000109210a63c058508cf454594f4e353635373930343941 
> Scan Result: Name: , Address: c7:69:d1:f4:79:4e 
> Found Device :-)... connecting to Server as client
>  - Created client

And it doesn't move further. Maybe crashed? I'm wondering what ways to I have to debug it.