nkolban/esp32-snippets #913: BLERemoteCharacteristic::writeValue  with negative numbers

HeadTek, 2019-10-07:
I have an array of char, some values are signed. looking at the implementation of writeValue function, it's using uint8_t. Will this still work?

axa88, 2019-10-08:
Yes.
They are both 8 bit values, and uint8_t is the most encompassing type of the two.
An 8 bit value is sent regardless of the specific value the sender or receiver believes it to be

HeadTek, 2019-10-08:
@axa88 so, just to be clear, your saying even though i have signed values the unsigned uint8_T will still send the negative value?

axa88, 2019-10-09:
Yes.
You can just take that as your answer, but as a friendly recommendation I would recommend looking into the difference of signed and unsigned numbers, to get a better understanding of any implications when using them with signed and unsigned types